coolsnowwolf/lede #13200: 编译失败，错误信息：elfutils

jw10126121, 2024-12-11:
### 详细叙述

20241211最新master分支源码在MacOS环境下编译失败，找到类似的错误，但看回复已修复，但实测还是出现。

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

AX18

### 详细日志

make[3]: *** [Makefile:38: /Volumes/OpenWrt/wrt/build_dir/host/elfutils-0.191/.configured] Error 1
make[3]: Leaving directory '/Volumes/OpenWrt/wrt/tools/elfutils'
time: tools/elfutils/compile#4.35#1.58#7.57
    ERROR: tools/elfutils failed to build.
make[2]: *** [tools/Makefile:231: tools/elfutils/compile] Error 1
make[2]: Leaving directory '/Volumes/OpenWrt/wrt'
make[1]: *** [tools/Makefile:227: /Volumes/OpenWrt/wrt/staging_dir/host/stamp/.tools_compile_nyyynyyynnnyyyyynyyyyyyynyyynynyynnynnyyyyyynyyyyyyyyyyynynnyyyyynyyy] Error 2
make[1]: Leaving directory '/Volumes/OpenWrt/wrt'
make: *** [/Volumes/OpenWrt/wrt/include/toplevel.mk:231: world] Error 2

Young-045, 2025-02-14:
请问解决了吗，我也是mac编译报错了
checking whether the compiler generates build-ids... ./configure: line 6978: -n: command not found
no
configure: WARNING: compiler doesn't generate build-id by default
checking whether gcc supports -Wl,-z,relro... no
checking for __thread support... no
configure: error: __thread support required
make[3]: *** [Makefile:38: /Volumes/PortableSSD/lede/build_dir/host/elfutils-0.191/.configured] Error 1
make[3]: Leaving directory '/Volumes/PortableSSD/lede/tools/elfutils'
time: tools/elfutils/compile#3.47#0.89#5.86
    ERROR: tools/elfutils failed to build.
make[2]: *** [tools/Makefile:231: tools/elfutils/compile] Error 1
make[2]: Leaving directory '/Volumes/PortableSSD/lede'
make[1]: *** [tools/Makefile:227: /Volumes/PortableSSD/lede/staging_dir/host/stamp/.tools_compile_nyyynyyynnnyyyyynyyyyyyynyyynynyynnynnyyyyyynyyyyyyyyyyynynnyyyyyyyyy] Error 2
make[1]: Leaving directory '/Volumes/PortableSSD/lede'
make: *** [/Volumes/PortableSSD/lede/include/toplevel.mk:231: world] Error 2

coolsnowwolf, 2026-09-05:
已按当前源码核对。该日志来自 2024 年的 elfutils 0.191；当前已更新为 0.192。更关键的是，现行 `tools/Makefile` 在 `HOST_OS=Darwin` 时只加入 macOS 所需的 host bash，`dwarves -> elfutils` 依赖位于非 Darwin 分支，因此全新 macOS 编译不会再进入 `tools/elfutils`，原报错路径已不适用。

若旧工作目录仍命中 `tools/elfutils`，请先确认已完整更新源码和 feeds，并清理旧的 `build_dir`、`staging_dir`、`tmp` 后按 README 的 macOS 工具链步骤重建；不要复用 2024 年生成的 host 工具/依赖 stamp。按当前代码已解决关闭。

closed 2026-09-05 by coolsnowwolf