coolsnowwolf/lede #13202: 打不开luci

wsppiup, 2024-12-11:
### 详细叙述

和之前一样的编译方法，但是编译后的固件打不开luci

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

X86

### 详细日志

Error
Unhandled exception during request dispatching
/usr/lib/lua/luci/ucodebridge.lua:23: /usr/lib/lua/luci/template.lua:181: Failed to execute template 'themes/ifit/header'.
A runtime error occurred: /usr/lib/lua/luci/dispatcher.lua:68: attempt to index global '__entries' (a nil value)
stack traceback:
/usr/lib/lua/luci/dispatcher.lua:68: in function 'node'
[string "/usr/lib/lua/luci/view/themes/ifit/header.h..."]:20: in main chunk

In error(), file [C]
called from function run (/usr/lib/lua/luci/ucodebridge.lua:23)
called from function [anonymous function] (/usr/lib/lua/luci/ucodebridge.lua:37)
In [anonymous function](), file /usr/share/ucode/luci/runtime.uc, line 133, byte 10:
  called from function [arrow function] (/usr/share/ucode/luci/runtime.uc:141:63)
  called from function render ([C])
  called from function [anonymous function] (/usr/share/ucode/luci/runtime.uc:141:64)
  called from function [anonymous function] (/usr/share/ucode/luci/dispatcher.uc:941:44)
  called from anonymous function (/www/cgi-bin/luci:39:12)

 `            die(ex);`
  Near here --------^

tsingui, 2024-12-12:
试试安luci-compat插件

wsppiup, 2024-12-12:
默认安装了

coolsnowwolf, 2024-12-13:
别用不兼容的主题

wsppiup, 2024-12-13:
是的，主题问题，谢谢

coolsnowwolf, 2026-09-05:
日志中的异常来自第三方 `luci-theme-ifit`：其旧版 Lua 模板调用了当前 LuCI dispatcher 中已不存在的全局 `__entries`，并非 LuCI 核心无法启动。报告者已在评论中确认更换兼容主题后恢复正常，因此按已解决关闭。若该主题仍需适配，请向主题仓库提交 issue，并提供其准确版本。

closed 2026-09-05 by coolsnowwolf