coolsnowwolf/lede #13204: LEDE最新分支luci openwrt-23.05编译docker页面显示不完整仅显示“配置”页面

tsingui, 2024-12-12:
### 详细叙述

docker页面显示不完整仅显示“配置”页面
尝试从应用商店重新安装开始提示firewall与firewall4依赖相互冲突，不知是否有关于？
[image]
[image]

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

QEMU ARM Virtual Machine （N1）

### 详细日志

编译过程正常！

xingshu1990, 2024-12-12:
试着删除luci-app-docker，单独编译luci-app-dockerman?

tsingui, 2024-12-12:
> 试着删除luci-app-docker，单独编译luci-app-dockerman?

编译时没添加luci-app-docker插件，最新luci分支openwrt-23.05也没这

coolsnowwolf, 2024-12-12:
选上 dockerd 和 docker 就有了

lizheyo77, 2024-12-12:
同样的问题 已编译docker

tsingui, 2024-12-13:
> 选上 dockerd 和 docker 就有了
编译选docker-compose、dockerd 和 docker一起编译就是没出来，暂时解决办法从luci-app-dockerman插件作者拉取源码替换，重新编译后正常了

lizheyo77, 2024-12-13:
> > 选上 dockerd 和 docker 就有了
> > 编译选docker-compose、dockerd 和 docker一起编译就是没出来，暂时解决办法从luci-app-dockerman插件作者拉取源码替换，重新编译后正常了

算了 我直接改的docker的配置文件修改的参数 凑合用

lizheyo77, 2024-12-15:
> > 选上 dockerd 和 docker 就有了
> > 编译选docker-compose、dockerd 和 docker一起编译就是没出来，暂时解决办法从luci-app-dockerman插件作者拉取源码替换，重新编译后正常了
> > 链接供参考：https://github.com/lisaac/luci-app-dockerman

我发现一个新问题,我docker部分关闭后就无法再次开启了 好像也连接不上网络

xingshu1990, 2024-12-16:
> > > 选上 dockerd 和 docker 就有了
> > > 编译选docker-compose、dockerd 和 docker一起编译就是没出来，暂时解决办法从luci-app-dockerman插件作者拉取源码替换，重新编译后正常了
> > > 链接供参考：https://github.com/lisaac/luci-app-dockerman
> 
> 我发现一个新问题,我docker部分关闭后就无法再次开启了 好像也连接不上网络

有没有改动docker使用网络，另外有没有使用53端口之类的启用一些服务。

liwenjie119, 2024-12-21:
> 选上 dockerd 和 docker 就有了
选上了，第一次启动的时候还有完整显示，再起动就只有配置了，重置了也不行，很奇怪
[image]

xingshu1990, 2024-12-21:
> 选上 dockerd 和 docker 就有了
> 选上了，第一次启动的时候还有完整显示，再起动就只有配置了，重置了也不行，很奇怪 [...]

docker 和dockerman 两个选一个安装，至于会不会解决你的这个问题，暂时不知道。我这边在虚拟机里重新编译中

liwenjie119, 2024-12-21:
> > > 选上 dockerd 和 docker 就有了
> > > 选上了，第一次启动的时候还有完整显示，再起动就只有配置了，重置了也不行，很奇怪
> > > ![image](https://private-user-images.githubusercontent.com/25956275/397898545-54bc71bb-159b-40f1-922c-297538ba35a1.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzQ3NDcwODIsIm5iZiI6MTczNDc0Njc4MiwicGF0aCI6Ii8yNTk1NjI3NS8zOTc4OTg1NDUtNTRiYzcxYmItMTU5Yi00MGYxLTkyMmMtMjk3NTM4YmEzNWExLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEyMjElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMjIxVDAyMDYyMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTU0OTFmNzg4M2Y3MmM4MGYyMTYwNWZiODhkZDdiOGYzNjhjOWY0NjNhYjhkMzQ0YjVmYjgxMjZmMTBmODZmZWUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.UuuCubt3A1qBDz84ksCina665nesoU9kBZVYi5jU_6U)
> 
> docker 和dockerman 两个选一个安装， 如果留docker，你这里删除dockerman，dockerman-zh-cn。 至于会不会解决你的这个问题，暂时不知道。我这边在虚拟机里重新编译中。

我是添加了luci-app-dockerman，其他默认被勾选。看所有涉及docker的日志有如下，不知道有没有帮助：
```
Sat Dec 21 10:16:35 2024 user.notice dockerd-init: Bridge device 'docker0' already defined in network config
Sat Dec 21 10:16:35 2024 daemon.notice procd: /etc/rc.d/S99dockerd: uci: Entry not found
Sat Dec 21 10:16:35 2024 daemon.notice procd: /etc/rc.d/S99dockerd: uci: Entry not found
Sat Dec 21 10:16:35 2024 daemon.notice procd: /etc/rc.d/S99dockerd: uci: Entry not found
Sat Dec 21 10:16:35 2024 daemon.notice procd: /etc/rc.d/S99dockerd: uci: Entry not found
Sat Dec 21 10:16:35 2024 daemon.notice procd: /etc/rc.d/S99dockerd: uci: Entry not found
Sat Dec 21 10:16:35 2024 user.notice dockerd-init: Drop traffic from eth1 to docker0
Sat Dec 21 10:16:38 2024 daemon.err dockerd[7233]: time="2024-12-21T02:16:38.739638052Z" level=warning msg="failed to load plugin io.containerd.snapshotter.v1.devmapper" error="devmapper not configured"
Sat Dec 21 10:16:38 2024 daemon.err dockerd[7233]: time="2024-12-21T02:16:38.915865986Z" level=warning msg="could not use snapshotter devmapper in metadata plugin" error="devmapper not configured"
Sat Dec 21 10:16:41 2024 daemon.err dockerd[7233]: time="2024-12-21T02:16:41.332613935Z" level=error msg="failed to mount overlay: invalid argument" storage-driver=overlay2
Sat Dec 21 10:16:41 2024 daemon.err dockerd[7233]: time="2024-12-21T02:16:41.334677837Z" level=error msg="exec: \"fuse-overlayfs\": executable file not found in $PATH" storage-driver=fuse-overlayfs
Sat Dec 21 10:16:41 2024 kern.err kernel: [   21.429336] overlay: filesystem on /opt/docker/check-overlayfs-support3410440910/upper not supported as upperdir
Sat Dec 21 10:16:43 2024 daemon.err dockerd[7233]: time="2024-12-21T02:16:43.275304235Z" level=warning msg="Could not load necessary modules for IPSEC rules: protocol not supported"
Sat Dec 21 10:16:43 2024 daemon.info avahi-daemon[4158]: Joining mDNS multicast group on interface docker0.IPv4 with address 172.17.0.1.
Sat Dec 21 10:16:43 2024 daemon.info avahi-daemon[4158]: New relevant interface docker0.IPv4 for mDNS.
Sat Dec 21 10:16:43 2024 daemon.info avahi-daemon[4158]: Registering new address record for 172.17.0.1 on docker0.IPv4.
Sat Dec 21 10:16:44 2024 daemon.err dockerd[7233]: time="2024-12-21T02:16:44.275425475Z" level=warning msg="WARNING: No swap limit support"
Sat Dec 21 10:16:44 2024 daemon.err dockerd[7233]: time="2024-12-21T02:16:44.275461784Z" level=warning msg="WARNING: No cpu cfs quota support"
Sat Dec 21 10:16:44 2024 daemon.err dockerd[7233]: time="2024-12-21T02:16:44.275469680Z" level=warning msg="WARNING: No cpu cfs period support"
Sat Dec 21 10:16:44 2024 daemon.err dockerd[7233]: time="2024-12-21T02:16:44.275476165Z" level=warning msg="WARNING: No cpu shares support"
Sat Dec 21 10:16:44 2024 daemon.err dockerd[7233]: time="2024-12-21T02:16:44.275483444Z" level=warning msg="WARNING: No io.weight support"
Sat Dec 21 10:16:44 2024 daemon.err dockerd[7233]: time="2024-12-21T02:16:44.275488978Z" level=warning msg="WARNING: No io.weight (per device) support"
Sat Dec 21 10:16:44 2024 daemon.err dockerd[7233]: time="2024-12-21T02:16:44.275497871Z" level=warning msg="WARNING: No io.max (rbps) support"
Sat Dec 21 10:16:44 2024 daemon.err dockerd[7233]: time="2024-12-21T02:16:44.275503302Z" level=warning msg="WARNING: No io.max (wbps) support"
Sat Dec 21 10:16:44 2024 daemon.err dockerd[7233]: time="2024-12-21T02:16:44.275508902Z" level=warning msg="WARNING: No io.max (riops) support"
Sat Dec 21 10:16:44 2024 daemon.err dockerd[7233]: time="2024-12-21T02:16:44.275514330Z" level=warning msg="WARNING: No io.max (wiops) support"
```

liwenjie119, 2024-12-21:
突然发现另一个问题，选上dockerman后，adg，ssr和docker互相作用，还把ipv6搞废了，不知道算不算个例

gaoyaxuan, 2025-02-23:
我发现用istore 安装或卸载一个应用后,docker man 可以把缺失的页面显示出来,但是重启之后又会仅显示 配置 页面,(系统自带的opkg 安装卸载试过,无效)

coolsnowwolf, 2026-09-05:
已用当前 LuCI `openwrt-25.12` 分支复核并完成 x86_64 实际编译：runc、containerd、dockerd、docker-compose 及 `luci-app-dockerman` 全部通过，生成 `luci-app-dockerman_26.167.29608~ed082a1_all.ipk`（61774 bytes）。

2024 年旧版 Lua controller 会在生成菜单时检查 Docker socket 和 API `_ping()`；只要重启过程中 dockerd 尚未就绪就直接 `return`，因此恰好只显示先注册的“配置”页。现行版本已重写为 JS + ucode socket API，Overview、Containers、Images、Networks、Volumes 等菜单均静态登记，不再因 dockerd 启动时序删除菜单；依赖也会显式安装 docker、dockerd 等组件。原故障路径已经消失，按已解决关闭。

closed 2026-09-05 by coolsnowwolf