coolsnowwolf/lede #13205: bind-9.20.0 包编译有问题，

yulinsoft, 2024-12-12:
### 详细叙述

bind-9.20.0编译有问题，N1下云编译失败，同样的配置文件，就是在升级了这个包后就开始失败。

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

N1

### 详细日志

2024-12-11T10:55:35.7362939Z make[7]: Nothing to be done for 'all-am'.
2024-12-11T10:55:35.7364155Z make[7]: Leaving directory '/builder/openwrt/build_dir/target-aarch64_generic_musl/bind-9.20.0/fuzz'
2024-12-11T10:55:35.7365219Z make[6]: Leaving directory '/builder/openwrt/build_dir/target-aarch64_generic_musl/bind-9.20.0/fuzz'
2024-12-11T10:55:35.7367075Z Making all in bin
2024-12-11T10:55:35.7385512Z make[6]: Entering directory '/builder/openwrt/build_dir/target-aarch64_generic_musl/bind-9.20.0/bin'
2024-12-11T10:55:35.7416227Z Making all in named
2024-12-11T10:55:35.7468249Z make[7]: Entering directory '/builder/openwrt/build_dir/target-aarch64_generic_musl/bind-9.20.0/bin/named'
2024-12-11T10:55:35.7468773Z make  all-am
2024-12-11T10:55:35.7634027Z make[8]: Entering directory '/builder/openwrt/build_dir/target-aarch64_generic_musl/bind-9.20.0/bin/named'
2024-12-11T10:55:35.7651849Z depbase=`echo os.o | sed 's|[^/]*$|.deps/&|;s|\.o$||'`;\
2024-12-11T10:55:35.7658908Z aarch64-openwrt-linux-musl-gcc -DHAVE_CONFIG_H -I. -I../..  -U_FORTIFY_SOURCE -D_FORTIFY_SOURCE=2 -include ../../config.h -I./include -I../../include -I../../include -I../../lib/isc/include -I../../lib/isc/include -I../../lib/dns/include -I../../lib/dns/include -I../../lib/ns/include -I../../lib/isccc/include/ -I../../lib/isccfg/include -I/builder/openwrt/staging_dir/target-aarch64_generic_musl/usr/include -I/builder/openwrt/staging_dir/target-aarch64_generic_musl/usr/include    -I/builder/openwrt/staging_dir/target-aarch64_generic_musl/usr/include  -I/builder/openwrt/staging_dir/target-aarch64_generic_musl/usr/include  -I/builder/openwrt/staging_dir/target-aarch64_generic_musl/usr/include  -DNAMED_LOCALSTATEDIR=\"/var\" -DNAMED_SYSCONFDIR=\"/etc/bind\"  -I/builder/openwrt/staging_dir/toolchain-aarch64_generic_gcc-13.3.0_musl/usr/include -I/builder/openwrt/staging_dir/toolchain-aarch64_generic_gcc-13.3.0_musl/include -I/builder/openwrt/staging_dir/toolchain-aarch64_generic_gcc-13.3.0_musl/include/fortify    -Wall -Wextra -Wwrite-strings -Wpointer-arith -Wno-missing-field-initializers -Wformat -Wshadow -Werror=implicit-function-declaration -Werror=missing-prototypes -Werror=format-security -Werror=parentheses -Werror=implicit -Werror=strict-prototypes -Werror=vla -fno-strict-aliasing -fno-delete-null-pointer-checks -fdiagnostics-show-option -Os -pipe -mcpu=generic -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/builder/openwrt/build_dir/target-aarch64_generic_musl/bind-9.20.0=bind-9.20.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro    -I/builder/openwrt/staging_dir/target-aarch64_generic_musl/usr/include -pthread -MT os.o -MD -MP -MF $depbase.Tpo -c -o os.o os.c &&\
2024-12-11T10:55:35.7665263Z mv -f $depbase.Tpo $depbase.Po
2024-12-11T10:55:35.7769125Z <command-line>: warning: "_FORTIFY_SOURCE" redefined
2024-12-11T10:55:35.7769665Z <command-line>: note: this is the location of the previous definition
2024-12-11T10:55:35.8274529Z os.c:294:1: error: static declaration of 'setresuid' follows non-static declaration
2024-12-11T10:55:35.8275082Z   294 | setresuid(uid_t ruid, uid_t euid, uid_t suid) {
2024-12-11T10:55:35.8275424Z       | ^~~~~~~~~
2024-12-11T10:55:35.8275988Z In file included from /builder/openwrt/staging_dir/toolchain-aarch64_generic_gcc-13.3.0_musl/include/fortify/unistd.h:22,
2024-12-11T10:55:35.8276612Z                  from os.c:36:
2024-12-11T10:55:35.8277552Z /builder/openwrt/staging_dir/toolchain-aarch64_generic_gcc-13.3.0_musl/include/unistd.h:189:5: note: previous declaration of 'setresuid' with type 'int(uid_t,  uid_t,  uid_t)' {aka 'int(unsigned int,  unsigned int,  unsigned int)'}
2024-12-11T10:55:35.8278540Z   189 | int setresuid(uid_t, uid_t, uid_t);
2024-12-11T10:55:35.8278839Z       |     ^~~~~~~~~
2024-12-11T10:55:35.8357373Z make[8]: *** [Makefile:670: os.o] Error 1
2024-12-11T10:55:35.8358116Z make[8]: Leaving directory '/builder/openwrt/build_dir/target-aarch64_generic_musl/bind-9.20.0/bin/named'
2024-12-11T10:55:35.8359923Z make[7]: *** [Makefile:550: all] Error 2
2024-12-11T10:55:35.8360883Z make[7]: Leaving directory '/builder/openwrt/build_dir/target-aarch64_generic_musl/bind-9.20.0/bin/named'
2024-12-11T10:55:35.8366806Z make[6]: *** [Makefile:455: all-recursive] Error 1
2024-12-11T10:55:35.8367671Z make[6]: Leaving directory '/builder/openwrt/build_dir/target-aarch64_generic_musl/bind-9.20.0/bin'
2024-12-11T10:55:35.8372516Z make[5]: *** [Makefile:622: all-recursive] Error 1
2024-12-11T10:55:35.8373318Z make[5]: Leaving directory '/builder/openwrt/build_dir/target-aarch64_generic_musl/bind-9.20.0'
2024-12-11T10:55:35.8374794Z make[4]: *** [Makefile:529: all] Error 2
2024-12-11T10:55:35.8375541Z make[4]: Leaving directory '/builder/openwrt/build_dir/target-aarch64_generic_musl/bind-9.20.0'
2024-12-11T10:55:35.8377028Z make[3]: *** [Makefile:296: /builder/openwrt/build_dir/target-aarch64_generic_musl/bind-9.20.0/.built] Error 2
2024-12-11T10:55:35.8378056Z make[3]: Leaving directory '/builder/openwrt/feeds/packages/net/bind'
2024-12-11T10:55:35.8382913Z time: package/feeds/packages/bind/compile#0.45#0.14#0.57
2024-12-11T10:55:35.8389935Z     ERROR: package/feeds/packages/bind failed to build.
2024-12-11T10:55:35.8393369Z make[2]: *** [package/Makefile:116: package/feeds/packages/bind/compile] Error 1
2024-12-11T10:55:35.8398800Z make[2]: Leaving directory '/builder/openwrt'
2024-12-11T10:55:35.8430065Z make[1]: *** [package/Makefile:110: /builder/openwrt/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
2024-12-11T10:55:35.8433121Z make[1]: Leaving directory '/builder/openwrt'
2024-12-11T10:55:35.8460832Z make: *** [/builder/openwrt/include/toplevel.mk:231: world] Error 2
2024-12-11T10:55:35.8478127Z ##[error]Process completed with exit code 2.

loydcheung, 2024-12-12:
我也有同样的问题

ostrichii, 2024-12-12:
有同样的问题

mahakalaya, 2024-12-12:
是的，+1

coolsnowwolf, 2024-12-12:
修复了

closed 2024-12-13 by coolsnowwolf