coolsnowwolf/lede #13208: 小米路由器R3P，刷机后，重启无法运行。

likaiqiang0721, 2024-12-13:
### 详细叙述

[com7_2024-12-13_18_38_15.log](https://github.com/user-attachments/files/18125688/com7_2024-12-13_18_38_15.log)

CONFIG_TARGET_ramips=y
CONFIG_TARGET_ramips_mt7621=y
CONFIG_TARGET_ramips_mt7621_DEVICE_xiaomi_mi-router-3-pro=y

**可以正常工作的版本**
主机型号	Xiaomi Mi Router 3 Pro
架构	MediaTek MT7621 ver:1 eco:3 x 4
固件版本	OpenWrt R23.11.11 / LuCI Master (git-23.305.37706-43242b7)
内核版本	5.4.260

**新编译的固件，重启后无法运行（附日志）**
型号	Xiaomi Mi Router 3 Pro
架构	MediaTek MT7621 ver:1 eco:3
目标平台	ramips/mt7621
固件版本	LEDE R24.12.12 / LuCI openwrt-23.05 branch git-24.339.48605-e1135e2
内核版本	5.10.221

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

Xiaomi Mi Router 3 Pro

### 详细日志

附件

likaiqiang0721, 2024-12-14:
我测试了revert f0ab317，还是重启就启动不了。

4thanks, 2024-12-14:
额。。。不是吧，刚刚编译完，正准备刷上呢，要不试试我这个 https://github.com/4thanks/ctc-openwrt/releases/tag/mt7621_lede.git-master_24.12.14_10.05.31

coolsnowwolf, 2026-09-05:
已核对附件中的完整串口日志。该日志并不是内核没有启动：Linux 5.10.221 已成功 attach UBI、挂载 SquashFS 根文件系统、启动 procd，随后网卡及 Wi-Fi 驱动也完成加载并进入 BusyBox 控制台。实际失败点是旧的可写层损坏：`UBIFS error (ubi0:1): ubifs_recover_master_node: failed to recover master node`，因此 `/overlay` 无法挂载并退回只读根文件系统。

当前 R3P 仍使用标准的 `nand_do_upgrade`，源码中的 kernel/UBI 分区布局与日志一致；报告者也已确认同一时期另一份固件能够正常运行，因此没有可复现的设备源码回归。重新刷写时请不要保留旧配置（LuCI 取消“保留配置”，或命令行使用 `sysupgrade -n <固件>`），以重建 overlay。若仍能在当前代码及全新 overlay 上复现，请另开 issue 并附上从上电开始的完整串口日志和所用固件配置。

closed 2026-09-05 by coolsnowwolf