coolsnowwolf/lede #13210: ksmbd编译错误，可能与12月6日修改的代码有关

dodogao, 2024-12-14:
### 详细叙述

ksmbd编译错误，可能与12月6日修改的代码有关

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

LINKSYS WRT1900ACS V2

### 详细日志

cc1: some warnings being treated as errors
make[5]: *** [scripts/Makefile.build:286: /home/gaoh/lede/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/ksmbd-2024.10.14/vfs.o] Error 1
make[4]: *** [Makefile:1832: /home/gaoh/lede/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/ksmbd-2024.10.14] Error 2
make[4]: Leaving directory '/home/gaoh/lede/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/linux-5.10.221'
make[3]: *** [Makefile:73: /home/gaoh/lede/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/ksmbd-2024.10.14/.built] Error 2
make[3]: Leaving directory '/home/gaoh/lede/package/kernel/ksmbd'
time: package/kernel/ksmbd/compile#1.06#0.41#1.46
    ERROR: package/kernel/ksmbd failed to build.
make[2]: *** [package/Makefile:116: package/kernel/ksmbd/compile] Error 1
make[2]: Leaving directory '/home/gaoh/lede'
make[1]: *** [package/Makefile:110: /home/gaoh/lede/staging_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/gaoh/lede'
make: *** [/home/gaoh/lede/include/toplevel.mk:231: world] Error 2

flyhong666, 2024-12-16:
一样的问题。。。

flyhong666, 2024-12-19:
> 已修复
雕，你修的是5.10的，在6.6下还没修哦