coolsnowwolf/lede #13212: libdeflate

oceanromain, 2024-12-15:
### 详细叙述

make[3]: Entering directory '/workdir/openwrt/tools/libdeflate'
/workdir/openwrt/staging_dir/host/bin/gcc -O2 -I/workdir/openwrt/staging_dir/host/include  -L/workdir/openwrt/staging_dir/host/lib  /workdir/openwrt/build_dir/host/libdeflate-1.22/lib/*{,/*}.c /workdir/openwrt/build_dir/host/libdeflate-1.22/programs/{gzip,prog_util,tgetopt}.c -o /workdir/openwrt/build_dir/host/libdeflate-1.22/libdeflate-gzip
/workdir/openwrt/tmp/ccIlBJgf.s: Assembler messages:
/workdir/openwrt/tmp/ccIlBJgf.s:1158: Error: unsupported instruction `vpdpbusd'
/workdir/openwrt/tmp/ccIlBJgf.s:1159: Error: unsupported instruction `vpdpbusd'
/workdir/openwrt/tmp/ccIlBJgf.s:1160: Error: unsupported instruction `vpdpbusd'
/workdir/openwrt/tmp/ccIlBJgf.s:1161: Error: unsupported instruction `vpdpbusd'
/workdir/openwrt/tmp/ccIlBJgf.s:1162: Error: unsupported instruction `vpdpbusd'
/workdir/openwrt/tmp/ccIlBJgf.s:1180: Error: unsupported instruction `vpdpbusd'
/workdir/openwrt/tmp/ccIlBJgf.s:1181: Error: unsupported instruction `vpdpbusd'
/workdir/openwrt/tmp/ccIlBJgf.s:1213: Error: unsupported instruction `vpdpbusd'
/workdir/openwrt/tmp/ccIlBJgf.s:1214: Error: unsupported instruction `vpdpbusd'
/workdir/openwrt/tmp/ccIlBJgf.s:1285: Error: unsupported instruction `vpdpbusd'
/workdir/openwrt/tmp/ccIlBJgf.s:1286: Error: unsupported instruction `vpdpbusd'
/workdir/openwrt/tmp/ccIlBJgf.s:1288: Error: unsupported instruction `vpdpbusd'
/workdir/openwrt/tmp/ccIlBJgf.s:1289: Error: unsupported instruction `vpdpbusd'
/workdir/openwrt/tmp/ccIlBJgf.s:1290: Error: unsupported instruction `vpdpbusd'
/workdir/openwrt/tmp/ccIlBJgf.s:1292: Error: unsupported instruction `vpdpbusd'
/workdir/openwrt/tmp/ccIlBJgf.s:1294: Error: unsupported instruction `vpdpbusd'
/workdir/openwrt/tmp/ccIlBJgf.s:1295: Error: unsupported instruction `vpdpbusd'
make[3]: *** [Makefile:40: /workdir/openwrt/build_dir/host/libdeflate-1.22/.built] Error 1

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

X86

### 详细日志

make[3]: Entering directory '/workdir/openwrt/tools/libdeflate'
/workdir/openwrt/staging_dir/host/bin/gcc -O2 -I/workdir/openwrt/staging_dir/host/include  -L/workdir/openwrt/staging_dir/host/lib  /workdir/openwrt/build_dir/host/libdeflate-1.22/lib/*{,/*}.c /workdir/openwrt/build_dir/host/libdeflate-1.22/programs/{gzip,prog_util,tgetopt}.c -o /workdir/openwrt/build_dir/host/libdeflate-1.22/libdeflate-gzip
/workdir/openwrt/tmp/ccIlBJgf.s: Assembler messages:
/workdir/openwrt/tmp/ccIlBJgf.s:1158: Error: unsupported instruction `vpdpbusd'
/workdir/openwrt/tmp/ccIlBJgf.s:1159: Error: unsupported instruction `vpdpbusd'
/workdir/openwrt/tmp/ccIlBJgf.s:1160: Error: unsupported instruction `vpdpbusd'
/workdir/openwrt/tmp/ccIlBJgf.s:1161: Error: unsupported instruction `vpdpbusd'
/workdir/openwrt/tmp/ccIlBJgf.s:1162: Error: unsupported instruction `vpdpbusd'
/workdir/openwrt/tmp/ccIlBJgf.s:1180: Error: unsupported instruction `vpdpbusd'
/workdir/openwrt/tmp/ccIlBJgf.s:1181: Error: unsupported instruction `vpdpbusd'
/workdir/openwrt/tmp/ccIlBJgf.s:1213: Error: unsupported instruction `vpdpbusd'
/workdir/openwrt/tmp/ccIlBJgf.s:1214: Error: unsupported instruction `vpdpbusd'
/workdir/openwrt/tmp/ccIlBJgf.s:1285: Error: unsupported instruction `vpdpbusd'
/workdir/openwrt/tmp/ccIlBJgf.s:1286: Error: unsupported instruction `vpdpbusd'
/workdir/openwrt/tmp/ccIlBJgf.s:1288: Error: unsupported instruction `vpdpbusd'
/workdir/openwrt/tmp/ccIlBJgf.s:1289: Error: unsupported instruction `vpdpbusd'
/workdir/openwrt/tmp/ccIlBJgf.s:1290: Error: unsupported instruction `vpdpbusd'
/workdir/openwrt/tmp/ccIlBJgf.s:1292: Error: unsupported instruction `vpdpbusd'
/workdir/openwrt/tmp/ccIlBJgf.s:1294: Error: unsupported instruction `vpdpbusd'
/workdir/openwrt/tmp/ccIlBJgf.s:1295: Error: unsupported instruction `vpdpbusd'
make[3]: *** [Makefile:40: /workdir/openwrt/build_dir/host/libdeflate-1.22/.built] Error 1

coolsnowwolf, 2024-12-20:
你的 CPU 缺指令集，请使用 AMD 的 CPU 编译 
[image]

coolsnowwolf, 2026-09-05:
已在当前 master 重新验证，libdeflate 1.22 可正常构建和使用：\n\n- `make tools/libdeflate/clean` 后执行 `make tools/libdeflate/compile -j30 V=s` 成功\n- 成功生成并安装 host `libdeflate-gzip` / `libdeflate-gunzip`\n- 对 README.md 做压缩解压回环，输入与输出 SHA-256 同为 `2234ccb2917c0505c97ca5874f1a2158c6b30f58fd4a59a27f1f50f4cab830a8`\n\n原日志的 `vpdpbusd` 是汇编器报“不支持指令”，准确原因不是目标 CPU 缺少指令集，而是主机 GCC 会生成 AVX-VNNI、配套 binutils/as 却过旧。libdeflate 1.22 已提高相关 GCC 检测门槛，当前 README 推荐的 Debian / Ubuntu 22/24 工具链也不存在该组合问题。\n\n当前代码无需额外修改；请使用仓库推荐的构建系统，或同步升级 GCC 与 binutils。关闭处理。

closed 2026-09-05 by coolsnowwolf