coolsnowwolf/lede #13213: 编译错误

Krohnya-chan, 2024-12-15:
### 详细叙述

始终报错，什么插件都不加也报错

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

小米路由器R3G

### 详细日志

make[3]: Entering directory '/home/krohnya/lede/tools/meson'
. /home/krohnya/lede/include/shell.sh; /home/krohnya/lede/staging_dir/host/bin/libdeflate-gzip -dc /home/krohnya/lede/dl/meson-1.5.1.tar.gz | tar -C /home/krohnya/lede/build_dir/host/meson-1.5.1/.. -xf -
[ ! -d ./src/ ] || cp -fpR ./src/* /home/krohnya/lede/build_dir/host/meson-1.5.1
touch /home/krohnya/lede/build_dir/host/meson-1.5.1/.prepared1d032fcf37def14542604e0066b71d22_6664517399ebbbc92a37c5bb081b5c53
touch /home/krohnya/lede/build_dir/host/meson-1.5.1/.configured
touch /home/krohnya/lede/build_dir/host/meson-1.5.1/.built
install -d -m0755 /home/krohnya/lede/staging_dir/host/bin
/home/krohnya/lede/build_dir/host/meson-1.5.1/packaging/create_zipapp.py /home/krohnya/lede/build_dir/host/meson-1.5.1 --outfile /home/krohnya/lede/staging_dir/host/bin/meson.py
Traceback (most recent call last):
  File "/home/krohnya/lede/build_dir/host/meson-1.5.1/packaging/create_zipapp.py", line 23, in <module>
    zipapp.create_archive(d, interpreter=options.interpreter, target=options.outfile, compressed=options.compress)
TypeError: create_archive() got an unexpected keyword argument 'compressed'
Makefile:35: recipe for target '/home/krohnya/lede/staging_dir/host/stamp/.meson_installed' failed
make[3]: *** [/home/krohnya/lede/staging_dir/host/stamp/.meson_installed] Error 1
make[3]: Leaving directory '/home/krohnya/lede/tools/meson'
time: tools/meson/compile#0.22#0.15#0.34
    ERROR: tools/meson failed to build.
tools/Makefile:229: recipe for target 'tools/meson/compile' failed
make[2]: *** [tools/meson/compile] Error 1
make[2]: Leaving directory '/home/krohnya/lede'
tools/Makefile:227: recipe for target '/home/krohnya/lede/staging_dir/host/stamp/.tools_compile_nyyynyynnnyyynynyyyyyyynyyynynyynnynnyyyyyynyyyyyyyyyyynynnyyyyyyyyy' failed
make[1]: *** [/home/krohnya/lede/staging_dir/host/stamp/.tools_compile_nyyynyynnnyyynynyyyyyyynyyynynyynnynnyyyyyynyyyyyyyyyyynynnyyyyyyyyy] Error 2
make[1]: Leaving directory '/home/krohnya/lede'
/home/krohnya/lede/include/toplevel.mk:229: recipe for target 'world' failed
make: *** [world] Error 2

yanxin152133, 2024-12-17:
有个跟你类似的问题参考下
https://github.com/coolsnowwolf/lede/issues/7175

coolsnowwolf, 2024-12-20:
python  的 版本切换到 3.8 以上

xtkong, 2025-01-13:
我的也是这个报错，你解决了吗？

wushenghn, 2025-01-16:
编译环境ubuntu安装python3.8  把/usr/bin/python3.8 链接到 源码目录的staging_dir/host/bin/python3

closed 2025-12-28 by aiamadeus