coolsnowwolf/lede #13217: 偶发运营商强制pppoe下线后端口也断开还不恢复

Suwipe, 2024-12-17:
### 详细叙述

x86软路由
每2天运营商会在半夜强制pppoe下线重播，今天5点断线后端口出现断开而没有自动恢复，以前都是就算出现端口断开也会自动连接上然后开始拨号，查看网线端口灯正常，光猫那边网线灯也亮的，手动拔掉网线插上网口灯有变化能亮，但是系统里没变化还是一直未连接，日志也没有提示，最后手动把wan连接重启就恢复了。
一直都觉得pppoe下线的时候网线端口也会断开过一会再重新连接很奇怪，这次这问题，让我是不是就我有这情况网口跟着pppoe一起下线
下面是系统日志

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86

### 详细日志

Tue Dec 17 05:46:06 2024 daemon.info pppd[28808]: LCP terminated by peer
Tue Dec 17 05:46:06 2024 daemon.info pppd[28808]: Connect time 2826.0 minutes.
Tue Dec 17 05:46:06 2024 daemon.info pppd[28808]: Sent 1614010079 bytes, received 1733083980 bytes.
Tue Dec 17 05:46:06 2024 daemon.notice ttyd[8582]: [2024/12/17 05:46:06:8621] N: rops_handle_POLLIN_netlink: DELADDR
Tue Dec 17 05:46:06 2024 daemon.err miniupnpd[22077]: ioctl(s, SIOCGIFADDR, ...): Address not available
Tue Dec 17 05:46:06 2024 daemon.err miniupnpd[22077]: Failed to get IP for interface pppoe-wan
Tue Dec 17 05:46:06 2024 daemon.warn miniupnpd[22077]: SendNATPMPPublicAddressChangeNotification: cannot get public IP address, stopping
Tue Dec 17 05:46:06 2024 daemon.notice ttyd[8582]: [2024/12/17 05:46:06:8626] N: rops_handle_POLLIN_netlink: DELADDR
Tue Dec 17 05:46:06 2024 daemon.notice ttyd[8582]: [2024/12/17 05:46:06:8628] N: rops_handle_POLLIN_netlink: DELADDR
Tue Dec 17 05:46:06 2024 daemon.notice netifd: Network device 'pppoe-wan' link is down
Tue Dec 17 05:46:06 2024 daemon.err odhcp6c[29000]: Failed to send RELEASE message to ff02::1:2 (Permission denied)
Tue Dec 17 05:46:06 2024 daemon.notice netifd: Network alias 'pppoe-wan' link is down
Tue Dec 17 05:46:06 2024 daemon.notice netifd: Interface 'wan_6' has link connectivity loss
Tue Dec 17 05:46:06 2024 daemon.notice pppd[28808]: Modem hangup
Tue Dec 17 05:46:06 2024 daemon.notice pppd[28808]: Connection terminated.
Tue Dec 17 05:46:06 2024 daemon.info pppd[28808]: Connect time 2826.0 minutes.
Tue Dec 17 05:46:06 2024 daemon.info pppd[28808]: Sent 1614010079 bytes, received 1733083980 bytes.
Tue Dec 17 05:46:06 2024 daemon.notice netifd: Interface 'wan' has lost the connection
Tue Dec 17 05:46:06 2024 daemon.notice netifd: wan_6 (29000): Command failed: ubus call network.interface notify_proto { "action": 0, "link-up": false, "keep": false, "interface": "wan_6" } (Permission denied)
Tue Dec 17 05:46:06 2024 daemon.info pppd[28808]: Exit.
Tue Dec 17 05:46:06 2024 daemon.notice netifd: Interface 'wan' is now down
Tue Dec 17 05:46:07 2024 daemon.notice ttyd[8582]: [2024/12/17 05:46:07:0225] N: rops_handle_POLLIN_netlink: DELADDR
Tue Dec 17 05:46:07 2024 daemon.notice netifd: Interface 'wan' is disabled
Tue Dec 17 05:46:07 2024 daemon.notice netifd: Interface 'wan' is enabled
Tue Dec 17 05:46:07 2024 daemon.notice netifd: Interface 'wan' is setting up now
Tue Dec 17 05:46:07 2024 daemon.notice netifd: Network device 'eth4' link is down
Tue Dec 17 05:46:07 2024 daemon.notice netifd: Interface 'wan' has link connectivity loss
Tue Dec 17 05:46:07 2024 daemon.notice netifd: Interface 'wan' is now down
Tue Dec 17 05:46:07 2024 daemon.notice netifd: Interface 'wan' is disabled
Tue Dec 17 05:46:07 2024 daemon.notice netifd: Interface 'wan' is enabled
Tue Dec 17 05:46:07 2024 daemon.err odhcp6c[29000]: Failed to send SOLICIT message to ff02::1:2 (Permission denied)
Tue Dec 17 05:46:07 2024 daemon.notice netifd: Interface 'wan_6' is now down
Tue Dec 17 05:46:07 2024 daemon.notice netifd: Interface 'wan_6' is disabled
Tue Dec 17 05:46:07 2024 kern.info kernel: [924699.167309] netifd[18350]: segfault at 7ff7f681a868 ip 000000000040df7d sp 00007fffb47c5d80 error 4 in netifd[404000+1b000] likely on CPU 3 (core 3, socket 0)
Tue Dec 17 05:46:07 2024 kern.info kernel: [924699.181647] Code: 00 00 01 75 2f 48 8b 53 58 31 c0 be 34 fd 41 00 bf 02 00 00 00 e8 6d 89 ff ff 48 89 df e8 f9 e7 ff ff 31 c0 89 83 90 00 00 00 <48> 8b 7b 68 5b e9 78 77 ff ff 5b c3 41 57 41 89 cf 41 56 49 89 d6
Tue Dec 17 05:46:11 2024 kern.info kernel: [924703.066677] igc 0000:05:00.0 eth4: NIC Link is Up 2500 Mbps Full Duplex, Flow Control: RX
Tue Dec 17 05:46:12 2024 user.notice : Added device handler type: 8021ad
Tue Dec 17 05:46:12 2024 user.notice : Added device handler type: 8021q
Tue Dec 17 05:46:12 2024 user.notice : Added device handler type: macvlan
Tue Dec 17 05:46:12 2024 user.notice : Added device handler type: veth
Tue Dec 17 05:46:12 2024 user.notice : Added device handler type: bridge
Tue Dec 17 05:46:12 2024 user.notice : Added device handler type: Network device
Tue Dec 17 05:46:12 2024 user.notice : Added device handler type: tunnel
Tue Dec 17 05:46:12 2024 kern.info kernel: [924704.276445] br-lan: port 2(eth1) entered disabled state
Tue Dec 17 05:46:12 2024 kern.info kernel: [924704.281834] br-lan: port 1(eth0) entered disabled state
Tue Dec 17 05:46:12 2024 daemon.notice ttyd[8582]: [2024/12/17 05:46:12:4586] N: rops_handle_POLLIN_netlink: DELADDR
Tue Dec 17 05:46:12 2024 daemon.notice ttyd[8582]: [2024/12/17 05:46:12:4587] N: rops_handle_POLLIN_netlink: DELADDR
Tue Dec 17 05:46:12 2024 kern.info kernel: [924704.287679] igc 0000:04:00.0 eth3: left allmulticast mode
Tue Dec 17 05:46:12 2024 kern.info kernel: [924704.293219] br-lan: port 4(eth3) entered disabled state
Tue Dec 17 05:46:12 2024 kern.info kernel: [924704.386517] igc 0000:03:00.0 eth2: left allmulticast mode
Tue Dec 17 05:46:12 2024 kern.info kernel: [924704.392058] br-lan: port 3(eth2) entered disabled state
Tue Dec 17 05:46:12 2024 kern.info kernel: [924704.466473] igc 0000:02:00.0 eth1: left allmulticast mode
Tue Dec 17 05:46:12 2024 kern.info kernel: [924704.472104] br-lan: port 2(eth1) entered disabled state
[... 169 lines cut ...]
Sun Dec 15 03:55:18 2024 daemon.notice miniupnpd[18910]: Listening for NAT-PMP/PCP traffic on port 5351
Sun Dec 15 03:55:18 2024 daemon.err mosdns[19533]: 2024-12-14T19:55:18.720Z	INFO	working directory changed	{"path": "/etc/mosdns"}
Sun Dec 15 03:55:18 2024 daemon.err mosdns[19533]: 2024-12-14T19:55:18.721Z	INFO	main config loaded	{"file": "/var/etc/mosdns.json"}
Sun Dec 15 03:55:18 2024 user.notice nlbwmon: Reloading nlbwmon due to ifup of loopback (lo)
Sun Dec 15 03:55:19 2024 daemon.err odhcpd[2365]: Failed to send to ff02::1%lan@br-lan (Address not available)
Sun Dec 15 03:55:19 2024 daemon.notice netifd: bridge 'br-lan' link is down
Sun Dec 15 03:55:19 2024 daemon.notice netifd: Interface 'lan' has link connectivity loss
Sun Dec 15 03:55:19 2024 daemon.err mosdns[20752]: 2024-12-14T19:55:19.311Z	INFO	working directory changed	{"path": "/etc/mosdns"}
Sun Dec 15 03:55:19 2024 daemon.err mosdns[20752]: 2024-12-14T19:55:19.311Z	INFO	main config loaded	{"file": "/var/etc/mosdns.json"}
Sun Dec 15 03:55:21 2024 daemon.notice netifd: Network device 'eth4' link is up
Sun Dec 15 03:55:21 2024 daemon.notice netifd: Interface 'wan' has link connectivity
Sun Dec 15 03:55:21 2024 daemon.notice netifd: Interface 'wan' is setting up now
Sun Dec 15 03:55:21 2024 kern.info kernel: [745255.576457] igc 0000:05:00.0 eth4: NIC Link is Up 2500 Mbps Full Duplex, Flow Control: RX
Sun Dec 15 03:55:21 2024 daemon.notice netifd: wan (20791): uci: Entry not found
Sun Dec 15 03:55:21 2024 daemon.notice netifd: wan (20791): sh: out of range
Sun Dec 15 03:55:21 2024 daemon.info pppd[20799]: Plugin pppoe.so loaded.
Sun Dec 15 03:55:21 2024 daemon.info pppd[20799]: PPPoE plugin from pppd 2.4.9
Sun Dec 15 03:55:21 2024 daemon.notice pppd[20799]: pppd 2.4.9 started by root, uid 0
Sun Dec 15 03:55:21 2024 kern.info kernel: [745255.766456] igc 0000:01:00.0 eth0: NIC Link is Up 2500 Mbps Full Duplex, Flow Control: RX
Sun Dec 15 03:55:21 2024 kern.info kernel: [745255.774805] br-lan: port 1(eth0) entered blocking state
Sun Dec 15 03:55:21 2024 daemon.notice netifd: Network device 'eth0' link is up
Sun Dec 15 03:55:21 2024 daemon.notice netifd: bridge 'br-lan' link is up
Sun Dec 15 03:55:21 2024 daemon.notice netifd: Interface 'lan' has link connectivity
Sun Dec 15 03:55:21 2024 kern.info kernel: [745255.780151] br-lan: port 1(eth0) entered forwarding state
Sun Dec 15 03:55:21 2024 daemon.info pppd[20799]: PPP session is 52429

t96mytell, 2024-12-17:
有个折中的办法，定时每天某个时间重启wan

yanxin152133, 2024-12-17:
不是有个插件，watchcat，设置成监测到断网后让它重拨或者重启
[image]

Suwipe, 2024-12-17:
> 不是有个插件，watchcat，设置成监测到断网后让它重拨或者重启 [...]

这个可以，之前我还遇到过更奇葩的重播获取ipv4正常但是ipv6获取失败导致连带影响ipv4访问外网都异常，逼着我写了个脚本检测IP获取情况然后重播，用这个能解决几种pppoe问题了

closed 2024-12-28 by Suwipe