coolsnowwolf/lede #13223: getting issue when compiling :g++: error: unrecognized command line option '-std=gnu++20'; did you mean '-std=gnu++2a'?

shelliao, 2024-12-20:
### 详细叙述

issue got when do compiling node-v22.11.0
g++: error: unrecognized command line option '-std=gnu++20'; did you mean '-std=gnu++2a'?

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

R2S

### 详细日志

WARNING: C++ compiler (CXX=g++, 9.4.0) too old, need g++ 10.1.0 or clang++ 8.0.0
INFO: Using floating patch "tools/icu/patches/75/source/tools/toolutil/pkg_genc.h" from "tools/icu"
INFO: Using floating patch "tools/icu/patches/75/source/tools/toolutil/pkg_genc.cpp" from "tools/icu"
INFO: Using floating patch "tools/icu/patches/75/source/tools/genccode/genccode.c" from "tools/icu"
WARNING: warnings were emitted in the configure phase
INFO: configure completed successfully
touch /home/shell/lede/build_dir/hostpkg/node-v22.11.0/.configured
CFLAGS="-O2 -I/home/shell/lede/staging_dir/host/include -I/home/shell/lede/staging_dir/hostpkg/include -I/home/shell/lede/staging_dir/target-aarch64_generic_musl/host/include" CPPFLAGS="-I/home/shell/lede/staging_dir/host/include -I/home/shell/lede/staging_dir/hostpkg/include -I/home/shell/lede/staging_dir/target-aarch64_generic_musl/host/include" CXXFLAGS="-O2 -I/home/shell/lede/staging_dir/host/include -I/home/shell/lede/staging_dir/hostpkg/include -I/home/shell/lede/staging_dir/target-aarch64_generic_musl/host/include" LDFLAGS="-L/home/shell/lede/staging_dir/host/lib -L/home/shell/lede/staging_dir/hostpkg/lib -L/home/shell/lede/staging_dir/target-aarch64_generic_musl/host/lib" NO_LOAD='cctest.target.mk embedtest.target.mk overlapped-checker.target.mk' make  -C /home/shell/lede/build_dir/hostpkg/node-v22.11.0  
make[4]: Entering directory '/home/shell/lede/build_dir/hostpkg/node-v22.11.0'
make -C out BUILDTYPE=Release V=s
  g++ -o /home/shell/lede/build_dir/hostpkg/node-v22.11.0/out/Release/obj.target/ada/deps/ada/ada.o ../deps/ada/ada.cpp '-D_GLIBCXX_USE_CXX11_ABI=1' '-DNODE_OPENSSL_CONF_NAME=nodejs_conf' '-DNODE_OPENSSL_HAS_QUIC' '-DICU_NO_USER_DATA_OVERRIDE' '-D__STDC_FORMAT_MACROS' '-DOPENSSL_NO_PINSHARED' '-DOPENSSL_THREADS' -I../deps/ada  -pthread -Wall -Wextra -Wno-unused-parameter -m64 -O3 -fno-omit-frame-pointer -fno-rtti -fno-exceptions -fno-strict-aliasing -std=gnu++20 -MMD -MF /home/shell/lede/build_dir/hostpkg/node-v22.11.0/out/Release/.deps//home/shell/lede/build_dir/hostpkg/node-v22.11.0/out/Release/obj.target/ada/deps/ada/ada.o.d.raw -I/home/shell/lede/staging_dir/host/include -I/home/shell/lede/staging_dir/hostpkg/include -I/home/shell/lede/staging_dir/target-aarch64_generic_musl/host/include -O2 -I/home/shell/lede/staging_dir/host/include -I/home/shell/lede/staging_dir/hostpkg/include -I/home/shell/lede/staging_dir/target-aarch64_generic_musl/host/include -c
g++: error: unrecognized command line option '-std=gnu++20'; did you mean '-std=gnu++2a'?
make[5]: *** [deps/ada/ada.target.mk:86: /home/shell/lede/build_dir/hostpkg/node-v22.11.0/out/Release/obj.target/ada/deps/ada/ada.o] Error 1
make[4]: *** [Makefile:137: node] Error 2
make[4]: Leaving directory '/home/shell/lede/build_dir/hostpkg/node-v22.11.0'
make[3]: *** [Makefile:244: /home/shell/lede/build_dir/hostpkg/node-v22.11.0/.built] Error 2
make[3]: Leaving directory '/home/shell/lede/feeds/packages/lang/node'
time: package/feeds/packages/node/host-compile#11.88#4.11#32.55
    ERROR: package/feeds/packages/node [host] failed to build.
make[2]: *** [package/Makefile:114: package/feeds/packages/node/host/compile] Error 1
make[2]: Leaving directory '/home/shell/lede'
make[1]: *** [package/Makefile:110: /home/shell/lede/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/shell/lede'
make: *** [/home/shell/lede/include/toplevel.mk:231: world] Error 2

coolsnowwolf, 2024-12-20:
主机的 gcc 要升级到 gcc11

shelliao, 2024-12-23:
是否能修改一下文档中安装依赖的内容，以便可以一键安装gcc 11

qingtian110, 2024-12-23:
虽然不知道 LEDE 源码能不能使用。但不妨试试： https://github.com/sbwml/feeds_packages_lang_node-prebuilt

10秒完成 node 的编译总比花 30 分钟的源码编译来的快

Hans-Deng, 2024-12-24:
> 虽然不知道 LEDE 源码能不能使用。但不妨试试： https://github.com/sbwml/feeds_packages_lang_node-prebuilt
> 
> 10秒完成 node 的编译总比花 30 分钟的源码编译来的快

实测有用 感谢帮助

jw10126121, 2025-01-06:
> > 虽然不知道 LEDE 源码能不能使用。但不妨试试： https://github.com/sbwml/feeds_packages_lang_node-prebuilt
> > 10秒完成 node 的编译总比花 30 分钟的源码编译来的快
> 
> 实测有用 感谢帮助

请问，要替换哪个版本的node呢，想说，lean支持的op版本是多少，就对应取哪个版本，但是op的版本，是多少，这个怎么获取到呢

coolsnowwolf, 2026-09-05:
已验证并修正说明：\n\n- 原日志使用 g++ 9.4，Node 22 的源码以 `-std=gnu++20` 编译，官方 BUILDING.md 与 configure 都要求 GCC >= 10.1（或 Clang >= 8.0），因此原失败属于主机编译器版本过旧\n- 当前 LEDE README 已推荐 Debian / Ubuntu LTS 22/24，不再推荐 Ubuntu 20.04；这些推荐环境的默认工具链满足要求\n- 在当前 g++ 13.3 上从 clean 状态执行 `make package/feeds/packages/node/host/compile -j30 V=s` 成功，最终 host Node 可执行文件运行输出 `v22.23.0`（V8 12.4.254.21-node.56）\n- packages 中原有帮助文本仍误写 gcc 8.3，已通过独立提交 1857925054d886dab3708f24b6fb4594045602f4 修正为 GCC 10.1 / Clang 8.0，并已推送、核对远端 master\n\n请使用当前 README 推荐系统或升级主机编译器。问题已确认处理，关闭。

closed 2026-09-05 by coolsnowwolf