coolsnowwolf/lede #13225: 编译最新6.12内核固件成功，但是无法启动

shjy10000, 2024-12-20:
### 详细叙述

[    7.524165] request_module: modprobe binfmt-4c46 cannot be processed, kmod busy with 50 threads for more than 5 se                                                                                                                                     conds now
[    7.536204] Starting init: /sbin/init exists but couldn't execute it (error -8)
[    7.538038] Run /etc/init as init process
[    7.545912] Run /bin/init as init process
[    7.552653] Run /bin/sh as init process
[   12.857505] request_module: modprobe binfmt-4c46 cannot be processed, kmod busy with 50 threads for more than 5 se                                                                                                                                     conds now
[   12.869460] Starting init: /bin/sh exists but couldn't execute it (error -8)
[   12.871178] Kernel panic - not syncing: No working init found.  Try passing init= option to kernel. See Linux Docu                                                                                                                                     mentation/admin-guide/init.rst for guidance.
[   12.872442] CPU: 0 UID: 0 PID: 1 Comm: swapper/0 Not tainted 6.12.6 #0
[   12.873045] Hardware name: Radxa E25 Carrier Board (DT)
[   12.873523] Call trace:
[   12.873754]  dump_backtrace+0xa0/0xe0
[   12.874127]  show_stack+0x18/0x24
[   12.874452]  dump_stack_lvl+0x34/0x8c
[   12.874806]  dump_stack+0x18/0x24
[   12.875126]  panic+0x350/0x364
[   12.875428]  kernel_init+0x194/0x1d0
[   12.875782]  ret_from_fork+0x10/0x20
[   12.876128] SMP: stopping secondary CPUs
[   12.876503] Kernel Offset: disabled
   12.876823] CPU features: 0x0c,00000004,00280908,4200700b
[   12.877317] Memory Limit: none
[   12.877608] ---[ end Kernel panic - not syncing: No working init found.  Try passing init= option to kernel. See L                                                                                                                                     inux Documentation/admin-guide/init.rst for guidance. ]---

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

radxa e25

### 详细日志

固件是可以编译成功，但是刷入后死活启动不了，报错日志如上

aiamadeus, 2024-12-21:
[image]
没问题啊

shjy10000, 2024-12-21:
> ![图像](https://private-user-images.githubusercontent.com/42570690/397908871-4cdad88d-d532-4651-86da-3e54d89e7d80.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzQ3ODExNDAsIm5iZiI6MTczNDc4MDg0MCwicGF0aCI6Ii80MjU3MDY5MC8zOTc5MDg4NzEtNGNkYWQ4OGQtZDUzMi00NjUxLTg2ZGEtM2U1NGQ4OWU3ZDgwLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEyMjElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMjIxVDExMzQwMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWM2ZGZlZjY2OWVkMjFjMTQyY2Q0NzE5MDZlYWMxZjNlZDgyNThiNDVmNjRlODlhYzQxZTkxM2NjZDA0YzAxMWYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.8tDSXQC2UUd6NU_pb5uqvQJLe5ZZKnAckR8SN0PNeY8)没问题啊

全部重新编译了一下，可以开机了，谢谢

shjy10000, 2024-12-21:
> 全部重新编译了一下，可以开机了，谢谢

大佬，可以问一下，这个e25设备刷了lede后，无法使用按钮进入loader模式了，之前使用F大的可以正常使用按钮进入loader模式，请问这个可以解决一下么，现在每次都要使用命令进入maskrom模式才能刷机，非常的麻烦，希望解决一下这个问题

closed 2024-12-23 by aiamadeus