coolsnowwolf/lede #13228: 2024-12-21更新代码后，编译失败

jesee, 2024-12-21:
### 详细叙述

今天更新代码重新编译就报错了
环境wsl ubuntu20.0
之后放到docker debian环境重头跑了一遍也是同样报错
查了下issue，没找到相关解决办法

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

Nokia EA0326GMP

### 详细日志

```
[32/301] /debian-home/lede/staging_dir/host/bin/g++ -O2 -I/debian-home/lede/staging_dir/host/include     -D_FILE_OFFSET_BITS=64    -DCMAKE_BOOTSTRAP    -DCMake_HAVE_CXX_MAKE_UNIQUE=1 -DCMake_HAVE_CXX_FILESYSTEM=1   -I/debian-home/lede/build_dir/host/cmake-3.30.5/Bootstrap.cmk   -I/debian-home/lede/build_dir/host/cmake-3.30.5/Source   -I/debian-home/lede/build_dir/host/cmake-3.30.5/Source/LexerParser   -I/debian-home/lede/build_dir/host/cmake-3.30.5/Utilities/std   -I/debian-home/lede/build_dir/host/cmake-3.30.5/Utilities  -c /debian-home/lede/build_dir/host/cmake-3.30.5/Source/cmList.cxx -o cmList.o
ninja: build stopped: subcommand failed.
---------------------------------------------
Error when bootstrapping CMake:
Problem while running /debian-home/lede/staging_dir/host/bin/ninja
---------------------------------------------
Log of errors: /debian-home/lede/build_dir/host/cmake-3.30.5/Bootstrap.cmk/cmake_bootstrap.log
---------------------------------------------
make[3]: *** [Makefile:59: /debian-home/lede/build_dir/host/cmake-3.30.5/.configured] Error 9
make[3]: Leaving directory '/debian-home/lede/tools/cmake'
time: tools/cmake/compile#104.39#126.94#72.03
    ERROR: tools/cmake failed to build.
make[2]: *** [tools/Makefile:231: tools/cmake/compile] Error 1
make[2]: Leaving directory '/debian-home/lede'
make[1]: *** [tools/Makefile:227: /debian-home/lede/staging_dir/host/stamp/.tools_compile_nyyynyynnnyyynynyyyyyyynyyynynyynnynnyyyyyynyyyyyyyyyyynynnyyyyynyyy] Error 2
make[1]: Leaving directory '/debian-home/lede'
make: *** [/debian-home/lede/include/toplevel.mk:231: world] Error 2
```

另外查出日志文件中的错误：
```
cat /debian-home/lede/build_dir/host/cmake-3.30.5/Bootstrap.cmk/cmake_bootstrap.log | grep error                                                                                                             
# error "The CMAKE_C_COMPILER is set to a C++ compiler"                                                                 
#error "On AIX with GNU we need the -pthread flag."                                                                     
#error "On Solaris we need C99."                                                                                        
#error "On HP-UX we need GCC 4.9 or higher."                                                                            
#error "Compiler is not in a mode aware of C++11."                                                                      
#error "On AIX with GNU we need the -pthread flag."                                                                     
#error "SunPro <= 5.13 mode not supported due to bug in move semantics."                                                
#error "On HP-UX we need GCC 4.9 or higher."                                                                            
  // the wrong constructor which generate error in template instantiation.                                              
#      error "utimensat not available on macOS < 10.13"                                                                 
#      error "utimensat not available on macOS < 10.13"                                                                 
#      error "utimensat not available on macOS < 10.13"                                                                 
/debian-home/lede/build_dir/host/cmake-3.30.5/Source/kwsys/kwsysPlatformTestsCXX.cxx:53:13: error: 'environ' was not declared in this scope; did you mean 'union'?                                                                              
#      error "utimensat not available on macOS < 10.13"                                                                 
#      error "utimensat not available on macOS < 10.13"
```

aiamadeus, 2024-12-21:
你这是不是编译机内存不够啊

jesee, 2024-12-21:
> 你这是不是编译机内存不够啊

[image]
不清楚呢，这个是我任务管理器截图情况

jesee, 2024-12-21:
> 你这是不是编译机内存不够啊

[image]
ssd 100%不知道是不是这个原因

jesee, 2024-12-21:
> 你这是不是编译机内存不够啊

[image]
不知道咋处理，我试试放别的电脑跑看看

jesee, 2024-12-21:
换了个电脑跑，每次跑都2个报错，跑一次一个小时，不想搞了。。。

jesee, 2024-12-21:
> 你这是不是编译机内存不够啊

我又检查了下，free -h发现wsl配置的内存太小了，将.wslconfig配置的内存改大一点重新跑了下，没报错了。

closed 2024-12-21 by jesee