coolsnowwolf/lede #13229: ax6s 主分支默认编译后lan接口插入无反应，且无5Gwifi

nebula888, 2024-12-21:
### 详细叙述

ax6s 主分支默认编译，仅设置型号和root分区大小，其他均保持不变，编译后刷入有线无法链接。pc查看发现没有ip地址。用2.4Gwifi LEDE进入路由器之后发现所有lan口均显示未连接，且获取不到cpu温度。而且没有5gwifi选项

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

ax6s

### 详细日志

编译日记未保存提供config
CONFIG_TARGET_mediatek=y
CONFIG_TARGET_mediatek_mt7622=y
CONFIG_TARGET_mediatek_mt7622_DEVICE_xiaomi_redmi-router-ax6s=y
CONFIG_PACKAGE_6in4=y
# CONFIG_PACKAGE_firewall is not set
CONFIG_PACKAGE_firewall4=y
# CONFIG_PACKAGE_iptables-mod-conntrack-extra is not set
# CONFIG_PACKAGE_iptables-mod-fullconenat is not set
CONFIG_PACKAGE_ipv6helper=y
CONFIG_PACKAGE_jansson=y
CONFIG_PACKAGE_kmod-crypto-crc32c=y
# CONFIG_PACKAGE_kmod-ipt-conntrack-extra is not set
# CONFIG_PACKAGE_kmod-ipt-fullconenat is not set
CONFIG_PACKAGE_kmod-iptunnel=y
CONFIG_PACKAGE_kmod-iptunnel4=y
CONFIG_PACKAGE_kmod-lib-crc32c=y
CONFIG_PACKAGE_kmod-nft-core=y
CONFIG_PACKAGE_kmod-nft-fib=y
CONFIG_PACKAGE_kmod-nft-nat=y
CONFIG_PACKAGE_kmod-nft-offload=y
CONFIG_PACKAGE_kmod-sit=y
CONFIG_PACKAGE_libnftnl=y
CONFIG_PACKAGE_nftables-json=y
CONFIG_TARGET_ROOTFS_PARTSIZE=50

Bincyx, 2025-01-22:
一樣

coolsnowwolf, 2026-09-05:
已参考 OpenWrt 上游修复并提交：e66721c4e321e1a71702bbe1afe10b322fa97b0f\n\n修复内容：\n- 将 AX6S 的 MT7531 MDIO 地址从错误的 0 改为硬件实际地址 31，修复新版 DSA 驱动按 DTS 地址探测后 LAN switch 无法初始化的问题\n- 将 MT7622 2.4 GHz 与 MT7915 5 GHz 的校准数据改为当前 nvmem EEPROM 描述，保证两个 radio 都能取得正确校准数据\n- 补齐 memory 节点的地址与 `device_type`，不再依赖 U-Boot 可选 fixup\n- 保留 LEDE 当前分区与启动布局，没有带入上游新的 UBI 布局\n\n验证：\n- AX6S / Linux 6.12.108：`make target/linux/compile -j30 V=s` 成功\n- 设备 DTS 成功生成 23877-byte DTB；反编译确认 `switch@1f reg = <0x1f>`、两块 EEPROM nvmem、2.4/5 GHz radio 引用和 `cpu-thermal` 节点均存在\n- mac80211 6.18.39 及当前 mt76、MT7615、MT7915 模块和固件编译成功\n- 提交已推送并确认远端 master 与本地一致\n\n据此关闭。

closed 2026-09-05 by coolsnowwolf