coolsnowwolf/lede #13230: ERROR: package/feeds/helloworld/naiveproxy failed to build

jzhou404, 2024-12-23:
### 详细叙述

make download -j8

make[2] -C feeds/packages/libs/liburing download
 make[2] -C feeds/packages/libs/libuv download
 make[2] -C feeds/packages/libs/libwebsockets download
 make[2] -C feeds/packages/utils/lm-sensors download
    ERROR: package/feeds/helloworld/naiveproxy failed to build.
 make[2] -C feeds/packages/utils/lvm2 download
make package/download: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
 make[1] target/download
 make[2] -C target/linux download

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x64编译

### 详细日志

make[2] -C feeds/packages/libs/keyutils download
 make[2] -C feeds/packages/utils/kmod download
 make[2] -C feeds/packages/net/ksmbd-tools download
 make[2] -C feeds/packages/libs/libdaemon download
 make[2] -C feeds/packages/libs/libev download
 make[2] -C feeds/packages/libs/libnatpmp download
 make[2] -C feeds/packages/libs/libsodium download
 make[2] -C feeds/packages/libs/libtasn1 download
 make[2] -C feeds/packages/libs/libtirpc download
 make[2] -C feeds/packages/libs/liburing download
 make[2] -C feeds/packages/libs/libuv download
 make[2] -C feeds/packages/libs/libwebsockets download
 make[2] -C feeds/packages/utils/lm-sensors download
    ERROR: package/feeds/helloworld/naiveproxy failed to build.
 make[2] -C feeds/packages/utils/lvm2 download
make package/download: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
 make[1] target/download
 make[2] -C target/linux download

coolveyron, 2024-12-24:
你的编译环境是啥？自己在家里的单机环境还是用的云服务器？

clion007, 2025-02-06:
用action ubuntu-22.04环境编译xray-core也是报错，我看了一下，是go的版本问题，要求1.23，当前是1.22。请问应该如何解决？

clion007, 2025-02-07:
> 用action ubuntu-22.04环境编译xray-core也是报错，我看了一下，是go的版本问题，要求1.23，当前是1.22。请问应该如何解决？

已解决，更新feeds中的golang版本就好了

coolsnowwolf, 2026-09-05:
已在当前代码重新验证，此下载/编译问题已经不存在：\n\n- 平台：x86_64（与 issue 一致）\n- naiveproxy：150.0.7871.63-1\n- `make package/feeds/helloworld/naiveproxy/download -j1 V=s` 成功\n- 源码、Chromium Clang 与 PGO profile 三项下载文件的 SHA-256 均与 Makefile 完全一致\n- 临时启用 `CONFIG_PACKAGE_naiveproxy=m` 后，从 clean 状态执行 `make package/feeds/helloworld/naiveproxy/compile -j30 V=s` 成功\n- Ninja 4942/4942 完成并成功链接 `naive`\n- 生成 `naiveproxy_150.0.7871.63.1-1_x86_64.ipk`（3847576 bytes）\n\n原报告只有并行下载的汇总尾部，没有实际错误；当前版本已经完整验证通过，无需代码修改，关闭处理。

closed 2026-09-05 by coolsnowwolf