coolsnowwolf/lede #13231: 最新版 京东云鲁班AX3000编译失败

wcloud520, 2024-12-23:
### 详细叙述

 京东云鲁班AX3000编译失败
[image]

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

京东云鲁班AX3000

### 详细日志

musl-objcopy -O binary -R .reginfo -R .notes -R .note -R .comment -R .mdebug -R .note.gnu.build-id -S /home/ua/lede/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq50xx/linux-6.1.119/vmlinux /home/ua/lede/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq50xx/vmlinux-initramfs; aarch64-openwrt-linux-musl-objcopy -R .reginfo -R .notes -R .note -R .comment -R .mdebug -R .note.gnu.build-id -S /home/ua/lede/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq50xx/linux-6.1.119/vmlinux /home/ua/lede/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq50xx/vmlinux-initramfs.elf; cp -fpR /home/ua/lede/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq50xx/linux-6.1.119/vmlinux /home/ua/lede/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq50xx/vmlinux-initramfs.debug;  cp -fpR /home/ua/lede/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq50xx/linux-6.1.119/arch/arm64/boot//Image /home/ua/lede/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq50xx/Image-initramfs;  }
touch /home/ua/lede/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq50xx/linux-6.1.119/.image
make -C image compile install TARGET_BUILD=
make[5]: Entering directory '/home/ua/lede/target/linux/qualcommax/image'
make[5]: Nothing to be done for 'compile'.
aarch64-openwrt-linux-musl-cpp -nostdinc -x assembler-with-cpp  -I/home/ua/lede/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq50xx/linux-6.1.119/arch/arm64/boot/dts -I/home/ua/lede/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq50xx/linux-6.1.119/arch/arm64/boot/dts/include -I/home/ua/lede/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq50xx/linux-6.1.119/include/ -I/home/ua/lede/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq50xx/linux-6.1.119/scripts/dtc/include-prefixes -undef -D__DTS__  -o /home/ua/lede/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq50xx/image-ipq5018-gl-b3000.dtb.tmp /home/ua/lede/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq50xx/linux-6.1.119/arch/arm64/boot/dts/qcom/ipq5018-gl-b3000.dts
/home/ua/lede/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq50xx/linux-6.1.119/scripts/dtc/dtc -O dtb -i/home/ua/lede/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq50xx/linux-6.1.119/arch/arm64/boot/dts/qcom/ -Wno-unit_address_vs_reg -Wno-simple_bus_reg -Wno-unit_address_format -Wno-pci_bridge -Wno-pci_device_bus_num -Wno-pci_device_reg -Wno-avoid_unnecessary_addr_size -Wno-alias_paths -Wno-graph_child_address -Wno-graph_port -Wno-unique_unit_address    -o /home/ua/lede/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq50xx/image-ipq5018-gl-b3000.dtb /home/ua/lede/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq50xx/image-ipq5018-gl-b3000.dtb.tmp
Error: /home/ua/lede/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq50xx/linux-6.1.119/arch/arm64/boot/dts/qcom/ipq5018-gl-b3000.dts:86.1-8 Label or path crypto not found
Error: /home/ua/lede/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq50xx/linux-6.1.119/arch/arm64/boot/dts/qcom/ipq5018-gl-b3000.dts:90.1-11 Label or path cryptobam not found
Error: /home/ua/lede/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq50xx/linux-6.1.119/arch/arm64/boot/dts/qcom/ipq5018-gl-b3000.dts:108.1-8 Label or path ge_phy not found
Error: /home/ua/lede/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq50xx/linux-6.1.119/arch/arm64/boot/dts/qcom/ipq5018-gl-b3000.dts:192.1-6 Label or path prng not found
Error: /home/ua/lede/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq50xx/linux-6.1.119/arch/arm64/boot/dts/qcom/ipq5018-gl-b3000.dts:196.1-11 Label or path q6v5_wcss not found
Error: /home/ua/lede/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq50xx/linux-6.1.119/arch/arm64/boot/dts/qcom/ipq5018-gl-b3000.dts:274.1-10 Label or path qpic_bam not found
Error: /home/ua/lede/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq50xx/linux-6.1.119/arch/arm64/boot/dts/qcom/ipq5018-gl-b3000.dts:278.1-11 Label or path qpic_nand not found
Error: /home/ua/lede/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq50xx/linux-6.1.119/arch/arm64/boot/dts/qcom/ipq5018-gl-b3000.dts:385.1-7 Label or path wifi0 not found
Error: /home/ua/lede/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq50xx/linux-6.1.119/arch/arm64/boot/dts/qcom/ipq5018-gl-b3000.dts:394.1-7 Label or path wifi1 not found
FATAL ERROR: Syntax error parsing input tree
make[5]: *** [Makefile:18: /home/ua/lede/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq50xx/image-ipq5018-gl-b3000.dtb] Error 1
make[5]: Leaving directory '/home/ua/lede/target/linux/qualcommax/image'
make[4]: *** [Makefile:23: install] Error 2
make[4]: Leaving directory '/home/ua/lede/target/linux/qualcommax'
make[3]: *** [Makefile:11: install] Error 2
make[3]: Leaving directory '/home/ua/lede/target/linux'
time: target/linux/install#67.89#77.60#139.40
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:30: target/linux/install] Error 1
make[2]: Leaving directory '/home/ua/lede'
make[1]: *** [target/Makefile:24: /home/ua/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.target_install] Error 2
make[1]: Leaving directory '/home/ua/lede'
make: *** [/home/ua/lede/include/toplevel.mk:231：world] 错误 2
root@ua-virtual-machine:/home/ua/lede# cd /home/ua/lede/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq50xx/linux-6.1.119/arch/arm64/boot/dts/qcom/^C
root@ua-virtual-machine:/home/ua/lede# uname -a
Linux ua-virtual-machine 6.8.0-49-generic #49~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Wed Nov  6 17:42:15 UTC 2 x86_64 x86_64 x86_64 GNU/Linux
root@ua-virtual-machine:/home/ua/lede# uname -srm
Linux 6.8.0-49-generic x86_64
root@ua-virtual-machine:/home/ua/lede# cd^C
root@ua-virtual-machine:/home/ua/lede# make menuconfig

lurenjia1213, 2025-02-20:
重新pull一下代码？

coolsnowwolf, 2026-09-05:
已在当前 master（c1dc394ac6a）重新验证，问题已经不存在：\n\n- 目标配置：`qualcommax/ipq50xx/jdcloud_re-cs-03`\n- 内核版本：Linux 6.12.108\n- `make target/linux/compile -j30 V=s` 编译成功\n- 按当前镜像的 CPP+DTC 规则分别编译 `ipq5018-re-cs-03.dts` 和旧日志实际报错的 `ipq5018-gl-b3000.dts`，均成功生成 DTB（26674 / 24989 bytes）\n- 旧日志中的 `crypto`、`cryptobam`、`ge_phy`、`prng`、`q6v5_wcss`、`qpic_bam`、`qpic_nand`、`wifi0`、`wifi1` 未定义错误均未复现\n\n该 issue 来自旧的 Linux 6.1.119 源码状态，当前代码无需额外修改，关闭处理。

closed 2026-09-05 by coolsnowwolf