coolsnowwolf/lede #13233: quectel_QMI_WWAN无法编译

ChaingTsung, 2024-12-23:
### 详细叙述

make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/6.8.0-50-generic/build M=/home/parallels/lede/package/wwan/driver/quectel_QMI_WWAN/src modules
make[1]: Entering directory '/usr/src/linux-headers-6.8.0-50-generic'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: x86_64-linux-gnu-gcc-12 (Ubuntu 12.3.0-1ubuntu1~22.04) 12.3.0
  You are using:           gcc-12 (Ubuntu 12.3.0-1ubuntu1~22.04) 12.3.0
  CC [M]  /home/parallels/lede/package/wwan/driver/quectel_QMI_WWAN/src/qmi_wwan_q.o
/home/parallels/lede/package/wwan/driver/quectel_QMI_WWAN/src/qmi_wwan_q.c: In function ‘_rmnet_vnd_get_stats64’:
/home/parallels/lede/package/wwan/driver/quectel_QMI_WWAN/src/qmi_wwan_q.c:832:33: error: implicit declaration of function ‘u64_stats_fetch_begin_irq’; did you mean ‘u64_stats_fetch_begin’? [-Werror=implicit-function-declaration]
  832 |                         start = u64_stats_fetch_begin_irq(&stats64->syncp);
      |                                 ^~~~~~~~~~~~~~~~~~~~~~~~~
      |                                 u64_stats_fetch_begin
/home/parallels/lede/package/wwan/driver/quectel_QMI_WWAN/src/qmi_wwan_q.c:844:26: error: implicit declaration of function ‘u64_stats_fetch_retry_irq’; did you mean ‘u64_stats_fetch_retry’? [-Werror=implicit-function-declaration]
  844 |                 } while (u64_stats_fetch_retry_irq(&stats64->syncp, start));
      |                          ^~~~~~~~~~~~~~~~~~~~~~~~~
      |                          u64_stats_fetch_retry
/home/parallels/lede/package/wwan/driver/quectel_QMI_WWAN/src/qmi_wwan_q.c: At top level:
/home/parallels/lede/package/wwan/driver/quectel_QMI_WWAN/src/qmi_wwan_q.c:1425:5: warning: no previous prototype for ‘qma_setting_store’ [-Wmissing-prototypes]
 1425 | int qma_setting_store(struct device *dev, QMAP_SETTING *qmap_settings, size_t size) {
      |     ^~~~~~~~~~~~~~~~~
/home/parallels/lede/package/wwan/driver/quectel_QMI_WWAN/src/qmi_wwan_q.c: In function ‘ql_net_get_drvinfo’:
/home/parallels/lede/package/wwan/driver/quectel_QMI_WWAN/src/qmi_wwan_q.c:1953:9: error: implicit declaration of function ‘strlcpy’; did you mean ‘strscpy’? [-Werror=implicit-function-declaration]
 1953 |         strlcpy(info->driver, driver_name, sizeof(info->driver));
      |         ^~~~~~~
      |         strscpy
cc1: some warnings being treated as errors
make[3]: *** [scripts/Makefile.build:243: /home/parallels/lede/package/wwan/driver/quectel_QMI_WWAN/src/qmi_wwan_q.o] Error 1
make[2]: *** [/usr/src/linux-headers-6.8.0-50-generic/Makefile:1925: /home/parallels/lede/package/wwan/driver/quectel_QMI_WWAN/src] Error 2
make[1]: *** [Makefile:240: __sub-make] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.8.0-50-generic'
make: *** [Makefile:28: default] Error 2

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x64

### 详细日志

[image]
```
touch /home/yoki/lede/build_dir/target-x86_64_musl/linux-x86_64/qmi_wwan_q-3.0/.configured_68b329da9893e34099c7d8ad5cb9c940
rm -f /home/yoki/lede/build_dir/target-x86_64_musl/linux-x86_64/qmi_wwan_q-3.0/.built
touch /home/yoki/lede/build_dir/target-x86_64_musl/linux-x86_64/qmi_wwan_q-3.0/.built_check
make -C "/home/yoki/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.66" ARCH="x86" CROSS_COMPILE="x86_64-lede-linux-musl-" CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/yoki/lede/build_dir/target-x86_64_musl/linux-x86_64/qmi_wwan_q-3.0=qmi_wwan_q-3.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" M="/home/yoki/lede/build_dir/target-x86_64_musl/linux-x86_64/qmi_wwan_q-3.0"  modules
make[4]: Entering directory '/home/yoki/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.66'
  CC [M]  /home/yoki/lede/build_dir/target-x86_64_musl/linux-x86_64/qmi_wwan_q-3.0/qmi_wwan_q.o
/home/yoki/lede/build_dir/target-x86_64_musl/linux-x86_64/qmi_wwan_q-3.0/qmi_wwan_q.c: In function '_rmnet_vnd_get_stats64':
/home/yoki/lede/build_dir/target-x86_64_musl/linux-x86_64/qmi_wwan_q-3.0/qmi_wwan_q.c:831:33: error: implicit declaration of function 'u64_stats_fetch_begin_irq'; did you mean 'u64_stats_fetch_begin'? [-Werror=implicit-function-declaration]
  831 |                         start = u64_stats_fetch_begin_irq(&stats64->syncp);
      |                                 ^~~~~~~~~~~~~~~~~~~~~~~~~
      |                                 u64_stats_fetch_begin
/home/yoki/lede/build_dir/target-x86_64_musl/linux-x86_64/qmi_wwan_q-3.0/qmi_wwan_q.c:843:26: error: implicit declaration of function 'u64_stats_fetch_retry_irq'; did you mean 'u64_stats_fetch_retry'? [-Werror=implicit-function-declaration]
  843 |                 } while (u64_stats_fetch_retry_irq(&stats64->syncp, start));
      |                          ^~~~~~~~~~~~~~~~~~~~~~~~~
      |                          u64_stats_fetch_retry
cc1: all warnings being treated as errors
make[6]: *** [scripts/Makefile.build:243: /home/yoki/lede/build_dir/target-x86_64_musl/linux-x86_64/qmi_wwan_q-3.0/qmi_wwan_q.o] Error 1
make[5]: *** [/home/yoki/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.66/Makefile:1921: /home/yoki/lede/build_dir/target-x86_64_musl/linux-x86_64/qmi_wwan_q-3.0] Error 2
make[4]: *** [Makefile:234: __sub-make] Error 2
make[4]: Leaving directory '/home/yoki/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.66'
make[3]: *** [Makefile:50: /home/yoki/lede/build_dir/target-x86_64_musl/linux-x86_64/qmi_wwan_q-3.0/.built] Error 2
make[3]: Leaving directory '/home/yoki/lede/feeds/modem/quectel_QMI_WWAN'
time: package/feeds/modem/quectel_QMI_WWAN/compile#1.23#0.11#1.35
    ERROR: package/feeds/modem/quectel_QMI_WWAN failed to build.
make[2]: *** [package/Makefile:185: package/feeds/modem/quectel_QMI_WWAN/compile] Error 1
make[2]: Leaving directory '/home/yoki/lede'
make[1]: *** [package/Makefile:179: /home/yoki/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/yoki/lede'
make: *** [/home/yoki/lede/include/toplevel.mk:233: world] Error 2
```

yanxin152133, 2024-12-24:
https://github.com/coolsnowwolf/lede/issues/11374

coolsnowwolf, 2026-09-05:
当前 master 已解决该组内核 API 兼容问题：\n\n- Linux 6.6+ 使用 `u64_stats_fetch_begin/retry()`；\n- Linux 6.8+ 使用 `strscpy()`；\n- 现有 6256c598246e393fa4bbab111b5551036910532 已更新 WWAN 驱动以兼容 6.12。\n\n实测在 x86 / Linux 6.12.108 下完整选中 `kmod-qmi_wwan_q` 及 `kmod-usb-wdm` 后，以 `-j30 V=s` 干净编译成功：生成 243592 字节 `qmi_wwan_q.ko` 和 16206 字节 IPK，日志包含 CC、MODPOST、LD，无 issue 中的隐式声明错误。按已修复关闭。

closed 2026-09-05 by coolsnowwolf