coolsnowwolf/lede #13242: Turbo ACC 网络加速设置不生效

Shawleevip, 2024-12-28:
### 详细叙述

L大有时间帮忙看看，谢谢

固件版本   LuCI openwrt-23.05 branch git-24.339.48605-e1135e2

[image]

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86

### 详细日志

无

xiangfeidexiaohuo, 2024-12-28:
你说的NAT1 不显示状态吧，Lean之前说修好了，应该是没推。

xiangfeidexiaohuo, 2024-12-28:
https://github.com/coolsnowwolf/luci/issues/312

Jejz168, 2024-12-31:
> https://github.com/coolsnowwolf/luci/issues/312

大雕修复的是1806的😂

xiangfeidexiaohuo, 2024-12-31:
> > [coolsnowwolf/luci#312](https://github.com/coolsnowwolf/luci/issues/312)
> 
> 大雕修复的是1806的😂

默默等待2305的

coolsnowwolf, 2024-12-31:
保存配置冲突了而已

richaradlee, 2025-01-02:
请问coolsnowwolf该如何修复呢？
我把：
1、openwrt恢复默认设置
2、从新编译固件后选择不保存配置更新
都不行，配置完后还是无法保存。
还请赐教如何修复？

shjy10000, 2025-01-03:
你在编译的时候选择了使用FW4的插件或者某些使用FW4的插件自带fullconenat,不要在编译的时候选择mihomo、homeproxy这些插件，这几个插件都会导致这个问题的

Shawleevip, 2025-01-03:
谢谢解答，终于搞明白了。

Shawleevip, 2025-01-03:
> 你在编译的时候选择了使用FW4的插件或者某些使用FW4的插件自带fullconenat,不要在编译的时候选择mihomo、homeproxy这些插件，这几个插件都会导致这个问题的

谢谢解答，终于搞明白了。

closed 2025-03-27 by Shawleevip