coolsnowwolf/lede #13246: 最新云编译出错！（ vpdpbusd）

NeedforGit, 2024-12-30:
### 详细叙述

云编译 X86，提示错误：

/workdir/openwrt/build_dir/host/libdeflate-1.22/libdeflate-gzip
/workdir/openwrt/tmp/ccvwe81g.s: Assembler messages:
/workdir/openwrt/tmp/ccvwe81g.s:1158: Error: unsupported instruction `vpdpbusd'

详细的请看日志，请问如何解决？

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

X86

### 详细日志

make[3]: Entering directory '/workdir/openwrt/tools/libdeflate'
/workdir/openwrt/staging_dir/host/bin/gcc -O2 -I/workdir/openwrt/staging_dir/host/include  -L/workdir/openwrt/staging_dir/host/lib  /workdir/openwrt/build_dir/host/libdeflate-1.22/lib/*{,/*}.c /workdir/openwrt/build_dir/host/libdeflate-1.22/programs/{gzip,prog_util,tgetopt}.c -o /workdir/openwrt/build_dir/host/libdeflate-1.22/libdeflate-gzip
/workdir/openwrt/tmp/ccvwe81g.s: Assembler messages:
/workdir/openwrt/tmp/ccvwe81g.s:1158: Error: unsupported instruction `vpdpbusd'
/workdir/openwrt/tmp/ccvwe81g.s:1159: Error: unsupported instruction `vpdpbusd'
/workdir/openwrt/tmp/ccvwe81g.s:1160: Error: unsupported instruction `vpdpbusd'
/workdir/openwrt/tmp/ccvwe81g.s:1161: Error: unsupported instruction `vpdpbusd'
/workdir/openwrt/tmp/ccvwe81g.s:1162: Error: unsupported instruction `vpdpbusd'
/workdir/openwrt/tmp/ccvwe81g.s:1180: Error: unsupported instruction `vpdpbusd'
/workdir/openwrt/tmp/ccvwe81g.s:1181: Error: unsupported instruction `vpdpbusd'
/workdir/openwrt/tmp/ccvwe81g.s:1213: Error: unsupported instruction `vpdpbusd'
/workdir/openwrt/tmp/ccvwe81g.s:1214: Error: unsupported instruction `vpdpbusd'
/workdir/openwrt/tmp/ccvwe81g.s:1285: Error: unsupported instruction `vpdpbusd'
/workdir/openwrt/tmp/ccvwe81g.s:1286: Error: unsupported instruction `vpdpbusd'
/workdir/openwrt/tmp/ccvwe81g.s:1288: Error: unsupported instruction `vpdpbusd'
/workdir/openwrt/tmp/ccvwe81g.s:1289: Error: unsupported instruction `vpdpbusd'
/workdir/openwrt/tmp/ccvwe81g.s:1290: Error: unsupported instruction `vpdpbusd'
/workdir/openwrt/tmp/ccvwe81g.s:1292: Error: unsupported instruction `vpdpbusd'
/workdir/openwrt/tmp/ccvwe81g.s:1294: Error: unsupported instruction `vpdpbusd'
/workdir/openwrt/tmp/ccvwe[81](https://github.com/NeedforGit/Actions-OpenWrt/actions/runs/12541368083/job/34969834665#step:13:82)g.s:1295: Error: unsupported instruction `vpdpbusd'
make[3]: *** [Makefile:40: /workdir/openwrt/build_dir/host/libdeflate-1.22/.built] Error 1
make[3]: Leaving directory '/workdir/openwrt/tools/libdeflate'
time: tools/libdeflate/compile#3.91#0.44#4.35
    ERROR: tools/libdeflate failed to build.
make[2]: *** [tools/Makefile:231: tools/libdeflate/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [tools/Makefile:227: /workdir/openwrt/staging_dir/host/stamp/.tools_compile_nyyynyynnnyyynynyyyyyyynyyynynyynnynnyyyyyynyyyyyyyyyyynynnyyyyynyyy] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 2

liyu-cloud, 2025-01-01:
刚刚本地编译也是一样的错误

yanxin152133, 2025-01-06:
内存太小吧
https://github.com/coolsnowwolf/lede/issues/13228
https://github.com/coolsnowwolf/lede/issues/12015

311303, 2026-01-13:
用ubuntu 20.04编译最新源码也报了这错。找了一圈最后在[libdeflate那里](https://github.com/ebiggers/libdeflate/issues/365)找到了答案。是本机binutils的版本太旧的祸，升级到2.36以后即可。

coolsnowwolf, 2026-09-05:
已确认原因是宿主机 binutils 版本过旧，无法汇编 libdeflate 使用的 `vpdpbusd` 指令；评论中已验证升级到 binutils 2.36+ 可解决，属于云编译环境问题而非 LEDE 源码问题，现关闭。

closed 2026-09-05 by coolsnowwolf