coolsnowwolf/lede #13248: ksmbd编译失败！

Fly432, 2025-01-01:
### 详细叙述

ksmbd 在kernel6.6.66内核编译失败！

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

X86

### 详细日志

make[3]: Entering directory '/home/gyk/lede/package/kernel/ksmbd'
rm -f /home/gyk/lede/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-2024.10.14/.built
touch /home/gyk/lede/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-2024.10.14/.built_check
make -C /home/gyk/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.66 KCFLAGS="-fmacro-prefix-map=/home/gyk/lede/build_dir/target-x86_64_musl=target-x86_64_musl -fno-caller-saves " HOSTCFLAGS="-O2 -I/home/gyk/lede/staging_dir/host/include -I/home/gyk/lede/staging_dir/hostpkg/include -I/home/gyk/lede/staging_dir/target-x86_64_musl/host/include -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="x86_64-openwrt-linux-musl-" ARCH="x86" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Mon Dec 16 12:30:22 2024" KBUILD_BUILD_VERSION="0" KBUILD_HOSTLDFLAGS="-L/home/gyk/lede/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls= KBUILD_EXTRA_SYMBOLS="/home/gyk/lede/build_dir/target-x86_64_musl/linux-x86_64/symvers/cryptodev-linux.symvers /home/gyk/lede/build_dir/target-x86_64_musl/linux-x86_64/symvers/fullconenat.symvers" KERNELRELEASE=6.6.66 M="/home/gyk/lede/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-2024.10.14" EXTRA_CFLAGS="-DCONFIG_SMB_INSECURE_SERVER=1" CONFIG_SMB_INSECURE_SERVER=y CONFIG_SMB_SERVER=m modules
make[4]: Entering directory '/home/gyk/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.66'
  CC [M]  /home/gyk/lede/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-2024.10.14/smb1pdu.o
/home/gyk/lede/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-2024.10.14/smb1pdu.c: In function 'smb_rename':
/home/gyk/lede/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-2024.10.14/smb1pdu.c:864:14: error: implicit declaration of function 'ksmbd_vfs_kern_path'; did you mean 'ksmbd_vfs_zero_data'? [-Werror=implicit-function-declaration]
  864 |         rc = ksmbd_vfs_kern_path(work, newname, LOOKUP_NO_SYMLINKS, &path, 1);
      |              ^~~~~~~~~~~~~~~~~~~
      |              ksmbd_vfs_zero_data
/home/gyk/lede/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-2024.10.14/smb1pdu.c:890:14: error: implicit declaration of function 'ksmbd_vfs_fp_rename'; did you mean 'ksmbd_vfs_rename'? [-Werror=implicit-function-declaration]
  890 |         rc = ksmbd_vfs_fp_rename(work, fp, newname);
      |              ^~~~~~~~~~~~~~~~~~~
      |              ksmbd_vfs_rename
In file included from ./include/linux/string.h:294,
                 from ./include/linux/bitmap.h:11,
                 from ./include/linux/cpumask.h:12,
                 from ./arch/x86/include/asm/paravirt.h:17,
                 from ./arch/x86/include/asm/cpuid.h:62,
                 from ./arch/x86/include/asm/processor.h:19,
                 from ./arch/x86/include/asm/cpufeature.h:5,
                 from ./arch/x86/include/asm/thread_info.h:53,
                 from ./include/linux/thread_info.h:60,
                 from ./arch/x86/include/asm/preempt.h:9,
                 from ./include/linux/preempt.h:79,
                 from ./include/linux/spinlock.h:56,
                 from ./include/linux/wait.h:9,
                 from ./include/linux/wait_bit.h:8,
                 from ./include/linux/fs.h:6,
                 from /home/gyk/lede/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-2024.10.14/smb1pdu.c:7:
In function 'fortify_memset_chk',
    inlined from 'query_file_info_pipe' at /home/gyk/lede/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-2024.10.14/smb1pdu.c:6833:2,
    inlined from 'query_file_info' at /home/gyk/lede/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-2024.10.14/smb1pdu.c:6872:10:
./include/linux/fortify-string.h:507:25: warning: call to '__write_overflow_field' declared with attribute warning: detected write beyond size of field (1st parameter); maybe use struct_group()? [-Wattribute-warning]
  507 |                         __write_overflow_field(p_size_field, size);
      |                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In function 'fortify_memset_chk',
    inlined from 'query_file_info' at /home/gyk/lede/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-2024.10.14/smb1pdu.c:6917:3:
./include/linux/fortify-string.h:507:25: warning: call to '__write_overflow_field' declared with attribute warning: detected write beyond size of field (1st parameter); maybe use struct_group()? [-Wattribute-warning]
  507 |                         __write_overflow_field(p_size_field, size);
      |                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In function 'fortify_memset_chk',
    inlined from 'query_file_info' at /home/gyk/lede/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-2024.10.14/smb1pdu.c:6952:3:
./include/linux/fortify-string.h:507:25: warning: call to '__write_overflow_field' declared with attribute warning: detected write beyond size of field (1st parameter); maybe use struct_group()? [-Wattribute-warning]
  507 |                         __write_overflow_field(p_size_field, size);
      |                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In function 'fortify_memset_chk',
    inlined from 'query_file_info' at /home/gyk/lede/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-2024.10.14/smb1pdu.c:6991:3:
./include/linux/fortify-string.h:507:25: warning: call to '__write_overflow_field' declared with attribute warning: detected write beyond size of field (1st parameter); maybe use struct_group()? [-Wattribute-warning]
  507 |                         __write_overflow_field(p_size_field, size);
      |                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In function 'fortify_memset_chk',
    inlined from 'query_file_info' at /home/gyk/lede/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-2024.10.14/smb1pdu.c:7023:3:
./include/linux/fortify-string.h:507:25: warning: call to '__write_overflow_field' declared with attribute warning: detected write beyond size of field (1st parameter); maybe use struct_group()? [-Wattribute-warning]
  507 |                         __write_overflow_field(p_size_field, size);
[... 94 lines cut ...]
      |                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In function 'fortify_memset_chk',
    inlined from 'smb_open_andx' at /home/gyk/lede/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-2024.10.14/smb1pdu.c:8323:3:
./include/linux/fortify-string.h:507:25: warning: call to '__write_overflow_field' declared with attribute warning: detected write beyond size of field (1st parameter); maybe use struct_group()? [-Wattribute-warning]
  507 |                         __write_overflow_field(p_size_field, size);
      |                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In function 'fortify_memset_chk',
    inlined from 'smb_open_andx' at /home/gyk/lede/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-2024.10.14/smb1pdu.c:8386:4:
./include/linux/fortify-string.h:507:25: warning: call to '__write_overflow_field' declared with attribute warning: detected write beyond size of field (1st parameter); maybe use struct_group()? [-Wattribute-warning]
  507 |                         __write_overflow_field(p_size_field, size);
      |                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
make[6]: *** [scripts/Makefile.build:243: /home/gyk/lede/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-2024.10.14/smb1pdu.o] Error 1
make[5]: *** [/home/gyk/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.66/Makefile:1921: /home/gyk/lede/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-2024.10.14] Error 2
make[4]: *** [Makefile:234: __sub-make] Error 2
make[4]: Leaving directory '/home/gyk/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.66'
make[3]: *** [Makefile:73: /home/gyk/lede/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-2024.10.14/.built] Error 2
make[3]: Leaving directory '/home/gyk/lede/package/kernel/ksmbd'
time: package/kernel/ksmbd/compile#2.59#0.35#4.03
    ERROR: package/kernel/ksmbd failed to build.
make[2]: *** [package/Makefile:116: package/kernel/ksmbd/compile] Error 1
make[2]: Leaving directory '/home/gyk/lede'
make[1]: *** [package/Makefile:110: /home/gyk/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/gyk/lede'
make: *** [/home/gyk/lede/include/toplevel.mk:231: world] Error 2

yanxin152133, 2025-01-03:
这个之前不是说修复了吗

xbclub, 2025-01-05:
我也遇到了ksmbd 报错

coolsnowwolf, 2025-01-06:
你代码没更新

closed 2025-01-13 by coolsnowwolf