coolsnowwolf/lede #13253: fine3399无法启动

aguoxing, 2025-01-02:
### 详细叙述

U-Boot TPL 2024.10-OpenWrt-r84-d379cee20 (Jan 02 2025 - 03:38:20)
rk3399_dmc_of_to_plat: Cannot read rockchip,sdram-params -1
DRAM init failed: -1

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

fine3399

### 详细日志

无

aiamadeus, 2025-01-03:
```
--- a/package/boot/uboot-rockchip/patches/302-rockchip-rk3399-Add-support-for-rumu3f-fine3399.patch
+++ b/package/boot/uboot-rockchip/patches/302-rockchip-rk3399-Add-support-for-rumu3f-fine3399.patch
@@ -1,4 +1,21 @@
 --- /dev/null
++++ b/arch/arm/dts/rk3399-fine-3399-u-boot.dtsi
+@@ -0,0 +1,14 @@
++// SPDX-License-Identifier: GPL-2.0+
++
++#include "rk3399-u-boot.dtsi"
++#include "rk3399-sdram-lpddr4-100.dtsi"
++
++/ {
++	chosen {
++		u-boot,spl-boot-order = "same-as-spl", &sdhci, &sdmmc;
++	};
++};
++
++&vdd_log {
++	regulator-init-microvolt = <950000>;
++};
+--- /dev/null
 +++ b/arch/arm/dts/rk3399-fine-3399.dts
 @@ -0,0 +1,789 @@
 +// SPDX-License-Identifier: (GPL-2.0+ OR MIT)

```
这样子呢

aguoxing, 2025-01-03:
内存丝印是S5RG2G20CMS-CGCJ  查了下应该是ddr4的 加上补丁用另外一块板子试了报同样的错

aiamadeus, 2025-01-03:
`make package/uboot-rockchip/{clean,compile} V=s` 编译完然后重刷试过了吗？
rk3399_dmc_of_to_plat 就是要 "rk3399-sdram-lpddr4-100.dtsi" 的参数

aguoxing, 2025-01-03:
我是直接make编译后重新刷过 可能补丁没打上？
不过暂时试不了 进不了maskrom模式

aguoxing, 2025-01-04:
之前确实是补丁没打上  现在能正常启动了