coolsnowwolf/lede #13254: 拉取最新源码编译6.6内核固件报错

bi4tms, 2025-01-05:
### 详细叙述

如题，编译R2S固件报错
@coolsnowwolf 

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

R2S

### 详细日志

Applying /home/r2s.6.6/lede/target/linux/rockchip/patches-6.6/220-phy-rockchip-naneng-combphy-fix-phy-reset.patch using plaintext: 
patching file arch/arm64/boot/dts/rockchip/rk3568.dtsi
patching file arch/arm64/boot/dts/rockchip/rk356x.dtsi
patching file drivers/phy/rockchip/phy-rockchip-naneng-combphy.c
Hunk #1 FAILED at 324.
1 out of 1 hunk FAILED -- saving rejects to file drivers/phy/rockchip/phy-rockchip-naneng-combphy.c.rej
Patch failed!  Please fix /home/r2s.6.6/lede/target/linux/rockchip/patches-6.6/220-phy-rockchip-naneng-combphy-fix-phy-reset.patch!
make[4]: *** [Makefile:34: /home/r2s.6.6/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-6.6.69/.prepared_22b537c0fe76b1beeaef27bde817e058] Error 1
make[4]: Leaving directory '/home/r2s.6.6/lede/target/linux/rockchip'
make[3]: *** [Makefile:11: compile] Error 2
make[3]: Leaving directory '/home/r2s.6.6/lede/target/linux'
time: target/linux/compile#17.11#11.11#26.76
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:30: target/linux/compile] Error 1
make[2]: Leaving directory '/home/r2s.6.6/lede'
make[1]: *** [target/Makefile:23: /home/r2s.6.6/lede/staging_dir/target-aarch64_generic_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/home/r2s.6.6/lede'
make: *** [/home/r2s.6.6/lede/include/toplevel.mk:231：world] 错误 2

gssiwmd, 2025-01-05:
R5S一样报错

findnr, 2025-01-06:
r68s也是报同样的错误

commit f4046790 (2025-01-06): rockchip: sync upstream fixes

Fixes: #13254
closed 2025-01-06 by aiamadeus