coolsnowwolf/lede #13255: 编译mediatek报错

audyfan, 2025-01-05:
### 详细叙述

最新源码编译报错

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

H3C NX30 Pro

### 详细日志

install ./include/openssl/idea.h -> /home/audy/lede/build_dir/target-aarch64_cortex-a53_musl/openssl-3.0.15/ipkg-install/usr/include/openssl/idea.h
install ./include/openssl/kdf.h -> /home/audy/lede/build_dir/target-aarch64_cortex-a53_musl/openssl-3.0.15/ipkg-install/usr/include/openssl/kdf.h
install ./include/openssl/kdferr.h -> /home/audy/lede/build_dir/target-aarch64_cortex-a53_musl/openssl-3.0.15/ipkg-install/usr/include/openssl/kdferr.h
install ./include/openssl/lhash.h -> /home/audy/lede/build_dir/target-aarch64_cortex-a53_musl/openssl-3.0.15/ipkg-install/usr/include/openssl/lhash.h
install ./include/openssl/macros.h -> /home/audy/lede/build_dir/target-aarch64_cortex-a53_musl/openssl-3.0.15/ipkg-install/usr/include/openssl/macros.h
install ./include/openssl/md2.h -> /home/audy/lede/build_dir/target-aarch64_cortex-a53_musl/openssl-3.0.15/ipkg-install/usr/include/openssl/md2.h
install ./include/openssl/md4.h -> /home/audy/lede/build_dir/target-aarch64_cortex-a53_musl/openssl-3.0.15/ipkg-install/usr/include/openssl/md4.h
install ./include/openssl/md5.h -> /home/audy/lede/build_dir/target-aarch64_cortex-a53_musl/openssl-3.0.15/ipkg-install/usr/include/openssl/md5.h
install ./include/openssl/mdc2.h -> /home/audy/lede/build_dir/target-aarch64_cortex-a53_musl/openssl-3.0.15/ipkg-install/usr/include/openssl/mdc2.h
install ./include/openssl/modes.h -> /home/audy/lede/build_dir/target-aarch64_cortex-a53_musl/openssl-3.0.15/ipkg-install/usr/include/openssl/modes.h
install ./include/openssl/obj_mac.h -> /home/audy/lede/build_dir/target-aarch64_cortex-a53_musl/openssl-3.0.15/ipkg-install/usr/include/openssl/obj_mac.h
install ./include/openssl/objects.h -> /home/audy/lede/build_dir/target-aarch64_cortex-a53_musl/openssl-3.0.15/ipkg-install/usr/include/openssl/objects.h
install ./include/openssl/objectserr.h -> /home/audy/lede/build_dir/target-aarch64_cortex-a53_musl/openssl-3.0.15/ipkg-install/usr/include/openssl/objectserr.h
install ./include/openssl/ocsp.h -> /home/audy/lede/build_dir/target-aarch64_cortex-a53_musl/openssl-3.0.15/ipkg-install/usr/include/openssl/ocsp.h
install ./include/openssl/ocsperr.h -> /home/audy/lede/build_dir/target-aarch64_cortex-a53_musl/openssl-3.0.15/ipkg-install/usr/include/openssl/ocsperr.h
install ./include/openssl/opensslconf.h -> /home/audy/lede/build_dir/target-aarch64_cortex-a53_musl/openssl-3.0.15/ipkg-install/usr/include/openssl/opensslconf.h
install ./include/openssl/opensslv.h -> /home/audy/lede/build_dir/target-aarch64_cortex-a53_musl/openssl-3.0.15/ipkg-install/usr/include/openssl/opensslv.h
install ./include/openssl/ossl_typ.h -> /home/audy/lede/build_dir/target-aarch64_cortex-a53_musl/openssl-3.0.15/ipkg-install/usr/include/openssl/ossl_typ.h
install ./include/openssl/param_build.h -> /home/audy/lede/build_dir/target-aarch64_cortex-a53_musl/openssl-3.0.15/ipkg-install/usr/include/openssl/param_build.h
install ./include/openssl/params.h -> /home/audy/lede/build_dir/target-aarch64_cortex-a53_musl/openssl-3.0.15/ipkg-install/usr/include/openssl/params.h
install ./include/openssl/pem.h -> /home/audy/lede/build_dir/target-aarch64_cortex-a53_musl/openssl-3.0.15/ipkg-install/usr/include/openssl/pem.h
install ./include/openssl/pem2.h -> /home/audy/lede/build_dir/target-aarch64_cortex-a53_musl/openssl-3.0.15/ipkg-install/usr/include/openssl/pem2.h
install ./include/openssl/pemerr.h -> /home/audy/lede/build_dir/target-aarch64_cortex-a53_musl/openssl-3.0.15/ipkg-install/usr/include/openssl/pemerr.h
install ./include/openssl/pkcs12.h -> /home/audy/lede/build_dir/target-aarch64_cortex-a53_musl/openssl-3.0.15/ipkg-install/usr/include/openssl/pkcs12.h
install ./include/openssl/pkcs12err.h -> /home/audy/lede/build_dir/target-aarch64_cortex-a53_musl/openssl-3.0.15/ipkg-install/usr/include/openssl/pkcs12err.h
install ./include/openssl/pkcs7.h -> /home/audy/lede/build_dir/target-aarch64_cortex-a53_musl/openssl-3.0.15/ipkg-install/usr/include/openssl/pkcs7.h
install ./include/openssl/pkcs7err.h -> /home/audy/lede/build_dir/target-aarch64_cortex-a53_musl/openssl-3.0.15/ipkg-install/usr/include/openssl/pkcs7err.h
install ./include/openssl/prov_ssl.h -> /home/audy/lede/build_dir/target-aarch64_cortex-a53_musl/openssl-3.0.15/ipkg-install/usr/include/openssl/prov_ssl.h
install ./include/openssl/proverr.h -> /home/audy/lede/build_dir/target-aarch64_cortex-a53_musl/openssl-3.0.15/ipkg-install/usr/include/openssl/proverr.h
install ./include/openssl/provider.h -> /home/audy/lede/build_dir/target-aarch64_cortex-a53_musl/openssl-3.0.15/ipkg-install/usr/include/openssl/provider.h
install ./include/openssl/rand.h -> /home/audy/lede/build_dir/target-aarch64_cortex-a53_musl/openssl-3.0.15/ipkg-install/usr/include/openssl/rand.h
install ./include/openssl/randerr.h -> /home/audy/lede/build_dir/target-aarch64_cortex-a53_musl/openssl-3.0.15/ipkg-install/usr/include/openssl/randerr.h
install ./include/openssl/rc2.h -> /home/audy/lede/build_dir/target-aarch64_cortex-a53_musl/openssl-3.0.15/ipkg-install/usr/include/openssl/rc2.h
install ./include/openssl/rc4.h -> /home/audy/lede/build_dir/target-aarch64_cortex-a53_musl/openssl-3.0.15/ipkg-install/usr/include/openssl/rc4.h
install ./include/openssl/rc5.h -> /home/audy/lede/build_dir/target-aarch64_cortex-a53_musl/openssl-3.0.15/ipkg-install/usr/include/openssl/rc5.h
install ./include/openssl/ripemd.h -> /home/audy/lede/build_dir/target-aarch64_cortex-a53_musl/openssl-3.0.15/ipkg-install/usr/include/openssl/ripemd.h
install ./include/openssl/rsa.h -> /home/audy/lede/build_dir/target-aarch64_cortex-a53_musl/openssl-3.0.15/ipkg-install/usr/include/openssl/rsa.h
install ./include/openssl/rsaerr.h -> /home/audy/lede/build_dir/target-aarch64_cortex-a53_musl/openssl-3.0.15/ipkg-install/usr/include/openssl/rsaerr.h
install ./include/openssl/safestack.h -> /home/audy/lede/build_dir/target-aarch64_cortex-a53_musl/openssl-3.0.15/ipkg-install/usr/include/openssl/safestack.h
install ./include/openssl/seed.h -> /home/audy/lede/build_dir/target-aarch64_cortex-a53_musl/openssl-3.0.15/ipkg-install/usr/include/openssl/seed.h
install ./include/openssl/self_test.h -> /home/audy/lede/build_dir/target-aarch64_cortex-a53_musl/openssl-3.0.15/ipkg-install/usr/include/openssl/self_test.h
install ./include/openssl/sha.h -> /home/audy/lede/build_dir/target-aarch64_cortex-a53_musl/openssl-3.0.15/ipkg-install/usr/include/openssl/sha.h
install ./include/openssl/srp.h -> /home/audy/lede/build_dir/target-aarch64_cortex-a53_musl/openssl-3.0.15/ipkg-install/usr/include/openssl/srp.h
install ./include/openssl/srtp.h -> /home/audy/lede/build_dir/target-aarch64_cortex-a53_musl/openssl-3.0.15/ipkg-install/usr/include/openssl/srtp.h
install ./include/openssl/ssl.h -> /home/audy/lede/build_dir/target-aarch64_cortex-a53_musl/openssl-3.0.15/ipkg-install/usr/include/openssl/ssl.h
install ./include/openssl/ssl2.h -> /home/audy/lede/build_dir/target-aarch64_cortex-a53_musl/openssl-3.0.15/ipkg-install/usr/include/openssl/ssl2.h
install ./include/openssl/ssl3.h -> /home/audy/lede/build_dir/target-aarch64_cortex-a53_musl/openssl-3.0.15/ipkg-install/usr/include/openssl/ssl3.h
install ./include/openssl/sslerr.h -> /home/audy/lede/build_dir/target-aarch64_cortex-a53_musl/openssl-3.0.15/ipkg-install/usr/include/openssl/sslerr.h
install ./include/openssl/sslerr_legacy.h -> /home/audy/lede/build_dir/target-aarch64_cortex-a53_musl/openssl-3.0.15/ipkg-install/usr/include/openssl/sslerr_legacy.h
install ./include/openssl/stack.h -> /home/audy/lede/build_dir/target-aarch64_cortex-a53_musl/openssl-3.0.15/ipkg-install/usr/include/openssl/stack.h
install ./include/openssl/store.h -> /home/audy/lede/build_dir/target-aarch64_cortex-a53_musl/openssl-3.0.15/ipkg-install/usr/include/openssl/store.h
install ./include/openssl/storeerr.h -> /home/audy/lede/build_dir/target-aarch64_cortex-a53_musl/openssl-3.0.15/ipkg-install/usr/include/openssl/storeerr.h
install ./include/openssl/symhacks.h -> /home/audy/lede/build_dir/target-aarch64_cortex-a53_musl/openssl-3.0.15/ipkg-install/usr/include/openssl/symhacks.h
install ./include/openssl/tls1.h -> /home/audy/lede/build_dir/target-aarch64_cortex-a53_musl/openssl-3.0.15/ipkg-install/usr/include/openssl/tls1.h
install ./include/openssl/trace.h -> /home/audy/lede/build_dir/target-aarch64_cortex-a53_musl/openssl-3.0.15/ipkg-install/usr/include/openssl/trace.h
[... cut ...]
echo "openssl-util" >> /home/audy/lede/staging_dir/target-aarch64_cortex-a53_musl/pkginfo/openssl.default.install
make[3]: Leaving directory '/home/audy/lede/package/libs/openssl'
time: package/libs/openssl/compile#425.93#50.25#191.71
make[2]: Leaving directory '/home/audy/lede'
make[1]: *** [package/Makefile:110: /home/audy/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/audy/lede'
make: *** [/home/audy/lede/include/toplevel.mk:231: world] Error 2

yanxin152133, 2025-01-06:
https://github.com/coolsnowwolf/lede/issues/11772

audyfan, 2025-01-06:
> #11772

谢谢大佬回复

closed 2025-01-06 by audyfan