coolsnowwolf/lede #13256: 最新源码编译mediatek报错

audyfan, 2025-01-06:
### 详细叙述

使用了kenzo大插件库，多次编译都是一直卡在这个错误

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

RAX3000M emmc

### 详细日志

.a/libgnu_gpl_la-sys_socket.o .libs/libcmd-certtool.lax/libcerttool-cfg.a/libgnu_gpl_la-threadlib.o .libs/libcmd-certtool.lax/libcerttool-cfg.a/libgnu_gpl_la-time_rz.o .libs/libcmd-certtool.lax/libcerttool-cfg.a/libgnu_gpl_la-timegm.o .libs/libcmd-certtool.lax/libcerttool-cfg.a/libgnu_gpl_la-timespec.o .libs/libcmd-certtool.lax/libcerttool-cfg.a/libgnu_gpl_la-tls.o .libs/libcmd-certtool.lax/libcerttool-cfg.a/libgnu_gpl_la-unistd.o .libs/libcmd-certtool.lax/libcerttool-cfg.a/libgnu_gpl_la-xalloc-die.o .libs/libcmd-certtool.lax/libcerttool-cfg.a/libgnu_gpl_la-xmalloc.o .libs/libcmd-certtool.lax/libcerttool-cfg.a/libgnu_gpl_la-xsize.o .libs/libcmd-certtool.lax/libcerttool-cfg.a/libgnu_la-basename-lgpl.o .libs/libcmd-certtool.lax/libcerttool-cfg.a/libgnu_la-bitrotate.o .libs/libcmd-certtool.lax/libcerttool-cfg.a/libgnu_la-c-ctype.o .libs/libcmd-certtool.lax/libcerttool-cfg.a/libgnu_la-c-strcasecmp.o .libs/libcmd-certtool.lax/libcerttool-cfg.a/libgnu_la-c-strncasecmp.o .libs/libcmd-certtool.lax/libcerttool-cfg.a/libgnu_la-canonicalize-lgpl.o .libs/libcmd-certtool.lax/libcerttool-cfg.a/libgnu_la-cloexec.o .libs/libcmd-certtool.lax/libcerttool-cfg.a/libgnu_la-dirname-lgpl.o .libs/libcmd-certtool.lax/libcerttool-cfg.a/libgnu_la-fcntl.o .libs/libcmd-certtool.lax/libcerttool-cfg.a/libgnu_la-fd-hook.o .libs/libcmd-certtool.lax/libcerttool-cfg.a/libgnu_la-free.o .libs/libcmd-certtool.lax/libcerttool-cfg.a/libgnu_la-getdelim.o .libs/libcmd-certtool.lax/libcerttool-cfg.a/libgnu_la-getline.o .libs/libcmd-certtool.lax/libcerttool-cfg.a/libgnu_la-gl_linkedhash_list.o .libs/libcmd-certtool.lax/libcerttool-cfg.a/libgnu_la-gl_list.o .libs/libcmd-certtool.lax/libcerttool-cfg.a/libgnu_la-hash-pjw-bare.o .libs/libcmd-certtool.lax/libcerttool-cfg.a/libgnu_la-hash.o .libs/libcmd-certtool.lax/libcerttool-cfg.a/libgnu_la-lock.o .libs/libcmd-certtool.lax/libcerttool-cfg.a/libgnu_la-malloca.o .libs/libcmd-certtool.lax/libcerttool-cfg.a/libgnu_la-memchr.o .libs/libcmd-certtool.lax/libcerttool-cfg.a/libgnu_la-memset_explicit.o .libs/libcmd-certtool.lax/libcerttool-cfg.a/libgnu_la-rawmemchr.o .libs/libcmd-certtool.lax/libcerttool-cfg.a/libgnu_la-read-file.o .libs/libcmd-certtool.lax/libcerttool-cfg.a/libgnu_la-scratch_buffer_grow.o .libs/libcmd-certtool.lax/libcerttool-cfg.a/libgnu_la-scratch_buffer_grow_preserve.o .libs/libcmd-certtool.lax/libcerttool-cfg.a/libgnu_la-scratch_buffer_set_array_size.o .libs/libcmd-certtool.lax/libcerttool-cfg.a/libgnu_la-stat-time.o .libs/libcmd-certtool.lax/libcerttool-cfg.a/libgnu_la-stripslash.o .libs/libcmd-certtool.lax/libcerttool-cfg.a/libgnu_la-sys_socket.o .libs/libcmd-certtool.lax/libcerttool-cfg.a/libgnu_la-threadlib.o .libs/libcmd-certtool.lax/libcerttool-cfg.a/libgnu_la-tls.o .libs/libcmd-certtool.lax/libcerttool-cfg.a/libgnu_la-unistd.o .libs/libcmd-certtool.lax/libcerttool-cfg.a/libgnu_la-xsize.o .libs/libcmd-certtool.lax/libcerttool-cfg.a/lt1-asnprintf.o .libs/libcmd-certtool.lax/libcerttool-cfg.a/lt2-printf-args.o .libs/libcmd-certtool.lax/libcerttool-cfg.a/lt3-printf-parse.o .libs/libcmd-certtool.lax/libcerttool-cfg.a/lt4-vasnprintf.o .libs/libcmd-certtool.lax/libcerttool-cfg.a/mktime.o .libs/libcmd-certtool.lax/libcerttool-cfg.a/printf-args.o .libs/libcmd-certtool.lax/libcerttool-cfg.a/printf-parse.o .libs/libcmd-certtool.lax/libcerttool-cfg.a/vasnprintf.o .libs/libcmd-certtool.lax/lt2-asnprintf.o .libs/libcmd-certtool.lax/lt3-libgnu_gpl_la-basename-lgpl.o .libs/libcmd-certtool.lax/lt4-libgnu_gpl_la-bitrotate.o .libs/libcmd-certtool.lax/lt5-libgnu_gpl_la-c-ctype.o .libs/libcmd-certtool.lax/lt6-libgnu_gpl_la-c-strcasecmp.o .libs/libcmd-certtool.lax/lt7-libgnu_gpl_la-c-strncasecmp.o .libs/libcmd-certtool.lax/lt8-libgnu_gpl_la-canonicalize-lgpl.o .libs/libcmd-certtool.lax/lt9-libgnu_gpl_la-cloexec.o .libs/libcmd-certtool.lax/lt10-libgnu_gpl_la-close-stream.o .libs/libcmd-certtool.lax/lt11-libgnu_gpl_la-error.o .libs/libcmd-certtool.lax/lt12-libgnu_gpl_la-exitfail.o .libs/libcmd-certtool.lax/lt13-libgnu_gpl_la-fcntl.o .libs/libcmd-certtool.lax/lt14-libgnu_gpl_la-fd-hook.o .libs/libcmd-certtool.lax/lt15-libgnu_gpl_la-free.o .libs/libcmd-certtool.lax/lt16-libgnu_gpl_la-fseek.o .libs/libcmd-certtool.lax/lt17-libgnu_gpl_la-fseeko.o .libs/libcmd-certtool.lax/lt18-libgnu_gpl_la-getdelim.o .libs/libcmd-certtool.lax/lt19-libgnu_gpl_la-getline.o .libs/libcmd-certtool.lax/lt20-libgnu_gpl_la-getprogname.o .libs/libcmd-certtool.lax/lt21-libgnu_gpl_la-gettime.o .libs/libcmd-certtool.lax/lt22-libgnu_gpl_la-gl_linked_list.o .libs/libcmd-certtool.lax/lt23-libgnu_gpl_la-gl_linkedhash_list.o .libs/libcmd-certtool.lax/lt24-libgnu_gpl_la-gl_list.o .libs/libcmd-certtool.lax/lt25-libgnu_gpl_la-gl_xlist.o .libs/libcmd-certtool.lax/lt26-libgnu_gpl_la-hash-pjw-bare.o .libs/libcmd-certtool.lax/lt27-libgnu_gpl_la-hash.o .libs/libcmd-certtool.lax/lt28-libgnu_gpl_la-ialloc.o .libs/libcmd-certtool.lax/lt29-libgnu_gpl_la-lock.o .libs/libcmd-certtool.lax/lt30-libgnu_gpl_la-malloca.o .libs/libcmd-certtool.lax/lt31-libgnu_gpl_la-memchr.o .libs/libcmd-certtool.lax/lt32-libgnu_gpl_la-memset_explicit.o .libs/libcmd-certtool.lax/lt33-libgnu_gpl_la-nstrftime.o .libs/libcmd-certtool.lax/lt34-libgnu_gpl_la-parse-datetime.o .libs/libcmd-certtool.lax/lt35-libgnu_gpl_la-progname.o .libs/libcmd-certtool.lax/lt36-libgnu_gpl_la-rawmemchr.o .libs/libcmd-certtool.lax/lt37-libgnu_gpl_la-read-file.o .libs/libcmd-certtool.lax/lt38-libgnu_gpl_la-scratch_buffer_grow.o .libs/libcmd-certtool.lax/lt39-libgnu_gpl_la-scratch_buffer_grow_preserve.o .libs/libcmd-certtool.lax/lt40-libgnu_gpl_la-scratch_buffer_set_array_size.o .libs/libcmd-certtool.lax/lt41-libgnu_gpl_la-sockets.o .libs/libcmd-certtool.lax/lt42-libgnu_gpl_la-stat-time.o .libs/libcmd-certtool.lax/lt43-libgnu_gpl_la-sys_socket.o .libs/libcmd-certtool.lax/lt44-libgnu_gpl_la-threadlib.o .libs/libcmd-certtool.lax/lt45-libgnu_gpl_la-time_rz.o .libs/libcmd-certtool.lax/lt46-libgnu_gpl_la-timegm.o .libs/libcmd-certtool.lax/lt47-libgnu_gpl_la-timespec.o .libs/libcmd-certtool.lax/lt48-libgnu_gpl_la-tls.o .libs/libcmd-certtool.lax/lt49-libgnu_gpl_la-unistd.o .libs/libcmd-certtool.lax/lt50-libgnu_gpl_la-xalloc-die.o .libs/libcmd-certtool.lax/lt51-libgnu_gpl_la-xmalloc.o .libs/libcmd-certtool.lax/lt52-libgnu_gpl_la-xsize.o .libs/libcmd-certtool.lax/lt53-mktime.o .libs/libcmd-certtool.lax/lt54-printf-args.o .libs/libcmd-certtool.lax/lt55-printf-parse.o .libs/libcmd-certtool.lax/lt56-vasnprintf.o .libs/libcmd-certtool.lax/lt57-asnprintf.o .libs/libcmd-certtool.lax/lt58-libgnu_la-basename-lgpl.o .libs/libcmd-certtool.lax/lt59-libgnu_la-bitrotate.o .libs/libcmd-certtool.lax/lt60-libgnu_la-c-ctype.o .libs/libcmd-certtool.lax/lt61-libgnu_la-c-strcasecmp.o .libs/libcmd-certtool.lax/lt62-libgnu_la-c-strncasecmp.o .libs/libcmd-certtool.lax/lt63-libgnu_la-canonicalize-lgpl.o .libs/libcmd-certtool.lax/lt64-libgnu_la-cloexec.o .libs/libcmd-certtool.lax/lt65-libgnu_la-dirname-lgpl.o .libs/libcmd-certtool.lax/lt66-libgnu_la-fcntl.o .libs/libcmd-certtool.lax/lt67-libgnu_la-fd-hook.o .libs/libcmd-certtool.lax/lt68-libgnu_la-free.o .libs/libcmd-certtool.lax/lt69-libgnu_la-getdelim.o .libs/libcmd-certtool.lax/lt70-libgnu_la-getline.o .libs/libcmd-certtool.lax/lt71-libgnu_la-gl_linkedhash_list.o .libs/libcmd-certtool.lax/lt72-libgnu_la-gl_list.o .libs/libcmd-certtool.lax/lt73-libgnu_la-hash-pjw-bare.o .libs/libcmd-certtool.lax/lt74-libgnu_la-hash.o .libs/libcmd-certtool.lax/lt75-libgnu_la-lock.o .libs/libcmd-certtool.lax/lt76-libgnu_la-malloca.o .libs/libcmd-certtool.lax/lt77-libgnu_la-memchr.o .libs/libcmd-certtool.lax/lt78-libgnu_la-memset_explicit.o .libs/libcmd-certtool.lax/lt79-libgnu_la-rawmemchr.o .libs/libcmd-certtool.lax/lt80-libgnu_la-read-file.o .libs/libcmd-certtool.lax/lt81-libgnu_la-scratch_buffer_grow.o .libs/libcmd-certtool.lax/lt82-libgnu_la-scratch_buffer_grow_preserve.o .libs/libcmd-certtool.lax/lt83-libgnu_la-scratch_buffer_set_array_size.o .libs/libcmd-certtool.lax/lt84-libgnu_la-stat-time.o .libs/libcmd-certtool.lax/lt85-libgnu_la-stripslash.o .libs/libcmd-certtool.lax/lt86-libgnu_la-sys_socket.o .libs/libcmd-certtool.lax/lt87-libgnu_la-threadlib.o .libs/libcmd-certtool.lax/lt88-libgnu_la-tls.o .libs/libcmd-certtool.lax/lt89-libgnu_la-unistd.o .libs/libcmd-certtool.lax/lt90-libgnu_la-xsize.o .libs/libcmd-certtool.lax/lt91-printf-args.o .libs/libcmd-certtool.lax/lt92-printf-parse.o .libs/libcmd-certtool.lax/lt93-vasnprintf.o  [... cut ...]
touch /home/audy/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.gnutls_installed
make[3]: Leaving directory '/home/audy/lede/feeds/packages/libs/gnutls'
time: package/feeds/packages/gnutls/compile#515.11#42.98#397.88
make[2]: Leaving directory '/home/audy/lede'
make[1]: *** [package/Makefile:110: /home/audy/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/audy/lede'
make: *** [/home/audy/lede/include/toplevel.mk:231: world] Error 2

lzf577, 2025-01-07:
是呀，我也想问，最新源码老是出错。

lzf577, 2025-01-07:
patchelf.h:226:10: error: 'void ElfFile<Elf_Ehdr, Elf_Phdr, Elf_Shdr, Elf_Addr, Elf_Off, Elf_Dyn, Elf_Sym, Elf_Versym, Elf_Verdef, Elf_Verdaux, Elf_Verneed, Elf_Vernaux, Elf_Rel, Elf_Rela, ElfClass>::changeRelocTableSymIds(const Elf_Shdr&, RemapFn&&) [with ElfRelType = Elf32_Rela; RemapFn = ElfFile<Elf_Ehdr, Elf_Phdr, Elf_Shdr, Elf_Addr, Elf_Off, Elf_Dyn, Elf_Sym, Elf_Versym, Elf_Verdef, Elf_Verdaux, Elf_Verneed, Elf_Vernaux, Elf_Rel, Elf_Rela, ElfClass>::rebuildGnuHashTable(span<char>, span<Elf_Sym>) [with Elf_Ehdr = Elf32_Ehdr; Elf_Phdr = Elf32_Phdr; Elf_Shdr = Elf32_Shdr; Elf_Addr = unsigned int; Elf_Off = unsigned int; Elf_Dyn = Elf32_Dyn; Elf_Sym = Elf32_Sym; Elf_Versym = short unsigned int; Elf_Verdef = Elf32_Verdef; Elf_Verdaux = Elf32_Verdaux; Elf_Verneed = Elf32_Verneed; Elf_Vernaux = Elf32_Vernaux; Elf_Rel = Elf32_Rel; Elf_Rela = Elf32_Rela; unsigned int ElfClass = 32]::<lambda(auto:7&)>&; Elf_Ehdr = Elf32_Ehdr; Elf_Phdr = Elf32_Phdr; Elf_Shdr = Elf32_Shdr; Elf_Addr = unsigned int; Elf_Off = unsigned int; Elf_Dyn = Elf32_Dyn; Elf_Sym = Elf32_Sym; Elf_Versym = short unsigned int; Elf_Verdef = Elf32_Verdef; Elf_Verdaux = Elf32_Verdaux; Elf_Verneed = Elf32_Verneed; Elf_Vernaux = Elf32_Vernaux; Elf_Rel = Elf32_Rel; Elf_Rela = Elf32_Rela; unsigned int ElfClass = 32]', declared using local type 'ElfFile<Elf_Ehdr, Elf_Phdr, Elf_Shdr, Elf_Addr, Elf_Off, Elf_Dyn, Elf_Sym, Elf_Versym, Elf_Verdef, Elf_Verdaux, Elf_Verneed, Elf_Vernaux, Elf_Rel, Elf_Rela, ElfClass>::rebuildGnuHashTable(span<char>, span<Elf_Sym>) [with Elf_Ehdr = Elf32_Ehdr; Elf_Phdr = Elf32_Phdr; Elf_Shdr = Elf32_Shdr; Elf_Addr = unsigned int; Elf_Off = unsigned int; Elf_Dyn = Elf32_Dyn; Elf_Sym = Elf32_Sym; Elf_Versym = short unsigned int; Elf_Verdef = Elf32_Verdef; Elf_Verdaux = Elf32_Verdaux; Elf_Verneed = Elf32_Verneed; Elf_Vernaux = Elf32_Vernaux; Elf_Rel = Elf32_Rel; Elf_Rela = Elf32_Rela; unsigned int ElfClass = 32]::<lambda(auto:7&)>', is used but never defined [-fpermissive]
patchelf.h:226:10: error: 'void ElfFile<Elf_Ehdr, Elf_Phdr, Elf_Shdr, Elf_Addr, Elf_Off, Elf_Dyn, Elf_Sym, Elf_Versym, Elf_Verdef, Elf_Verdaux, Elf_Verneed, Elf_Vernaux, Elf_Rel, Elf_Rela, ElfClass>::changeRelocTableSymIds(const Elf_Shdr&, RemapFn&&) [with ElfRelType = Elf64_Rel; RemapFn = ElfFile<Elf_Ehdr, Elf_Phdr, Elf_Shdr, Elf_Addr, Elf_Off, Elf_Dyn, Elf_Sym, Elf_Versym, Elf_Verdef, Elf_Verdaux, Elf_Verneed, Elf_Vernaux, Elf_Rel, Elf_Rela, ElfClass>::rebuildGnuHashTable(span<char>, span<Elf_Sym>) [with Elf_Ehdr = Elf64_Ehdr; Elf_Phdr = Elf64_Phdr; Elf_Shdr = Elf64_Shdr; Elf_Addr = long unsigned int; Elf_Off = long unsigned int; Elf_Dyn = Elf64_Dyn; Elf_Sym = Elf64_Sym; Elf_Versym = short unsigned int; Elf_Verdef = Elf64_Verdef; Elf_Verdaux = Elf64_Verdaux; Elf_Verneed = Elf64_Verneed; Elf_Vernaux = Elf64_Vernaux; Elf_Rel = Elf64_Rel; Elf_Rela = Elf64_Rela; unsigned int ElfClass = 64]::<lambda(auto:7&)>&; Elf_Ehdr = Elf64_Ehdr; Elf_Phdr = Elf64_Phdr; Elf_Shdr = Elf64_Shdr; Elf_Addr = long unsigned int; Elf_Off = long unsigned int; Elf_Dyn = Elf64_Dyn; Elf_Sym = Elf64_Sym; Elf_Versym = short unsigned int; Elf_Verdef = Elf64_Verdef; Elf_Verdaux = Elf64_Verdaux; Elf_Verneed = Elf64_Verneed; Elf_Vernaux = Elf64_Vernaux; Elf_Rel = Elf64_Rel; Elf_Rela = Elf64_Rela; unsigned int ElfClass = 64]', declared using local type 'ElfFile<Elf_Ehdr, Elf_Phdr, Elf_Shdr, Elf_Addr, Elf_Off, Elf_Dyn, Elf_Sym, Elf_Versym, Elf_Verdef, Elf_Verdaux, Elf_Verneed, Elf_Vernaux, Elf_Rel, Elf_Rela, ElfClass>::rebuildGnuHashTable(span<char>, span<Elf_Sym>) [with Elf_Ehdr = Elf64_Ehdr; Elf_Phdr = Elf64_Phdr; Elf_Shdr = Elf64_Shdr; Elf_Addr = long unsigned int; Elf_Off = long unsigned int; Elf_Dyn = Elf64_Dyn; Elf_Sym = Elf64_Sym; Elf_Versym = short unsigned int; Elf_Verdef = Elf64_Verdef; Elf_Verdaux = Elf64_Verdaux; Elf_Verneed = Elf64_Verneed; Elf_Vernaux = Elf64_Vernaux; Elf_Rel = Elf64_Rel; Elf_Rela = Elf64_Rela; unsigned int ElfClass = 64]::<lambda(auto:7&)>', is used but never defined [-fpermissive]
patchelf.h:226:10: error: 'void ElfFile<Elf_Ehdr, Elf_Phdr, Elf_Shdr, Elf_Addr, Elf_Off, Elf_Dyn, Elf_Sym, Elf_Versym, Elf_Verdef, Elf_Verdaux, Elf_Verneed, Elf_Vernaux, Elf_Rel, Elf_Rela, ElfClass>::changeRelocTableSymIds(const Elf_Shdr&, RemapFn&&) [with ElfRelType = Elf64_Rela; RemapFn = ElfFile<Elf_Ehdr, Elf_Phdr, Elf_Shdr, Elf_Addr, Elf_Off, Elf_Dyn, Elf_Sym, Elf_Versym, Elf_Verdef, Elf_Verdaux, Elf_Verneed, Elf_Vernaux, Elf_Rel, Elf_Rela, ElfClass>::rebuildGnuHashTable(span<char>, span<Elf_Sym>) [with Elf_Ehdr = Elf64_Ehdr; Elf_Phdr = Elf64_Phdr; Elf_Shdr = Elf64_Shdr; Elf_Addr = long unsigned int; Elf_Off = long unsigned int; Elf_Dyn = Elf64_Dyn; Elf_Sym = Elf64_Sym; Elf_Versym = short unsigned int; Elf_Verdef = Elf64_Verdef; Elf_Verdaux = Elf64_Verdaux; Elf_Verneed = Elf64_Verneed; Elf_Vernaux = Elf64_Vernaux; Elf_Rel = Elf64_Rel; Elf_Rela = Elf64_Rela; unsigned int ElfClass = 64]::<lambda(auto:7&)>&; Elf_Ehdr = Elf64_Ehdr; Elf_Phdr = Elf64_Phdr; Elf_Shdr = Elf64_Shdr; Elf_Addr = long unsigned int; Elf_Off = long unsigned int; Elf_Dyn = Elf64_Dyn; Elf_Sym = Elf64_Sym; Elf_Versym = short unsigned int; Elf_Verdef = Elf64_Verdef; Elf_Verdaux = Elf64_Verdaux; Elf_Verneed = Elf64_Verneed; Elf_Vernaux = Elf64_Vernaux; Elf_Rel = Elf64_Rel; Elf_Rela = Elf64_Rela; unsigned int ElfClass = 64]', declared using local type 'ElfFile<Elf_Ehdr, Elf_Phdr, Elf_Shdr, Elf_Addr, Elf_Off, Elf_Dyn, Elf_Sym, Elf_Versym, Elf_Verdef, Elf_Verdaux, Elf_Verneed, Elf_Vernaux, Elf_Rel, Elf_Rela, ElfClas [... cut ...]
make[2]: Leaving directory '/home/lzf/lede'
tools/Makefile:227: recipe for target '/home/lzf/lede/staging_dir/host/stamp/.tools_compile_nyyynyynnnyyynynyyyyyyynyyynynyynnynnyyyyyynyyyyyyyyyyynynnyyyyynyyy' failed
make[1]: *** [/home/lzf/lede/staging_dir/host/stamp/.tools_compile_nyyynyynnnyyynynyyyyyyynyyynynyynnynnyyyyyynyyyyyyyyyyynynnyyyyynyyy] Error 2
make[1]: Leaving directory '/home/lzf/lede'
/home/lzf/lede/include/toplevel.mk:229: recipe for target 'world' failed
make: *** [world] Error 2

closed 2025-01-12 by audyfan