coolsnowwolf/lede #13261: 3588aib现在所有编译的op都不能编辑wifi

JDKCHINA, 2025-01-09:
### 详细叙述

现在无法编辑wifi换了很多网卡都不行，无法进入编辑页面
[image]
[image]
[image]
麻烦开发者修复一下这个问题板子为rk3588aib,谢谢

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

rk3588aib

### 详细日志

TypeError
Cannot read properties of undefined (reading 'push')

aiamadeus, 2025-01-09:
换回旧 luci 就可以了
```
--- a/feeds.conf.default
+++ b/feeds.conf.default
@@ -1,10 +1,9 @@
 src-git packages https://github.com/coolsnowwolf/packages
-#src-git luci https://github.com/coolsnowwolf/luci
-src-git luci https://github.com/coolsnowwolf/luci.git;openwrt-23.05
+src-git luci https://github.com/coolsnowwolf/luci
+#src-git luci https://github.com/coolsnowwolf/luci.git;openwrt-23.05
 src-git routing https://github.com/coolsnowwolf/routing
-src-git telephony https://github.com/coolsnowwolf/telephony.git
-#src-git helloworld https://github.com/fw876/helloworld.git
-#src-git oui https://github.com/zhaojh329/oui.git
+src-git telephony https://github.com/coolsnowwolf/telephony.git
+src-git helloworld https://github.com/fw876/helloworld.git
 #src-git video https://github.com/openwrt/video.git
 #src-git targets https://github.com/openwrt/targets.git
 #src-git oldpackages http://git.openwrt.org/packages.git
```

JDKCHINA, 2025-01-09:
大大，旧的luci版本很多插件不支持呀，本来还支持mt7922,现在都不支持了，麻烦能修一下嘛

wherss1, 2025-01-16:
7922和nfa765在x86平台，也这样，换了旧版luci,编译报错，移植imm23.05的luci，7922和nfa765无线还是跟博主显示的报错一样，希望lede大佬能修复一下