coolsnowwolf/lede #13266: 编译lede固件修改后台地址编译不生效

hztbhtbh, 2025-01-11:
### 详细叙述

使用命令：vim package/base-files/files/bin/config_generate  修改IP地址为192.168.5.5编译头不生效还是默认的IP地址192.168.1.1
我拉取最新源码然后git checkout编译了以下内核5.4.288  5.15.175   6.1.120都是不生效，其它没有问题，请大佬解答，谢谢！
[image]

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

J4125软路由

### 详细日志

没有提供的日志

binge8, 2025-01-12:
[善于搜索](https://github.com/coolsnowwolf/lede/issues/13128)

hztbhtbh, 2025-01-12:
> [善于搜索](https://github.com/coolsnowwolf/lede/issues/13128) 点进去看看就知道了

感谢大佬回复，谢谢

4thanks, 2025-01-12:
抄V老的脚本，大概是这3个位置
```
#修改默认主题
$(find ./feeds/luci/collections/ -type f -name "Makefile")

#修改关联IP
$(find ./feeds/luci/modules/luci-mod-system/ -type f -name "flash.js")

#修改默认IP地址和默认主机名
package/base-files/luci2/bin/config_generate
```

xzchsia, 2025-02-13:
原来是换路径了。我fork的别人的编译环境一直无法修改ip和hostname。看GitHub里面的查询都还是老的方案。一会修改编译环境试试看新的路径是否可以。

hztbhtbh, 2025-06-26:
#更改LAN口的默认IP地址：
cd lede
vim package/base-files/files/bin/config_generate          #旧版luci 18.06
vim package/base-files/luci2/bin/config_generate         #新版luci 23.05

coolsnowwolf, 2026-09-05:
当前 master 已验证解决。现行 LuCI feed 带 `package.json`，`package/base-files/Makefile` 不再用旧 LuCI 专用目录覆盖脚本；最终 x86 rootfs 的 `/bin/config_generate` 与 `package/base-files/files/bin/config_generate` SHA-256 完全一致。\n\n因此现在修改 `package/base-files/files/bin/config_generate` 中 LAN 默认地址会进入固件。旧版 23.05 曾使用的 `luci2`/`luci` 路径仅适用于当时分支。按当前代码已解决关闭。

closed 2026-09-05 by coolsnowwolf