coolsnowwolf/lede #13270: mt7925的蓝牙问题

IDeLoveYou, 2025-01-12:
### 详细叙述

我使用最新的源码编译了6.12.8内核的openwrt，并使用以下官方的教程配置蓝牙
https://openwrt.org/docs/guide-user/hardware/bluetooth/bluetooth.speakers
但是，我在使用蓝牙时没有识别到设备

```
root@LEDE:~# bluetoothctl
Agent registered
[bluetooth]# scan on
No default controller available
```

以下是一些日志输出

```
root@LEDE:~# lsmod | grep btusb
bluetooth             438272 19 rfcomm,hidp,hci_uart,btusb,btrtl,btqca,btmtk,btintel,btbcm,bnep,bluetooth_6lowpan
btbcm                  16384  2 hci_uart,btusb
btintel                32768  1 btusb
btmtk                  20480  1 btusb
btrtl                  20480  1 btusb
btusb                  45056  0 
root@LEDE:~# dmesg | grep -i firmware
[    0.000000] psci: PSCIv1.1 detected in firmware.
[    0.643654] arm-scmi arm-scmi.0.auto: SCMI Protocol v2.0 'rockchip:' Firmware version 0x0
[    8.842199] mt7925e 0000:01:00.0: WM Firmware Version: ____000000, Build Time: 20231227093232
root@LEDE:~# dmesg | grep -i bluetooth
[    7.406413] Bluetooth: Core ver 2.22
[    7.406846] NET: Registered PF_BLUETOOTH protocol family
[    7.407354] Bluetooth: HCI device and connection manager initialized
[    7.407940] Bluetooth: HCI socket layer initialized
[    7.408382] Bluetooth: L2CAP socket layer initialized
[    7.408848] Bluetooth: SCO socket layer initialized
[    7.411543] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[    7.412039] Bluetooth: BNEP filters: protocol multicast
[    7.412515] Bluetooth: BNEP socket layer initialized
[    7.427671] Bluetooth: HCI UART driver ver 2.3
[    7.428094] Bluetooth: HCI UART protocol H4 registered
[    7.428550] Bluetooth: HCI UART protocol BCSP registered
[    7.429246] Bluetooth: HCI UART protocol Broadcom registered
[    7.429803] Bluetooth: HCI UART protocol QCA registered
[    7.431322] Bluetooth: HIDP (Human Interface Emulation) ver 1.2
[    7.431889] Bluetooth: HIDP socket layer initialized
[    8.252311] Bluetooth: RFCOMM TTY layer initialized
[    8.252787] Bluetooth: RFCOMM socket layer initialized
[    8.253275] Bluetooth: RFCOMM ver 1.11
root@LEDE:~# lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux 6.12.8 xhci-hcd xHCI Host Controller
Bus 004 Device 001: ID 1d6b:0002 Linux 6.12.8 ehci_hcd EHCI Host Controller
Bus 006 Device 001: ID 1d6b:0002 Linux 6.12.8 ehci_hcd EHCI Host Controller
Bus 001 Device 001: ID 1d6b:0002 Linux 6.12.8 xhci-hcd xHCI Host Controller
Bus 003 Device 001: ID 1d6b:0001 Linux 6.12.8 ohci_hcd Generic Platform OHCI controller
Bus 005 Device 001: ID 1d6b:0001 Linux 6.12.8 ohci_hcd Generic Platform OHCI controller
root@LEDE:~# hciconfig -a
root@LEDE:~# 

```

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

hinlink-h68k（rockchip-rk3568）

### 详细日志

null

aiamadeus, 2025-01-12:
lsusb 看看

IDeLoveYou, 2025-01-12:
root@LEDE:~# lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux 6.12.8 xhci-hcd xHCI Host Controller
Bus 004 Device 001: ID 1d6b:0002 Linux 6.12.8 ehci_hcd EHCI Host Controller
Bus 006 Device 001: ID 1d6b:0002 Linux 6.12.8 ehci_hcd EHCI Host Controller
Bus 001 Device 001: ID 1d6b:0002 Linux 6.12.8 xhci-hcd xHCI Host Controller
Bus 003 Device 001: ID 1d6b:0001 Linux 6.12.8 ohci_hcd Generic Platform OHCI controller
Bus 005 Device 001: ID 1d6b:0001 Linux 6.12.8 ohci_hcd Generic Platform OHCI controller
root@LEDE:~#

IDeLoveYou, 2025-01-12:
```
root@LEDE:~# lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux 6.12.8 xhci-hcd xHCI Host Controller
Bus 004 Device 001: ID 1d6b:0002 Linux 6.12.8 ehci_hcd EHCI Host Controller
Bus 006 Device 001: ID 1d6b:0002 Linux 6.12.8 ehci_hcd EHCI Host Controller
Bus 001 Device 001: ID 1d6b:0002 Linux 6.12.8 xhci-hcd xHCI Host Controller
Bus 003 Device 001: ID 1d6b:0001 Linux 6.12.8 ohci_hcd Generic Platform OHCI controller
Bus 005 Device 001: ID 1d6b:0001 Linux 6.12.8 ohci_hcd Generic Platform OHCI controller
root@LEDE:~#
```

IDeLoveYou, 2025-01-12:
兄弟，你头像好熟悉啊，pgyvpn是不是你做的

aiamadeus, 2025-01-12:
`cat /sys/kernel/debug/usb/devices` 看看

aiamadeus, 2025-01-12:
或者换其它网卡能认到 usb 吗（lsusb）？

IDeLoveYou, 2025-01-12:
```
root@LEDE:~# cat /sys/kernel/debug/usb/devices

T:  Bus=01 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=480  MxCh= 1
B:  Alloc=  0/800 us ( 0%), #Int=  0, #Iso=  0
D:  Ver= 2.00 Cls=09(hub  ) Sub=00 Prot=01 MxPS=64 #Cfgs=  1
P:  Vendor=1d6b ProdID=0002 Rev= 6.12
S:  Manufacturer=Linux 6.12.8 xhci-hcd
S:  Product=xHCI Host Controller
S:  SerialNumber=xhci-hcd.1.auto
C:* #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=  0mA
I:* If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub
E:  Ad=81(I) Atr=03(Int.) MxPS=   4 Ivl=256ms

T:  Bus=02 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=5000 MxCh= 1
B:  Alloc=  0/800 us ( 0%), #Int=  0, #Iso=  0
D:  Ver= 3.00 Cls=09(hub  ) Sub=00 Prot=03 MxPS= 9 #Cfgs=  1
P:  Vendor=1d6b ProdID=0003 Rev= 6.12
S:  Manufacturer=Linux 6.12.8 xhci-hcd
S:  Product=xHCI Host Controller
S:  SerialNumber=xhci-hcd.1.auto
C:* #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=  0mA
I:* If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub
E:  Ad=81(I) Atr=03(Int.) MxPS=   4 Ivl=256ms

T:  Bus=03 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=12   MxCh= 1
B:  Alloc=  0/900 us ( 0%), #Int=  0, #Iso=  0
D:  Ver= 1.10 Cls=09(hub  ) Sub=00 Prot=00 MxPS=64 #Cfgs=  1
P:  Vendor=1d6b ProdID=0001 Rev= 6.12
S:  Manufacturer=Linux 6.12.8 ohci_hcd
S:  Product=Generic Platform OHCI controller
S:  SerialNumber=fd840000.usb
C:* #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=  0mA
I:* If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub
E:  Ad=81(I) Atr=03(Int.) MxPS=   2 Ivl=255ms

T:  Bus=04 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=480  MxCh= 1
B:  Alloc=  0/800 us ( 0%), #Int=  0, #Iso=  0
D:  Ver= 2.00 Cls=09(hub  ) Sub=00 Prot=00 MxPS=64 #Cfgs=  1
P:  Vendor=1d6b ProdID=0002 Rev= 6.12
S:  Manufacturer=Linux 6.12.8 ehci_hcd
S:  Product=EHCI Host Controller
S:  SerialNumber=fd880000.usb
C:* #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=  0mA
I:* If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub
E:  Ad=81(I) Atr=03(Int.) MxPS=   4 Ivl=256ms

T:  Bus=05 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=12   MxCh= 1
B:  Alloc=  0/900 us ( 0%), #Int=  0, #Iso=  0
D:  Ver= 1.10 Cls=09(hub  ) Sub=00 Prot=00 MxPS=64 #Cfgs=  1
P:  Vendor=1d6b ProdID=0001 Rev= 6.12
S:  Manufacturer=Linux 6.12.8 ohci_hcd
S:  Product=Generic Platform OHCI controller
S:  SerialNumber=fd8c0000.usb
C:* #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=  0mA
I:* If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub
E:  Ad=81(I) Atr=03(Int.) MxPS=   2 Ivl=255ms

T:  Bus=06 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=480  MxCh= 1
B:  Alloc=  0/800 us ( 0%), #Int=  0, #Iso=  0
D:  Ver= 2.00 Cls=09(hub  ) Sub=00 Prot=00 MxPS=64 #Cfgs=  1
P:  Vendor=1d6b ProdID=0002 Rev= 6.12
S:  Manufacturer=Linux 6.12.8 ehci_hcd
S:  Product=EHCI Host Controller
S:  SerialNumber=fd800000.usb
C:* #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=  0mA
I:* If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub
E:  Ad=81(I) Atr=03(Int.) MxPS=   4 Ivl=256ms
root@LEDE:~#
```
我以前使用ax210的时候是可以的，但是换了mt7925就不行了，前阵子看commit说支持了蓝牙，就打算编译一个试试，然后就这情况

IDeLoveYou, 2025-01-12:
无线功能是没问题的，就是蓝牙好像有点问题，我是少编译什么了吗

aiamadeus, 2025-01-12:
完全没看到蓝牙 usb 出现，7925 插到电脑上有 usb 吗？

IDeLoveYou, 2025-01-12:
有的

IDeLoveYou, 2025-01-12:
win11可以识别蓝牙

aiamadeus, 2025-01-12:
你先换其它网卡试试有没有 usb 出现（排除固件问题）
其它网卡正常就可能是硬件兼容问题
得去问鸭王了（卖家）

IDeLoveYou, 2025-01-12:
好的，我试试

IDeLoveYou, 2025-01-12:
固件应该没问题，ax210是可以支持的，硬件兼容问题感觉难搞了，鸭子应该不负责自己改网卡

aiamadeus, 2025-01-12:
我记得旧版 h66k/h68k WiFi 槽没有接 usb 的，新版才有，我手里的还是旧版本。

IDeLoveYou, 2025-01-12:
我的是新版的，但我用的不是usb接口的网卡，用的pcie的

aiamadeus, 2025-01-12:
WiFi 走 pcie，蓝牙走 usb 啊…

IDeLoveYou, 2025-01-12:
新版是四个网口都在前面部分的吧，我的就是这样的，那按道理应该也支持蓝牙吧

aiamadeus, 2025-01-12:
其它网卡能认到 usb 蓝牙就支持，硬件 issue 你自己关了吧

IDeLoveYou, 2025-01-12:
@aiamadeus 哥们谢了，我找到原因了，不仅需要编译PACKAGE_kmod-mt7925e还需要加上PACKAGE_mt7925bt-firmware

closed 2025-01-12 by IDeLoveYou

aiamadeus, 2025-01-12:
啊？mt7925 蓝牙不是走 usb 的？

IDeLoveYou, 2025-01-16:
这个我还真不清楚，但是装了PACKAGE_mt7925bt-firmware 就可以用了