coolsnowwolf/lede #13275: ssr这个依赖冲突怎么搞

binge8, 2025-01-14:
### 详细叙述

编译报错，依赖冲突

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x64，armv8

### 详细日志

```bash
* check_data_file_clashes: Package libustream-openssl20201210 wants to install file /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/root-x86/lib/libustream-ssl.so
	But that file is already provided by package  * libustream-mbedtls20201210
* opkg_install_cmd: Cannot install package libustream-openssl20201210.
 * check_data_file_clashes: Package libustream-openssl20201210 wants to install file /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/root-x86/lib/libustream-ssl.so
	But that file is already provided by package  * libustream-mbedtls20201210
* opkg_install_cmd: Cannot install package luci-app-ssr-plus.
* resolve_conffiles: Existing conffile /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/root-x86/etc/ppp/options is different from the conffile in the new package. The new conffile will be placed at /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/root-x86/etc/ppp/options-opkg.
```

yanxin152133, 2025-01-15:
https://www.google.com.hk/search?q=But+that+file+is+already+provided+by+package+*+libustream-mbedtls20201210&sourceid=chrome&ie=UTF-8

binge8, 2025-01-15:
> https://www.google.com.hk/search?q=But+that+file+is+already+provided+by+package+*+libustream-mbedtls20201210&sourceid=chrome&ie=UTF-8

看了看对我等小白好像有点深奥

yanxin152133, 2025-01-15:
> > https://www.google.com.hk/search?q=But+that+file+is+already+provided+by+package+*+libustream-mbedtls20201210&sourceid=chrome&ie=UTF-8
> 
> 看了看对我等小白好像有点深奥

这就是你之前骂别人能力低素质低的底气啊

binge8, 2025-01-15:
> > > https://www.google.com.hk/search?q=But+that+file+is+already+provided+by+package+*+libustream-mbedtls20201210&sourceid=chrome&ie=UTF-8
> > 
> > 
> > 看了看对我等小白好像有点深奥
> 
> 这就是你之前骂别人能力低素质低的底气啊

你个逗比，没完了是吧，首先我学那位没素质的人的口气回复你：“发什么链什么接，会就说不会别发”。。。换了你你这么帮忙找思路，不管对不对，求助者这种口气回复你，你就这么认怂吗？你不会上火？其次，最后问题也不是他解决的，分析这分析那的还，要素质没素质，要技术没技术，啥也不是

yanxin152133, 2025-01-15:
> 你个逗比，没完了是吧，首先我学那位没素质的人的口气回复你：“发什么链什么接，会就说不会别发”。。。换了你你这么帮忙找思路，不管对不对，求助者这种口气回复你，你就这么认怂吗？你不会上火？其次，最后问题也不是他解决的，分析这分析那的还，要素质没素质，要技术没技术，啥也不是

所以你还是连个百度谷歌搜索关键词都不会还这么装

binge8, 2025-01-15:
> 所以你还是连个百度谷歌搜索关键词都不会还这么装

所以我装你爹了还是装你妈了？没事找事？关键词搜了我看不懂来这里问一问有什么问题吗？陈芝麻烂谷子的事都还记仇拉出来找骂？跟个娘们一样的还打抱不平呢？你俩可真是物以类聚，我继续学那位的口气就回复你，想帮就帮不帮绕道，贱嗖嗖的进来记仇，真贱

closed 2025-01-15 by binge8

binge8, 2025-01-17:
> 所以你还是连个百度谷歌搜索关键词都不会还这么装

你还真是个小乐色，搜个关键词就跑我这来装逼来了？你要会的话还用搜关键词吗？就我这样的业余玩家都能分析解决，你还搜个关键词来我这装逼。真是跟他一路货色，技术半吊子，还娘们兮兮记仇拉仇恨，对你这种没技术也就罢了，还没素质，不好好说话来找骂，不骂白不骂

yanxin152133, 2025-01-17:
> > 所以你还是连个百度谷歌搜索关键词都不会还这么装
> 
> 你还真是个小乐色，搜个关键词就跑我这来装逼来了？你要会的话还用搜关键词吗？就我这样的业余玩家都能分析解决，你还搜个关键词来我这装逼。真是跟他一路货色，技术半吊子，还娘们兮兮记仇拉仇恨，对你这种没技术也就罢了，还没素质，不好好说话来找骂，不骂白不骂

你看不到google标红的部分？一个个点开看看有多少处理意见。还有没素质的人是你，自己不会点开看你就是没脑子，只有泼妇才像你这样张口闭口带脏字，建议你去当人妖。
[image]

binge8, 2025-01-17:
> > > 所以你还是连个百度谷歌搜索关键词都不会还这么装
> > 
> > 
> > 你还真是个小乐色，搜个关键词就跑我这来装逼来了？你要会的话还用搜关键词吗？就我这样的业余玩家都能分析解决，你还搜个关键词来我这装逼。真是跟他一路货色，技术半吊子，还娘们兮兮记仇拉仇恨，对你这种没技术也就罢了，还没素质，不好好说话来找骂，不骂白不骂
> 
> 你看不到google标红的部分？一个个点开看看有多少处理意见。还有没素质的人是你，自己不会点开看你就是没脑子，只有泼妇才像你这样张口闭口带脏字，建议你去当人妖。 [image]

帮就帮，不帮就绕远，过去多久的事了娘们兮兮进来记仇拉仇恨，你还真懂人妖，泼妇生的娘们兮兮的人妖最适合你。那么多标红有一个适合咱这种情况的吗？况且就我这种业余选手都连搜都不用搜就能解决的问题

binge8, 2025-01-17:
进来就不好好说话，犯贱找骂