coolsnowwolf/lede #13276: IPQ50XX-jdcloud_re-cs-03编译qca-nss-ecm报错

oppen321, 2025-01-15:
### 详细叙述

qca-nss-ecm，现在IPQ50XX支持NSS吗、不是不支持吗

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

jdcloud_re-cs-03、linksys_mx2000、linksys_mx5500

### 详细日志

cc1: all warnings being treated as errors
make[5]: *** [scripts/Makefile.build:250: /home/runner/work/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq50xx/qca-nss-ecm-2023-10-20-82b27915/ecm_conntrack_notifier.o] Error 1
make[4]: *** [Makefile:2009: /home/runner/work/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq50xx/qca-nss-ecm-2023-10-20-82b27915] Error 2
make[4]: Leaving directory '/home/runner/work/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq50xx/linux-6.1.124'
make[3]: *** [Makefile:164: /home/runner/work/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq50xx/qca-nss-ecm-2023-10-20-82b27915/.built] Error 2
make[3]: Leaving directory '/home/runner/work/openwrt/package/qca/qca-nss-ecm'
time: package/qca/qca-nss-ecm/compile#1.09#0.25#1.31
    ERROR: package/qca/qca-nss-ecm failed to build.
make[2]: *** [package/Makefile:116: package/qca/qca-nss-ecm/compile] Error 1
make[2]: Leaving directory '/home/runner/work/openwrt'
make[1]: *** [package/Makefile:110: /home/runner/work/openwrt/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/runner/work/openwrt'
make: *** [/home/runner/work/openwrt/include/toplevel.mk:231: world] Error 2

snowrice, 2025-01-17:
我紅米ax6編譯也有這個報錯

aiamadeus, 2025-01-18:
[image]
能编过啊

sijyun, 2025-02-16:
rax3000q 完成編譯.

turboacc 使用預設 fastpath (disable)

lurenjia1213, 2025-02-20:
你能跑起来aarch64内核吗？能跑，可否发一下你的MIBIB，QSEE，APPSBL？

oppen321, 2025-02-21:
目前已经可以成功编译通过了

closed 2025-02-21 by oppen321

Mrdeng233, 2025-04-21:
> rax3000q 完成編譯.
> 
> turboacc 使用預設 FastPath （disable）

请问你的RAX3000QY能刷入自编译的固件吗，我编译出来没有任何报错但从uboot刷入后无法开机。