coolsnowwolf/lede #13279: 日志 EDID block 0 is all zeroes

fma89, 2025-01-16:
### 详细叙述

物理机安装自编译的L大的Lede，出现如下报错，基本十秒就新增一条记录，请问如何修复或屏蔽该信息？

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86-J1900

### 详细日志

[image]
[image]
[image]

wushenghn, 2025-01-17:
linux内核的问题等同步内核补丁

coolsnowwolf, 2025-01-22:
marked. /drivers/gpu/drm/drm_edid.c

coolsnowwolf, 2026-09-05:
已修复并推送：7aefa9a320346f700046c085cc1d99fad8799d7b\n\n将 DRM 的 `EDID block N is all zeroes` 从 notice 降为 debug，避免显示器/转接器返回全零 EDID 时被连接器轮询每约 10 秒刷屏；其他 EDID 损坏提示保持不变。\n\n验证：\n- 6.12.108 x86 内核 `-j30` 完整编译通过；\n- 实际构建源码确认该路径为 `pr_debug()`；\n- 同步提供 6.18 补丁，并在 6.18.49 源码 dry-run 通过；\n- 两个补丁均以临时文件 + 无时间戳 a/b diff 生成。

closed 2026-09-05 by coolsnowwolf