coolsnowwolf/lede #13287: Please remove that entry from $PATH :The relative path 'Files/Common'

tuzen2018, 2025-01-17:
### 详细叙述

sed -i "s/Installed-Time: .*/Installed-Time: 1737028921/" /home/tuzen/lede/build_dir/target-x86_64_musl/root-x86/usr/lib/opkg/status
rm -rf /home/tuzen/lede/build_dir/target-x86_64_musl/root-x86/boot /home/tuzen/lede/build_dir/target-x86_64_musl/root-x86/tmp/* /home/tuzen/lede/build_dir/target-x86_64_musl/root-x86/usr/lib/opkg/info/*.postinst* /home/tuzen/lede/build_dir/target-x86_64_musl/root-x86/usr/lib/opkg/lists/* /home/tuzen/lede/build_dir/target-x86_64_musl/root-x86/var/lock/*.lock
find /home/tuzen/lede/build_dir/target-x86_64_musl/root-x86/ -mindepth 1 -execdir touch -hcd "@1737028921" "{}" +
find: The relative path 'Files/Common' is included in the PATH environment variable, which is insecure in combination with the -execdir action of find.  Please remove that entry from $PATH
make[2]: *** [package/Makefile:73: package/install] Error 1
make[2]: Leaving directory '/home/tuzen/lede'
make[1]: *** [package/Makefile:111: /home/tuzen/lede/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/tuzen/lede'
make: *** [/home/tuzen/lede/include/toplevel.mk:231: world] Error 2

ubuntu22.4下编译，x86 ，.config是默认的，重装了几个系统，一直报这个错，没办法了，请教一下各位大神， 不清楚在哪个文件里删除PATH

### 重复 issue

- [x] 没有类似的 issue

### 具体型号

x86

### 详细日志

sed -i "s/Installed-Time: .*/Installed-Time: 1737028921/" /home/tuzen/lede/build_dir/target-x86_64_musl/root-x86/usr/lib/opkg/status
rm -rf /home/tuzen/lede/build_dir/target-x86_64_musl/root-x86/boot /home/tuzen/lede/build_dir/target-x86_64_musl/root-x86/tmp/* /home/tuzen/lede/build_dir/target-x86_64_musl/root-x86/usr/lib/opkg/info/*.postinst* /home/tuzen/lede/build_dir/target-x86_64_musl/root-x86/usr/lib/opkg/lists/* /home/tuzen/lede/build_dir/target-x86_64_musl/root-x86/var/lock/*.lock
find /home/tuzen/lede/build_dir/target-x86_64_musl/root-x86/ -mindepth 1 -execdir touch -hcd "@1737028921" "{}" +
find: The relative path 'Files/Common' is included in the PATH environment variable, which is insecure in combination with the -execdir action of find.  Please remove that entry from $PATH
make[2]: *** [package/Makefile:73: package/install] Error 1
make[2]: Leaving directory '/home/tuzen/lede'
make[1]: *** [package/Makefile:111: /home/tuzen/lede/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/tuzen/lede'
make: *** [/home/tuzen/lede/include/toplevel.mk:231: world] Error 2

yhpgi, 2025-02-09:
It looks like you are trying to compile it in WSL.
[this](https://openwrt.org/docs/guide-developer/toolchain/wsl) page may help you in this case.

yhpgi, 2025-02-09:
It looks like you are trying to compile it in WSL.
If that is the case, please read the [README in this repo](https://github.com/coolsnowwolf/lede?tab=readme-ov-file#%E4%BD%BF%E7%94%A8-wslwsl2-%E8%BF%9B%E8%A1%8C%E7%BC%96%E8%AF%91). there is a guide to do the workaround. or, read from [here](https://openwrt.org/docs/guide-developer/toolchain/wsl).

coolsnowwolf, 2026-09-05:
日志中的 `Files/Common` 来自 WSL 自动注入的 Windows PATH，并非 LEDE Makefile。仓库 README 的“使用 WSL/WSL2 进行编译”章节已经给出处理方式：在 make 前执行 `PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin`。已有明确解决方案，关闭 issue。

closed 2026-09-05 by coolsnowwolf