coolsnowwolf/lede #13301: 编译报错，看不出啥原因，求助。。。

drw2005, 2025-01-21:
### 详细叙述

6.6内核，23.05，全新编译两次都是提示这个错误，求助大佬这个是哪里的问题！
感谢！

### 重复 issue

- [x] 没有类似的 issue

### 具体型号

x86

### 详细日志

cp -fpR /home/drw/lede/build_dir/target-x86_64_musl/elfutils-0.192/ipkg-install/usr/include/* /home/drw/lede/tmp/stage-elfutils/usr/include/
install -d -m0755 /home/drw/lede/tmp/stage-elfutils/usr/lib
cp -fpR /home/drw/lede/build_dir/target-x86_64_musl/elfutils-0.192/ipkg-install/usr/lib/libasm*.{a,so*} /home/drw/lede/tmp/stage-elfutils/usr/lib/
cp -fpR /home/drw/lede/build_dir/target-x86_64_musl/elfutils-0.192/ipkg-install/usr/lib/libdw*.{a,so*} /home/drw/lede/tmp/stage-elfutils/usr/lib/
cp -fpR /home/drw/lede/build_dir/target-x86_64_musl/elfutils-0.192/ipkg-install/usr/lib/libelf*.{a,so*} /home/drw/lede/tmp/stage-elfutils/usr/lib/
install -d -m0755 /home/drw/lede/tmp/stage-elfutils/usr/lib/pkgconfig
cp -fpR /home/drw/lede/build_dir/target-x86_64_musl/elfutils-0.192/ipkg-install/usr/lib/pkgconfig/libelf.pc /home/drw/lede/tmp/stage-elfutils/usr/lib/pkgconfig/
cp -fpR /home/drw/lede/build_dir/target-x86_64_musl/elfutils-0.192/ipkg-install/usr/lib/pkgconfig/libdw.pc /home/drw/lede/tmp/stage-elfutils/usr/lib/pkgconfig/
find /home/drw/lede/tmp/stage-elfutils -name '*.la' | xargs -r rm -f;
if [ -f /home/drw/lede/staging_dir/target-x86_64_musl/packages/elfutils.list ]; then /home/drw/lede/scripts/clean-package.sh "/home/drw/lede/staging_dir/target-x86_64_musl/packages/elfutils.list" "/home/drw/lede/staging_dir/target-x86_64_musl"; fi
if [ -d /home/drw/lede/tmp/stage-elfutils ]; then (cd /home/drw/lede/tmp/stage-elfutils; find ./ > /home/drw/lede/tmp/stage-elfutils.files);         SHELL= flock /home/drw/lede/tmp/.staging-dir.flock -c ' mv /home/drw/lede/tmp/stage-elfutils.files /home/drw/lede/staging_dir/target-x86_64_musl/packages/elfutils.list && cp -fpR /home/drw/lede/tmp/stage-elfutils/* /home/drw/lede/staging_dir/target-x86_64_musl/; '; fi
rm -rf /home/drw/lede/tmp/stage-elfutils
touch /home/drw/lede/staging_dir/target-x86_64_musl/stamp/.elfutils_installed
make[3]: Leaving directory '/home/drw/lede/package/libs/elfutils'
time: package/libs/elfutils/compile#160.34#131.40#365.21
make[2]: Leaving directory '/home/drw/lede'
make[1]: *** [package/Makefile:110: /home/drw/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/drw/lede'
make: *** [/home/drw/lede/include/toplevel.mk:231: world] Error 2

Azbuzz, 2025-01-29:
老版本luci也过不了，正常默认都过不去~试了几次，就选了个ssr，其他都没动试了几次-.-

wj45853007, 2025-02-05:
一样的问题，我也是这个报错

drw2005, 2025-02-07:
今天重新编译顺利通过了。

closed 2025-02-07 by drw2005