coolsnowwolf/lede #13305: 看起来像是rust 404

Jason6111, 2025-01-23:
### 详细叙述

在线编译错误

### 重复 issue

- [x] 没有类似的 issue

### 具体型号

x86

### 详细日志

CARGO_HOME=/workdir/openwrt/dl/cargo TARGET_CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/workdir/openwrt/build_dir/target-x86_64_musl/rust-1.80.1=rust-1.80.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" python3 /workdir/openwrt/build_dir/target-x86_64_musl/host/rustc-1.80.1-src/x.py --build-dir /workdir/openwrt/build_dir/target-x86_64_musl/host/rustc-1.80.1-src/build --config /workdir/openwrt/build_dir/target-x86_64_musl/host/rustc-1.80.1-src/config.toml dist build-manifest cargo llvm-tools rustc rust-std rust-src
Building bootstrap
WARNING: The `change-id` is missing in the `config.toml`. This means that you will not be able to track the major changes made to the bootstrap configurations.
NOTE: to silence this warning, add `change-id = 125535` at the top of `config.toml`
downloading https://ci-artifacts.rust-lang.org/rustc-builds/3f5fd8dd41153bc5fdca9427e9e05be2c767ba23/rust-dev-1.80.1-x86_64-unknown-linux-gnu.tar.xz
curl: (22) The requested URL returned error: 404
ERROR: failed to download llvm from ci

    HELP: There could be two reasons behind this:
        1) The host triple is not supported for `download-ci-llvm`.
        2) Old builds get deleted after a certain time.
    HELP: In either case, disable `download-ci-llvm` in your config.toml:

    [llvm]
    download-ci-llvm = false
    
Build completed unsuccessfully in 0:00:00
make[3]: *** [Makefile:109: /workdir/openwrt/build_dir/target-x86_64_musl/host/rustc-1.80.1-src/.built] Error 1
make[3]: Leaving directory '/workdir/openwrt/feeds/packages/lang/rust'
time: package/feeds/packages/rust/host-compile#0.51#0.25#1.07
    ERROR: package/feeds/packages/rust [host] failed to build.
make[2]: *** [package/Makefile:114: package/feeds/packages/rust/host/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 2

coolveyron, 2025-01-23:
如果你是在腾讯云编译的话，腾讯云的所有服务器已经无法正常编译openwrt，换阿里云吧。所有腾讯云的服务器都无法下载部分组件。

目前我只知道有一个问题在git.openwrt.org上，能ping通，但是返回的是假地址，是无法从git.openwrt.org中获取组件的正确hash，所以下载都出错。其他组件的地址是不是都这样，没继续研究了。

已经问过腾讯云的客服了，他们知道，说是运营商的问题，正在积极协调解决。

美区？香港？日本？都是运营商的问题？但是阿里云没问题啊。

不懂腾讯云的操作是不是故意而为之。

所以只要是用腾讯云服务器编译openwrt下载错误的问题，就直接换云吧。

Jason6111, 2025-01-23:
> 如果你是在腾讯云编译的话，腾讯云的所有服务器已经无法正常编译openwrt，换阿里云吧。所有腾讯云的服务器都无法下载部分组件。
> 
> 目前我只知道有一个问题在git.openwrt.org上，能ping通，但是返回的是假地址，是无法从git.openwrt.org中获取组件的正确hash，所以下载都出错。其他组件的地址是不是都这样，没继续研究了。
> 
> 已经问过腾讯云的客服了，他们知道，说是运营商的问题，正在积极协调解决。
> 
> 美区？香港？日本？都是运营商的问题？但是阿里云没问题啊。
> 
> 不懂腾讯云的操作是不是故意而为之。
> 
> 所以只要是用腾讯云服务器编译openwrt下载错误的问题，就直接换云吧。

我用的github actions

201821143044, 2025-01-23:
我也是github actions遇到同样的问题

GreatMichaelLee, 2025-01-23:
同样问题，有点莫名其妙不知道怎么来的

lxl6125, 2025-01-23:
github actions，同样的问题

rshi0212, 2025-01-24:
fw876/helloworld
我也编译不过去

1WuNeiGui, 2025-01-24:
这个404的问题我在腾讯云上编译也出现了，这个现象好长时间了，我按照上面那位大佬说的，换到阿里云了确实可以编译成功，github actions的没试过

lxl6125, 2025-01-24:
[image]
本地编译也有这个问题

GreatMichaelLee, 2025-01-24:
> [image] 本地编译也有这个问题

我的就是本地编译，同样问题。我本地科学方面没问题。这个就是这个rust 的CI源失效了。换成rust 1.84的可以直接浏览器下载，用楼上代码pull下来的URL就是404，但是我翻遍了库也没找到在哪里换源，config.toml没有这个，我对这块不熟。还是请大佬看看尽量fix吧。多谢 @coolsnowwolf

closed 2025-01-24 by coolsnowwolf