coolsnowwolf/lede #13306: 编译找不到wireguard是不是下架了

bikioxo, 2025-01-23:
### 详细叙述

编译进入程序找不到wireguard是不是下架了

### 重复 issue

- [x] 没有类似的 issue

### 具体型号

X86

### 详细日志

[image]

1WuNeiGui, 2025-01-23:
这个直接在接口下有wireguard，编译那个就行了

bi4tms, 2025-01-23:
在protocol里面

palemoky, 2025-01-28:
在这 `Luci -> Protocols -> Wireguard`

Doraemon309, 2025-02-06:
> 在这 `Luci -> Protocols -> Wireguard`

我必须使用 ssh 吗？没有出现在界面上？
我计划添加 src-git master https://github.com/coolsnowwolf/luci 以包含 luci-app-wireguard！

palemoky, 2025-02-06:
> > 在这 `Luci -> Protocols -> Wireguard`
> 
> 我必须使用 ssh 吗？没有出现在界面上？ 我计划添加 src-git master [coolsnowwolf/luci](https://github.com/coolsnowwolf/luci) 以包含 luci-app-wireguard！

我是在本地编译的，拉取最新代码执行到 `make menuconfig` 这个步骤就可以在配置界面的上述路径中看到这个配置，不需要执行 `src-git`

Doraemon309, 2025-02-06:
> 我是在本地编译的，拉取最新代码执行到 `make menuconfig` 这个步骤就可以在配置界面的上述路径中看到这个配置，不需要执行 `src-git`

[image]
是的，实际上有很多高级说明不在 README 中！需要用户学习。
我也提取了最新版本，但是内核不支持 24.10 稳定版，所以我恢复到了 23.05.5！还合并了2个coolsnowwolf luci分支进行编译！编译时有luci-app-wireguard！

Doraemon309, 2025-02-06:
我不知道这是否会导致 coolsnowwolf/luci 冲突！因为我可以将 master 和 openwrt-23.05 分支都放在 feeds.conf.default 中！我仍在等待编译完成并进行测试！ @palemoky

closed 2025-02-17 by aiamadeus