coolsnowwolf/lede #13319: luci-theme-argon18.06

mcusee, 2025-02-04:
luci-theme-argon18.06

mcusee, 2025-02-04:
tmp/.config-package.in:80583:error: 检测到递归依赖关系！
tmp/.config-package.in:80583：符号 PACKAGE_iptables 由 PACKAGE_luci-app-passwall_Iptables_Transparent_Proxy 选定
tmp/.config-package.in:62758：符号 PACKAGE_luci-app-passwall_Iptables_Transparent_Proxy 默认可见，具体取决于 PACKAGE_firewall4
tmp/.config-package.in:346：符号 PACKAGE_firewall4 由 PACKAGE_luci-app-fchomo 选定
tmp/.config-package.in:62008：符号 PACKAGE_luci-app-fchomo 依赖于 PACKAGE_coreutils
tmp/.config-package.in:117726：符号 PACKAGE_coreutils 由 PACKAGE_luci-app-bypass 选定
tmp/.config-package.in:61566：符号 PACKAGE_luci-app-bypass依赖于 PACKAGE_iptables

coolveyron, 2025-02-10:
> tmp/.config-package.in:80583:error: 检测到递归依赖关系！ tmp/.config-package.in:80583：符号 PACKAGE_iptables 由 PACKAGE_luci-app-passwall_Iptables_Transparent_Proxy 选定 tmp/.config-package.in:62758：符号 [...]

./scripts/feeds install -a 之后，如果你不用bypass的话，在feeds文件夹里面删除bypass。

closed 2025-03-14 by coolsnowwolf