coolsnowwolf/lede #13326: 编译出错：no nfsdcltrack file or directory

GreatMichaelLee, 2025-02-07:
### 详细叙述

touch /home/builder/lede_x86/staging_dir/target-x86_64_musl/root-x86/stamp/.netperf_installed
echo "netperf" >> /home/builder/lede_x86/staging_dir/target-x86_64_musl/pkginfo/netperf.default.install
make[3]: Leaving directory '/home/builder/lede_x86/feeds/packages/net/netperf'
time: package/feeds/packages/netperf/compile#0.05#0.11#0.15
make[3]: Entering directory '/home/builder/lede_x86/feeds/packages/libs/libtirpc'
make[3]: Leaving directory '/home/builder/lede_x86/feeds/packages/libs/libtirpc'
time: package/feeds/packages/libtirpc/host-compile#0.05#0.11#0.15
make[3]: Entering directory '/home/builder/lede_x86/feeds/packages/net/nfs-kernel-server'
make[3]: Leaving directory '/home/builder/lede_x86/feeds/packages/net/nfs-kernel-server'
time: package/feeds/packages/nfs-kernel-server/host-compile#0.05#0.13#0.17
make[3]: Entering directory '/home/builder/lede_x86/feeds/packages/libs/tcp_wrappers'
mkdir -p /home/builder/lede_x86/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/builder/lede_x86/tmp/.root-copy.flock -c 'cp -fpR /home/builder/lede_x86/build_dir/target-x86_64_musl/tcp_wrappers_7.6/.pkgdir/libwrap/. /home/builder/lede_x86/staging_dir/target-x86_64_musl/root-x86/'
touch /home/builder/lede_x86/staging_dir/target-x86_64_musl/root-x86/stamp/.libwrap_installed
echo "libwrap" >> /home/builder/lede_x86/staging_dir/target-x86_64_musl/pkginfo/tcp_wrappers.default.install
make[3]: Leaving directory '/home/builder/lede_x86/feeds/packages/libs/tcp_wrappers'
time: package/feeds/packages/tcp_wrappers/compile#0.04#0.09#0.15
make[3]: Entering directory '/home/builder/lede_x86/feeds/packages/net/rpcbind'
mkdir -p /home/builder/lede_x86/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/builder/lede_x86/tmp/.root-copy.flock -c 'cp -fpR /home/builder/lede_x86/build_dir/target-x86_64_musl/rpcbind-1.2.6/.pkgdir/rpcbind/. /home/builder/lede_x86/staging_dir/target-x86_64_musl/root-x86/'
touch /home/builder/lede_x86/staging_dir/target-x86_64_musl/root-x86/stamp/.rpcbind_installed
echo "rpcbind" >> /home/builder/lede_x86/staging_dir/target-x86_64_musl/pkginfo/rpcbind.default.install
make[3]: Leaving directory '/home/builder/lede_x86/feeds/packages/net/rpcbind'
time: package/feeds/packages/rpcbind/compile#0.09#0.09#0.18
make[3]: Entering directory '/home/builder/lede_x86/feeds/packages/net/nfs-kernel-server'
mkdir -p /home/builder/lede_x86/bin/targets/x86/64/packages /home/builder/lede_x86/build_dir/target-x86_64_musl/nfs-utils-2.6.2/ipkg-x86_64/nfs-utils/CONTROL /home/builder/lede_x86/staging_dir/target-x86_64_musl/pkginfo
install -d -m0755 /home/builder/lede_x86/build_dir/target-x86_64_musl/nfs-utils-2.6.2/ipkg-x86_64/nfs-utils/sbin
install -d -m0755 /home/builder/lede_x86/build_dir/target-x86_64_musl/nfs-utils-2.6.2/ipkg-x86_64/nfs-utils/usr/sbin
install -m0755 /home/builder/lede_x86/build_dir/target-x86_64_musl/nfs-utils-2.6.2/ipkg-install/sbin/mount.nfs /home/builder/lede_x86/build_dir/target-x86_64_musl/nfs-utils-2.6.2/ipkg-x86_64/nfs-utils/sbin/
install -m0755 /home/builder/lede_x86/build_dir/target-x86_64_musl/nfs-utils-2.6.2/ipkg-install/sbin/nfsdcltrack /home/builder/lede_x86/build_dir/target-x86_64_musl/nfs-utils-2.6.2/ipkg-x86_64/nfs-utils/sbin/
install: cannot stat '/home/builder/lede_x86/build_dir/target-x86_64_musl/nfs-utils-2.6.2/ipkg-install/sbin/nfsdcltrack': No such file or directory
make[3]: *** [Makefile:223: /home/builder/lede_x86/bin/packages/x86_64/packages/nfs-utils_2.6.2-4_x86_64.ipk] Error 1
make[3]: Leaving directory '/home/builder/lede_x86/feeds/packages/net/nfs-kernel-server'
time: package/feeds/packages/nfs-kernel-server/compile#0.05#0.15#0.19
    ERROR: package/feeds/packages/nfs-kernel-server failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/nfs-kernel-server/compile] Error 1
make[2]: Leaving directory '/home/builder/lede_x86'
make[1]: *** [package/Makefile:110: /home/builder/lede_x86/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/builder/lede_x86'
make: *** [/home/builder/lede_x86/include/toplevel.mk:231：world] 错误 2
builder@Build-Server:/home/builder/lede_x86$

### 重复 issue

- [x] 没有类似的 issue

### 具体型号

x86

### 详细日志

rm -rf ./tmp && make clean之后依然报这个错，日志见上

woni928, 2025-02-07:
同样

GreatMichaelLee, 2025-02-07:
@coolsnowwolf 麻烦大佬看看呢，block住了，也不知道啥原因。

woni928, 2025-02-07:
等大佬解决

closed 2025-02-08 by coolsnowwolf