coolsnowwolf/lede #13332: luci23.05编译包含luci-app-dockerman的固件时提示存储空间不足

herouin, 2025-02-08:
### 详细叙述

旧版luci包含luci-app-dockerman时，KERNEL分区128M、ROOTFS分区4096M，外加一大堆其它插件从未提示磁盘存贮空间不足。
而自从luci更新至23.05后，即便KERNEL分区64M、ROOTFS分区1024M，哪怕是仅选择默认配置的插件+luci-app-dockerman，就会报磁盘存贮空间不足。在编译前已移除占用磁盘空间的非必要程序，下图是磁盘使用信息明细：

[image]

整个编译过程没有任何保报错，但因磁盘存贮空间不足无法生成固件文件。

请问结症在哪？

### 重复 issue

- [x] 没有类似的 issue

### 具体型号

X86

### 详细日志

[image]

yanxin152133, 2025-02-10:
试试这个清理磁盘的方法
[image]

coolsnowwolf, 2025-02-13:
4GB 云空间不够了

herouin, 2025-02-15:
> 4GB 云空间不够了

更新luci之后才出现的。我看大佬的Actions里也有空间不足你的报错，不知有没有解决方法？

[image]

lxl6125, 2025-02-17:
/mnt 这么大的空间为啥不用？

herouin, 2025-02-19:
> [image] /mnt 这么大的空间为啥不用？

感谢前辈指点迷津，已解决。