coolsnowwolf/lede #13335: 多次编译失败，换了系统也是同样的问题

ecco2005, 2025-02-10:
### 详细叙述

用ubuntu编译失败，后来换了debian，还是同样的问题

找到一个差不多的提问，但似乎没有解决方案

### 重复 issue

- [x] 没有类似的 issue

### 具体型号

x86

### 详细日志

make[3]: Entering directory '/home/maple/lede/feeds/packages/lang/rust'
. /home/maple/lede/include/shell.sh; /home/maple/lede/staging_dir/host/bin/libdeflate-gzip -dc /home/maple/lede/dl/rustc-1.84.0-src.tar.gz | tar -C /home/maple/lede/build_dir/target-x86_64_musl/host/rustc-1.84.0-src/.. -xf -
libdeflate-gzip: Out of memory
libdeflate-gzip: "/home/maple/lede/dl/rustc-1.84.0-src.tar.gz": file is probably too large to be processed by this program
tar: This does not look like a tar archive
tar: Exiting with failure status due to previous errors
make[3]: *** [Makefile:111: /home/maple/lede/build_dir/target-x86_64_musl/host/rustc-1.84.0-src/.prepared8509954dc350916ed516dd227a023c96_6664517399ebbbc92a37c5bb081b5c53] Error 2
make[3]: Leaving directory '/home/maple/lede/feeds/packages/lang/rust'
time: package/feeds/packages/rust/host-compile#6.80#2.39#17.84
    ERROR: package/feeds/packages/rust [host] failed to build.
make[2]: *** [package/Makefile:114: package/feeds/packages/rust/host/compile] Error 1
make[2]: Leaving directory '/home/maple/lede'
make[1]: *** [package/Makefile:110: /home/maple/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/maple/lede'
make: *** [/home/maple/lede/include/toplevel.mk:231: world] Error 2

coolveyron, 2025-02-10:
扩大一下内存试试，内存翻倍一下，存储空间也扩大一下。

ecco2005, 2025-02-10:
自己解决了，记录一下，问题出在这一句

. /home/maple/lede/include/shell.sh; /home/maple/lede/staging_dir/host/bin/libdeflate-gzip -dc /home/maple/lede/dl/rustc-1.84.0-src.tar.gz | tar -C /home/maple/lede/build_dir/target-x86_64_musl/host/rustc-1.84.0-src/.. -xf -
libdeflate-gzip: Out of memory
libdeflate-gzip: "/home/maple/lede/dl/rustc-1.84.0-src.tar.gz": file is probably too large to be processed by this program

/home/maple/lede/dl/rustc-1.84.0-src.tar.gz   这个文件699M，但是解压后占用了4个G的空间，共有36万个小文件。

解决方式是AI告诉我的

你需要解决根本原因——内存不足的问题。以下是详细的步骤，强调了最重要的部分：

1. 增加交换空间（至关重要）： 这是最有可能的解决方案。你的交换空间需要比你尝试解压的归档文件大得多。2GB 可能不够。尝试 4GB 或 8GB，如果你的磁盘空间足够的话。

`你需要解决根本原因——内存不足的问题。以下是详细的步骤，强调了最重要的部分：

1. 增加交换空间（至关重要）： 这是最有可能的解决方案。你的交换空间需要比你尝试解压的归档文件大得多。2GB 可能不够。尝试 4GB 或 8GB，如果你的磁盘空间足够的话。

Bash

sudo fallocate -l 4G /swapfile  # 或 8G，根据需要调整
sudo chmod 600 /swapfile
sudo mkswap /swapfile
sudo swapon /swapfile
echo "/swapfile none swap sw 0 0" | sudo tee -a /etc/fstab  # 使其永久生效
free -h  # 验证交换空间大小`

我改成了8G，重新make V=s -j1 
不再报错了

helllkz, 2025-02-14:
我很久没编译，前几天也是这个问题，还是看到这个错误，把虚拟机内存增大到8G后就没问题了

closed 2025-02-14 by coolsnowwolf