coolsnowwolf/lede #13338: 23.05-部分软件没有图标

woni928, 2025-02-12:
### 详细叙述

19.07正常
23.05目前发现没有的：
ipsec-server
softethervpn

哪位同学知道怎么解决，感谢！

### 重复 issue

- [x] 没有类似的 issue

### 具体型号

N1

### 详细日志

无

woni928, 2025-02-13:
难道没人遇到？

minfengyk, 2025-02-15:
> 难道没人遇到？

我也遇到，喜欢用softethervpn，有几月了，一直不见修复。早有人反映了，估计大佬没时间吧。改immortalwrt了，也很稳定，建议试试。

woni928, 2025-02-16:
> > 难道没人遇到？
> 
> 我也遇到，喜欢用softethervpn，有几月了，一直不见修复。早有人反映了，估计大佬没时间吧。改immortalwrt了，也很稳定，建议试试。

immortalwrt 正常？ 具体怎么操作，又得折腾。

woni928, 2025-02-17:
> > > > 难道没人遇到？
> > > 
> > > 
> > > 我也遇到，喜欢用softethervpn，有几月了，一直不见修复。早有人反映了，估计大佬没时间吧。改immortalwrt了，也很稳定，建议试试。
> > 
> > 
> > immortalwrt 正常？ 具体怎么操作，又得折腾。
> 
> 到ImmortalWrt项目下，拉取天灵大佬的源码编译啊，23.05分支和24.10分支都很稳定的。就是23.05下softethervpn设置好后要修改一下配置文件，要不然连不上。
> 
> 5.02版本softethervpn默认没有开启监听IPv4，导致连不上（大坑）。需要在$vpn_server.config配置文件内添加string ListenIP 0.0.0.0，避免UDP端口只监控IPv6地址。 开启IPv4监听方法：在SoftEther VPN Server管理器里点击“编辑设置”，导出配置文件，添加string ListenIP 0.0.0.0后，导入文件并应用。

感谢感谢
云编译了下天灵的源码，都失败。
有没有现场的，炒个作业🌚

minfengyk, 2025-02-17:
> 感谢感谢 云编译了下天灵的源码，都失败。 有没有现场的，炒个作业🌚

我都是本地编译的，云编译没经验。

closed 2025-03-27 by coolsnowwolf