coolsnowwolf/lede #13341: chrome 浏览器打开偶发XHR request timed out

yuhongwei380, 2025-02-13:
### 详细叙述

拉取的lede最新分支编译了固件。
用chrome打开lan口地址进行管理时，偶发XHR request timed out 报错；
然后切换用无痕模式，报错依旧；用edge 可以解决这个问题。

### 重复 issue

- [x] 没有类似的 issue

### 具体型号

R4S

### 详细日志

编译没有报错；

yuhongwei380, 2025-02-13:
[image]
也有XHR request aborted by browser

ncheng89, 2025-02-14:
我也是，固件用了好几天了，今天突然提示

[image]

xzchsia, 2025-02-15:
我用的20250215编译的最新版本，K3路由器，使用一会后，系统会变卡，使用edge浏览器打开无线网络的编辑也会出现Error
XHR request timed out
[image]

yanxin152133, 2025-02-21:
+1，chrome 还会中英文随机切换

yuhongwei380, 2025-02-21:
> +1，chrome 还会中英文随机切换

是的，我一开始还以为我的问题。

coolsnowwolf, 2025-03-14:
解决了

yuhongwei380, 2025-03-15:
> 解决了

好的，请问是最新分支吗

closed 2025-03-15 by yuhongwei380