coolsnowwolf/lede #13343: wan口老是重启

ncheng89, 2025-02-14:
### 详细叙述

x86设备，cpu是n5105
wan口老是重启, 这个是固件问题，还是硬件问题

### 重复 issue

- [x] 没有类似的 issue

### 具体型号

x86

### 详细日志

[284877.041239] igc 0000:13:00.0 eth2: Reset adapter
[284880.042788] igc 0000:13:00.0 eth2: NIC Link is Up 2500 Mbps Full Duplex, Flow Control: RX
[285809.943159] igc 0000:13:00.0 eth2: NETDEV WATCHDOG: CPU: 3: transmit queue 2 timed out 5290 ms
[285809.943803] igc 0000:13:00.0 eth2: Register Dump
[285809.944117] igc 0000:13:00.0 eth2: Register Name   Value
[285809.944441] igc 0000:13:00.0 eth2: CTRL            081c0641
[285809.944753] igc 0000:13:00.0 eth2: STATUS          40680683
[285809.945099] igc 0000:13:00.0 eth2: CTRL_EXT        10000040
[285809.945408] igc 0000:13:00.0 eth2: MDIC            1805d1e1
[285809.945731] igc 0000:13:00.0 eth2: ICR             00000081
[285809.946057] igc 0000:13:00.0 eth2: RCTL            04408022
[285809.946436] igc 0000:13:00.0 eth2: RDLEN[0-3]      00001000 00001000 00001000 00001000
[285809.946881] igc 0000:13:00.0 eth2: RDH[0-3]        0000007d 000000c9 000000b9 0000000a
[285809.947346] igc 0000:13:00.0 eth2: RDT[0-3]        0000007c 000000c8 000000b8 00000009
[285809.947785] igc 0000:13:00.0 eth2: RXDCTL[0-3]     02040808 02040808 02040808 02040808
[285809.948218] igc 0000:13:00.0 eth2: RDBAL[0-3]      08aa7000 0d20d000 0d256000 06c7f000
[285809.948665] igc 0000:13:00.0 eth2: RDBAH[0-3]      00000000 00000000 00000000 00000000
[285809.949112] igc 0000:13:00.0 eth2: TCTL            a503f0fa
[285809.949440] igc 0000:13:00.0 eth2: TDBAL[0-3]      0b5f2000 0b5fb000 0840e000 07daf000
[285809.949875] igc 0000:13:00.0 eth2: TDBAH[0-3]      00000000 00000000 00000000 00000000
[285809.950308] igc 0000:13:00.0 eth2: TDLEN[0-3]      00001000 00001000 00001000 00001000
[285809.950741] igc 0000:13:00.0 eth2: TDH[0-3]        00000045 0000000f 000000ba 000000b5
[285809.951174] igc 0000:13:00.0 eth2: TDT[0-3]        00000045 0000000f 000000ba 000000b5
[285809.951608] igc 0000:13:00.0 eth2: TXDCTL[0-3]     02100108 02100108 02100108 02100108
[285809.952035] igc 0000:13:00.0 eth2: Reset adapter
[285809.986806] igc 0000:13:00.0 eth2: Register Dump
[285809.987107] igc 0000:13:00.0 eth2: Register Name   Value
[285809.987427] igc 0000:13:00.0 eth2: CTRL            081c0641
[285809.987759] igc 0000:13:00.0 eth2: STATUS          40680681
[285809.988076] igc 0000:13:00.0 eth2: CTRL_EXT        10000040
[285809.988398] igc 0000:13:00.0 eth2: MDIC            18017969
[285809.988712] igc 0000:13:00.0 eth2: ICR             00000001
[285809.989033] igc 0000:13:00.0 eth2: RCTL            04408022
[285809.989350] igc 0000:13:00.0 eth2: RDLEN[0-3]      00001000 00001000 00001000 00001000
[285809.989785] igc 0000:13:00.0 eth2: RDH[0-3]        00000000 00000000 00000000 00000000
[285809.990220] igc 0000:13:00.0 eth2: RDT[0-3]        000000ff 000000ff 000000ff 000000ff
[285809.990655] igc 0000:13:00.0 eth2: RXDCTL[0-3]     02040808 02040808 02040808 02040808
[285809.991089] igc 0000:13:00.0 eth2: RDBAL[0-3]      08aa7000 0d20d000 0d256000 06c7f000
[285809.991528] igc 0000:13:00.0 eth2: RDBAH[0-3]      00000000 00000000 00000000 00000000
[285809.991958] igc 0000:13:00.0 eth2: TCTL            a50400fa
[285809.992269] igc 0000:13:00.0 eth2: TDBAL[0-3]      0b5f2000 0b5fb000 0840e000 07daf000
[285809.992702] igc 0000:13:00.0 eth2: TDBAH[0-3]      00000000 00000000 00000000 00000000
[285809.993173] igc 0000:13:00.0 eth2: TDLEN[0-3]      00001000 00001000 00001000 00001000
[285809.993614] igc 0000:13:00.0 eth2: TDH[0-3]        0000000d 00000000 00000000 00000000
[285809.994052] igc 0000:13:00.0 eth2: TDT[0-3]        00000010 00000000 00000000 00000000
[285809.994488] igc 0000:13:00.0 eth2: TXDCTL[0-3]     02100108 02100108 02100108 02100108
[285809.994916] igc 0000:13:00.0 eth2: Reset adapter
[285813.903482] igc 0000:13:00.0 eth2: NIC Link is Up 2500 Mbps Full Duplex, Flow Control: RX

aiamadeus, 2025-02-14:
i225 还是 i226 啊

ncheng89, 2025-02-14:
> i225 还是 i226 啊

i225

coolsnowwolf, 2025-02-17:
I225 的硬件bug，ASIC 死了，更换硬件吧，或者升级网卡 fw 试试

ncheng89, 2025-02-17:
可是我用5.15的内核版本没有这个问题，很稳定
[image]

ncheng89, 2025-04-17:
@coolsnowwolf 大佬如果把网卡换成226会不会还有这个问题

ncheng89, 2025-04-17:
5.15内核和6.6的内核的fw都是2014的版本，要怎么才可以解决这个问题：
```
root@OpenWrt:~# ethtool -i eth2
driver: igc
version: 6.6.75
firmware-version: 2014:8877
expansion-rom-version: 
bus-info: 0000:13:00.0
supports-statistics: yes
supports-test: yes
supports-eeprom-access: yes
supports-register-dump: yes
supports-priv-flags: yes
root@OpenWrt:~# lspci -nn | grep -i eth
04:00.0 Ethernet controller [0200]: VMware VMXNET3 Ethernet Controller [15ad:07b0] (rev 01)
0b:00.0 Ethernet controller [0200]: Intel Corporation Device [8086:125c] (rev 04)
13:00.0 Ethernet controller [0200]: Intel Corporation Device [8086:125c] (rev 04)
1b:00.0 Ethernet controller [0200]: Intel Corporation Device [8086:125c] (rev 04)
root@OpenWrt:~# exit
Connection to 192.168.22.1 closed.
[root@msjj-data]# ssh root@192.168.23.1
root@192.168.23.1's password: 

BusyBox v1.36.0 (2023-04-27 16:05:38 UTC) built-in shell (ash)

     _________
    /        /\      _    ___ ___  ___
   /  LE    /  \    | |  | __|   \| __|
  /    DE  /    \   | |__| _|| |) | _|
 /________/  LE  \  |____|___|___/|___|
 \        \   DE /
  \    LE  \    /  -------------------------------------------
   \  DE    \  /    OpenWrt SNAPSHOT, r5931-585c2ca50
    \________\/    -------------------------------------------

root@23X:~# ethtool -i eth2
driver: igc
version: 5.15.105
firmware-version: 2014:8877
expansion-rom-version: 
bus-info: 0000:0b:00.0
supports-statistics: yes
supports-test: yes
supports-eeprom-access: yes
supports-register-dump: yes
supports-priv-flags: yes
root@23X:~# lspci -nn | grep -i eth
02:01.0 Ethernet controller [0200]: Intel Corporation 82545EM Gigabit Ethernet Controller (Copper) [8086:100f] (rev 01)
03:00.0 Ethernet controller [0200]: Intel Corporation Device [8086:125c] (rev 04)
0b:00.0 Ethernet controller [0200]: Intel Corporation Device [8086:125c] (rev 04)
13:00.0 Ethernet controller [0200]: Intel Corporation Device [8086:125c] (rev 04)
root@23X:~# 
```

mmeiwenlong, 2025-07-08:
> 我仔细看了一下225和226都有这个问题

这问题还有么

closed 2026-09-05 by coolsnowwolf