coolsnowwolf/lede #13349: mac m4 芯片编译报错

Jiang00, 2025-02-18:
### 详细叙述

m4芯片报错

### 重复 issue

- [x] 没有类似的 issue

### 具体型号

mac m4 pro

### 详细日志

make[3]: *** [Makefile:38: /Volumes/OpenWrt/lede/build_dir/host/elfutils-0.191/.configured] Error 1
make[3]: Leaving directory '/Volumes/OpenWrt/lede/tools/elfutils'
time: tools/elfutils/compile#3.28#0.82#6.36
    ERROR: tools/elfutils failed to build.
make[2]: *** [tools/Makefile:231: tools/elfutils/compile] Error 1
make[2]: Leaving directory '/Volumes/OpenWrt/lede'
make[1]: *** [tools/Makefile:227: /Volumes/OpenWrt/lede/staging_dir/host/stamp/.tools_compile_nyyynyyynnnyyyyynyyyyyyynyyynynyynnynnyyyyyynyyyyyyyyyyynynnyyyyynyyy] Error 2
make[1]: Leaving directory '/Volumes/OpenWrt/lede'
make: *** [/Volumes/OpenWrt/lede/include/toplevel.mk:231: world] Error 2

canghaiwuhen, 2025-02-20:
@Jiang00 m4 编译需要多久

Jiang00, 2025-03-06:
@canghaiwuhen 不好意思才看到，我的是顶配的mac mini，一个小时以内，具体多久没注意，反正挺快，我之前用inter的macpro要一晚上

Margele, 2025-03-17:
我也遇到了，在尝试编译Immortalwrt的时候报找不到elftools，猜测可能与Python的elftools有关系。

pip3 install pyelftools

metaforget, 2025-03-30:
我用op官方源仓库不会有这个问题直接编译成功了，用lede就会一样，并且lede编译一次后就用不了make menuconfig了
用大模型改了一下变量export CC=gcc-11  export CXX=g++-11，又有别的问题了，不知道你们成功没 @Margele @Jiang00

coolsnowwolf, 2026-09-05:
该问题对应的 tools/elfutils macOS 构建失败已在提交 `68a496c92ab` 修复（处理 macOS common symbol 构建兼容性）。当前仓库已包含该修复，关闭 issue。

closed 2026-09-05 by coolsnowwolf