coolsnowwolf/lede #13352: 新编译的固件 没有 shadowsocksr-libev-ssr-local

rshi0212, 2025-02-19:
### 详细叙述

新编译的x86/rockchip/filogic固件 没有 SSR shadowsocksr-libev-ssr-local，shadowsocksr-libev相关的都没有，SS shadowsocks-libev-ss-local有，确认都勾选了。

### 重复 issue

- [x] 没有类似的 issue

### 具体型号

x86/rockchip/filogic

### 详细日志

x86/rockchip/filogic

Goldberg-Fan, 2025-02-21:
cd /home/你的id名字/lede/package/lean
rm -rf luci-app-ssr-plus
git clone https://github.com/fw876/helloworld.git

zhao-binyu, 2025-03-10:
一模一样的问题

coolsnowwolf, 2026-09-05:
已核对当前 feeds：`shadowsocksr-libev` 仍提供 `shadowsocksr-libev-ssr-local`、`-ssr-redir`、`-ssr-server`，且 luci-app-ssr-plus 通过 `CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_ShadowsocksR_Libev_Client` 控制依赖。该包并未从源码移除；需要在 LuCI 的 SSR Libev Client 选项中启用后才会进入固件。因此关闭此“缺包” issue。

closed 2026-09-05 by coolsnowwolf