coolsnowwolf/lede #13353: RAX3000Q(y)无法启动

lurenjia1213, 2025-02-19:
### 详细叙述

使用的uboot与分区
https://github.com/hzyitc/openwrt-redmi-ax3000/issues/73#issuecomment-2259591683

在使用此仓库编译的镜像时，uboot启动日志如下

`U-Boot 2016.01 (Nov 28 2023 - 20:41:48 +0800)

DRAM:  smem ram ptable found: ver: 1 len: 4
256 MiB
USB: Node Not found, skipping initia*9瞠QPIC controller support serial NAND
ID = 21e521e5
Vendor = e5
Device = 21
Serial Nand Device Found With ID : 0xe5 0x21
Serial NAND device Manufacturer:DS35M1GA
Device Size:128 MiB, Page size:2048, Spare Size:64, ECC:4-bit
SF: Unsupported flash IDs: manuf 00, jedec 0000, ext_jedec 0000
ipq_spi: SPI Flash not found (bus/cs/speed/mode) = (0/0/48000000/0)
128 MiB
MMC:   sdhci: Node Not found, skipping initialization

PCI0 is not defined in the device tree
PCI1 is not defined in the device tree
In:    serial@78AF000
Out:   serial@78AF000
Err:   serial@78AF000
machid: 8040000
eth0 MAC Address from ART is not valid
eth1 MAC Address from ART is not valid
in main_loop
Net:   cmbblk is stable 5
MAC0 addr:0:11:22:33:44:55
PHY ID1: 0x4d
PHY ID2: 0xd0c0
MAC1 addr:0:11:22:33:44:56
athrs17_reg_init: complete
athrs17_vlan_config ...done
S17c init  done
board_update_caldata: Unable to find slot-Id, Default CapIn/CapOut values used
eth0, eth1
btn init write value c3
Start Check button is pressd
Now GPIO RESET BTN IS 23
Compare gpio get value is 1 compare 0
Hit any key to stop autoboot:  0 
ubi0: attaching mtd1
ubi0: scanning is finished
ubi0: attached mtd1 (name "mtd=0", size 110 MiB)
ubi0: PEB size: 131072 bytes (128 KiB), LEB size: 126976 bytes
ubi0: min./max. I/O unit sizes: 2048/2048, sub-page size 2048
ubi0: VID header offset: 2048 (aligned 2048), data offset: 4096
ubi0: good PEBs: 880, bad PEBs: 0, corrupted PEBs: 0
ubi0: user volume: 3, internal volumes: 1, max. volumes count: 128
ubi0: max/mean erase counter: 1/0, WL threshold: 4096, image sequence number: 1739849100
ubi0: available PEBs: 0, total reserved PEBs: 880, PEBs reserved for bad PEB handling: 20
Read 0 bytes from volume kernel to 44000000
No size specified -> Using max size (5079040)
## Loading kernel from FIT Image at 44000000 ...
   Using 'config@mp02.1' configuration
   Trying 'kernel-1' kernel subimage
     Description:  ARM64 OpenWrt Linux-6.1.128
     Type:         Kernel Image
     Compression:  gzip compressed
     Data Start:   0x440000e8
     Data Size:    5001941 Bytes = 4.8 MiB
     Architecture: AArch64
     OS:           Linux
     Load Address: 0x41000000
     Entry Point:  0x41000000
     Hash algo:    crc32
     Hash value:   557b5d0e
     Hash algo:    sha1
     Hash value:   cee904f52e6c63b67c705cd1fd6f4a0eb2fb453a
   Verifying Hash Integrity ... crc32+ sha1+ OK
## Loading fdt from FIT Image at 44000000 ...
   Using 'config@mp02.1' configuration
   Trying 'fdt-1' fdt subimage
     Description:  ARM64 OpenWrt cmcc_rax3000q device tree blob
     Type:         Flat Device Tree
     Compression:  uncompressed
     Data Start:   0x444c5500
     Data Size:    24355 Bytes = 23.8 KiB
     Architecture: AArch64
     Hash algo:    crc32
     Hash value:   b77595f6
     Hash algo:    sha1
     Hash value:   508ddac5e0dfa4c61211c73e90c5b67d16a43784
   Verifying Hash Integrity ... crc32+ sha1+ OK
   Booting using the fdt blob at 0x444c5500
   Uncompressing Kernel Image ... OK
   Loading Device Tree to 4a3f7000, end 4a3fff22 ... OK
Could not find PCI in device tree
Could not find PCI in device tree
Using machid 0x8040000 from environment

Starting kernel ...

Jumping to AARCH64 kernel via monitor
`
修改内核加载地址为0x41080000,0x41208000 不能解决(我觉得应该不是这个的问题？)

32位镜像可以启动
https://github.com/kkstone/Actions-OpenWrt-RAX3000Q/releases/tag/RAX3000Q-2025.02.18-1737

### 重复 issue

- [x] 没有类似的 issue

### 具体型号

RAX3000-QY

### 详细日志

正常编译，没有报错

lurenjia1213, 2025-02-19:
> IPQ5018# smeminfo 
> ubi0: attaching mtd1
> ubi0: scanning is finished
> ubi0: attached mtd1 (name "mtd=0", size 110 MiB)
> ubi0: PEB size: 131072 bytes (128 KiB), LEB size: 126976 bytes
> ubi0: min./max. I/O unit sizes: 2048/2048, sub-page size 2048
> ubi0: VID header offset: 2048 (aligned 2048), data offset: 4096
> ubi0: good PEBs: 880, bad PEBs: 0, corrupted PEBs: 0
> ubi0: user volume: 3, internal volumes: 1, max. volumes count: 128
> ubi0: max/mean erase counter: 1/0, WL threshold: 4096, image sequence number: 1739849100
> ubi0: available PEBs: 0, total reserved PEBs: 880, PEBs reserved for bad PEB handling: 20
> flash_type:		0xb
> flash_index:		0x0
> flash_chip_select:	0x0
> flash_block_size:	0x20000
> flash_density:		0x80000
> partition table offset	0x0
> No.: Name             Attributes            Start             Size
>   0: 0:SBL1           0x0000ffff              0x0          0x80000
>   1: 0:MIBIB          0x0000ffff          0x80000          0x80000
>   2: 0:BOOTCONFIG     0x0000ffff         0x100000          0x40000
>   3: 0:BOOTCONFIG1    0x0000ffff         0x140000          0x40000
>   4: 0:QSEE           0x0000ffff         0x180000         0x100000
>   5: 0:QSEE_1         0x0000ffff         0x280000         0x100000
>   6: 0:DEVCFG         0x0000ffff         0x380000          0x40000
>   7: 0:DEVCFG_1       0x0000ffff         0x3c0000          0x40000
>   8: 0:CDT            0x0000ffff         0x400000          0x40000
>   9: 0:CDT_1          0x0000ffff         0x440000          0x40000
>  10: 0:APPSBLENV      0x0000ffff         0x480000          0x80000
>  11: 0:APPSBL         0x0000ffff         0x500000         0x140000
>  12: 0:APPSBL_1       0x0000ffff         0x640000         0x140000
>  13: 0:ART            0x0000ffff         0x780000         0x100000
>  14: 0:TRAINING       0x0000ffff         0x880000          0x80000
>  15: rootfs           0x0000ffff         0x900000        0x6e00000
> 	ubi vol 0 kernel
> 	ubi vol 1 rootfs
> 	ubi vol 2 rootfs_data
>

closed 2025-03-14 by coolsnowwolf

LoadingZhang, 2025-04-04:
@coolsnowwolf 同样的情况，日志也一样，卡在Jumping to AARCH64 kernel了
Could not find PCI in device tree
Could not find PCI in device tree
Using machid 0x8040000 from environment

Starting kernel ...

Jumping to AARCH64 kernel via monitor

lurenjia1213, 2025-04-04:
@coolsnowwolf 问题似乎还没解决
我注意到rax3000qy是ipq5000,但是[https://github.com/coolsnowwolf/lede/blob/7eb42a36189ae45a2ff66246bdd92092054710ed/target/linux/qualcommax/image/ipq50xx.mk#L11](代码)里面写的是ipq5018？这是否会导致启动失败

LoadingZhang, 2025-04-04:
@lurenjia1213 据说是因为内存不足已运行64bit的系统

LoadingZhang, 2025-04-04:
@lurenjia1213 如果有512M的内存的话可以换个试一下😅

Mrdeng233, 2025-04-21:
如果64位系统因为内存不足无法开机，那么如何编译32位系统呢，貌似我没有在编译界面找到32位的选项，隔壁的qsdk倒是可以编译32位的

lurenjia1213, 2025-04-21:
> 如果64位系统因为内存不足无法开机，那么如何编译32位系统呢，貌似我没有在编译界面找到32位的选项，隔壁的qsdk倒是可以编译32位的
参考这个
https://github.com/kkstone/Actions-OpenWrt-RAX3000Q/

说实话，我觉得应该不是内存不足导致的？
即使内存不足，内核最起码得在串口有输出吧？（不过我这条件有限，没法去深入分析这个问题

Mrdeng233, 2025-04-23:
> > 如果64位系统因为内存不足无法开机，那么如何编译32位系统呢，貌似我没有在编译界面找到32位的选项，隔壁的qsdk倒是可以编译32位的
> 
> 参考这个 https://github.com/kkstone/Actions-OpenWrt-RAX3000Q/
> 
> 说实话，我觉得应该不是内存不足导致的？ 即使内存不足，内核最起码得在串口有输出吧？（不过我这条件有限，没法去深入分析这个问题
很奇怪的是，我从uboot中刷入仅有内核的固件时，它会自动重启到原始状态，跟没刷之前一样，甚至WiFi的名字等等数据都还在

Mrdeng233, 2025-04-23:
> > 如果64位系统因为内存不足无法开机，那么如何编译32位系统呢，貌似我没有在编译界面找到32位的选项，隔壁的qsdk倒是可以编译32位的
> 
> 参考这个 https://github.com/kkstone/Actions-OpenWrt-RAX3000Q/
> 
> 说实话，我觉得应该不是内存不足导致的？ 即使内存不足，内核最起码得在串口有输出吧？（不过我这条件有限，没法去深入分析这个问题

我在uboot刷入仅带内核的固件时，会自动重启到没刷入的固件，数据都在的那种，原系统是nwrt。

lurenjia1213, 2025-04-25:
我来补充一点东西，如果有人有空可以看看
[qsdk-arm-trusted-firmware](https://git.codelinaro.org/clo/qsdk/atf-qsdk/arm-trusted-firmware/-/commits/NHSS.QSDK.12.4.5/?ref_type=heads)

LoadingZhang, 2025-04-26:
会不会跟刷了uboot有关，只能原版uboot启动

lurenjia1213, 2025-04-26:
> 会不会跟刷了uboot有关，只能原版uboot启动

应该和uboot无关，我当时用ida静态看过uboot，在scm call的时候应该是一样的。
scm call之后应该就进入另外一个东西了🤔
但是我没搞过arm trustzone😇

Mrdeng233, 2025-05-06:
1 > 会不会跟刷了uboot有关，只能原版uboot启动
> 
> 应该和uboot无关，我当时用ida静态看过uboot，在输出最后那行之后应该就进入另外一个东西了🤔，不归uboot 但是我没搞过arm trustzone😇
> 
> 会不会是传入参数有问题呢？有可能，但是我看大致是没毛病的

我个人感觉应该是64bit和32bit的问题,目前来看,我刷入你提供的kkstone的固件是可以正常启动的，并且我注意到nwrt和kkstone中的opkg软件源是32bit的arm_cortex-a7_neon-vfpv4而不是64bit的aarch64_cortex-A53。

Mrdeng233, 2025-05-06:
> 如果有512M的内存的话可以换个试一下 😅

这玩意内存是集成在soc中，想换都换不了。

lurenjia1213, 2025-05-07:
> 1 > 会不会跟刷了uboot有关，只能原版uboot启动
> > 
> > 应该和uboot无关，我当时用ida静态看过uboot，在输出最后那行之后应该就进入另外一个东西了🤔，不归uboot 但是我没搞过arm trustzone😇
> > 
> > 会不会是传入参数有问题呢？有可能，但是我看大致是没毛病的
> 
> 我个人感觉应该是64bit和32bit的问题,目前来看,我刷入你提供的kkstone的固件是可以正常启动的，并且我注意到nwrt和kkstone中的opkg软件源是32bit的arm_cortex-a7_neon-vfpv4而不是64bit的aarch64_cortex-A53。

疑惑的地方就是为什么64bit没法启动😂

Mrdeng233, 2025-05-07:
> 疑惑的地方就是为什么64bit没法启动 😂

要注意的是，这个固件在你提供的kkstone固件中貌似并没有加载nss驱动，在刷入之后，内存的最大占用达到了230MB（未使用任何固件）, 同时我使用的cr660x在开启passwall等代理工具时，内存占用仅有150多MB。我个人认为可能是由于qca-WiFi及nss硬件加速的驱动占用了过多的内存导致的，思路来源于zn-m2，zn-m2这类原版仅有256MB的ipq6000类处理器在使用nss硬件加速时打开WiFi仅能使用32bit固件，否则会导致内存溢出，参考网站为https://www.right.com.cn/forum/thread-7827262-1-1.html

lurenjia1213, 2025-05-07:
> 要注意的是，这个固件在你提供的kkstone固件中貌似并没有加载nss驱动，在刷入之后，内存的最大占用达到了230MB（未使用任何固件）, [...]

`[   15.986180] (unnamed net_device) (uninitialized): nss_dp_gmac: Registering netdev eth%d(qcom-id:1) with GMAC, mac_base: 0x90a60000
[   15.988314] (unnamed net_device) (uninitialized): nss_dp_gmac: Registering netdev eth%d(qcom-id:2) with GMAC, mac_base: 0x90ba0000
[   15.999753] **********************************************************
[   16.011884] * NSS Data Plane driver
[   16.015850] **********************************************************
[   16.165671] nss_driver - fw of size 781856  bytes copied to load addr: 40000000, nss_id : 0`

kkstone固件的dmesg。
我对此没太多经验，看起来应该是启动了？

不过这个固件的2.4G确实有问题

感觉你的观点是对的

Mrdeng233, 2025-05-07:
ipq6000可以在没有WiFi的情况下正常启动64bit的系统，带WiFi的则需要32bit。接下来我打算在内核中去除WiFi驱动来看看64bit是否可以正常启动。

Mrdeng233, 2025-05-07:
> > > > 1 > 会不会跟刷了uboot有关，只能原版uboot启动
> > > > > 应该和uboot无关，我当时用ida静态看过uboot，在输出最后那行之后应该就进入另外一个东西了🤔，不归uboot 但是我没搞过arm trustzone😇
> > > > > 会不会是传入参数有问题呢？有可能，但是我看大致是没毛病的
> > > > 
> > > > 
> > > > 我个人感觉应该是64bit和32bit的问题，目前来看，我刷入你提供的kkstone的固件是可以正常启动的，并且我注意到nwrt和kkstone中的opkg软件源是32bit的arm_cortex-a7_neon-vfpv4而不是64bit的aarch64_cortex-A53。
> > > 
> > > 
> > > 疑惑的地方就是为什么64bit没法启动 😂
> > 
> > 
> > 要注意的是，这个固件在你提供的kkstone固件中貌似并没有加载nss驱动，在刷入之后，内存的最大占用达到了230MB（未使用任何固件）, 同时我使用的cr660x在开启passwall等代理工具时，内存占用仅有150多MB。我个人认为可能是由于qca-WiFi及nss硬件加速的驱动占用了过多的内存导致的，思路来源于zn-m2，zn-m2这类原版仅有256MB的ipq6000类处理器在使用nss硬件加速时打开WiFi仅能使用32bit固件，否则会导致内存溢出，参考网站为https://www.right.com.cn/forum/thread-7827262-1-1.html
> 
> ```
> [   15.986180] (unnamed net_device) (uninitialized): nss_dp_gmac: Registering netdev eth%d(qcom-id:1) with GMAC, mac_base: 0x90a60000
> [   15.988314] (unnamed net_device) (uninitialized): nss_dp_gmac: Registering netdev eth%d(qcom-id:2) with GMAC, mac_base: 0x90ba0000
> [   15.999753] **********************************************************
> [   16.011884] * NSS Data Plane driver
> [   16.015850] **********************************************************
> [   16.165671] nss_driver - fw of size 781856  bytes copied to load addr: 40000000, nss_id : 0
> ```
> 
> kkstone固件的dmesg。 我对此没太多经验，看起来nss是启动了？
> 
> 不过这个固件的2.4G确实有问题 感觉ipq5000没有折腾的意义了 不过钻个牛角尖，根据你给的那个链接来看，ipq6000能在64bit下正常启动，只是可用内存很少。但ipq5000在用64bit固件时则是根本没有来自Linux内核的输出。

接下来我打算去除qca-WiFi驱动以确认这是由于内存的原因造成的。另外我需要指出的是，ipq系列处理器使用qsdk类编译出来的固件本身就有着较低的内存占用。

Mrdeng233, 2025-05-08:
> ```
> [   15.986180] (unnamed net_device) (uninitialized): nss_dp_gmac: Registering netdev eth%d(qcom-id:1) with GMAC, mac_base: 0x90a60000
> [   15.988314] (unnamed net_device) (uninitialized): [...]

我这次在编译时去除了qca-wifi驱动，然而刷入后仍然无法进入系统，可能是你之前说过的无法引导内核的问题。接下来是换ipq6000（bushi）

kzio111, 2025-06-09:
> > > > 大雕的说法是uboot有问题
> > > > [image]
> > > 
> > > 
> > > 那么我应该选择什么uboot呢，当然要自己编译的也可以
> > > 我也不太清楚，或许你可以去请教一下大雕，而且这个路由器好像有两种版本，具体表现为一种刷入nwrt固件一切正常，另外一种刷入nwrtlan口无法使用
> 
> 你居然能联系上大雕,uboot 目前只见过nwrt 的那个, 而且第二个分区也改了,要是换uboot 第二个分区也要跟着改吧.

应该是

lurenjia1213, 2025-06-15:
> 大雕的说法是uboot有问题
> 
> [image]

scm call的时候吗？我看nwrt那个uboot这部分似乎与git.codelinaro.org上的一样。能不能让大雕把正确的uboot放出来呢？

kzio111, 2025-06-15:
> > 大雕的说法是uboot有问题
> > [image]
> 
> scm call的时候吗？我看nwrt那个uboot这部分似乎与git.codelinaro.org上的一样。能不能让大雕把正确的uboot放出来呢？

大雕应该不会去做这个东西的uboot

lurenjia1213, 2025-06-21:
> > > 大雕的说法是uboot有问题
> > > [image]
> > 
> > 
> > scm call的时候吗？我看nwrt那个uboot这部分似乎与git.codelinaro.org上的一样。能不能让大雕把正确的uboot放出来呢？
> 
> 大雕应该不会去做这个东西的uboot

他不是都已经能启动了吗？那应该就是做好了吧

kzio111, 2025-06-21:
> > > > 大雕的说法是uboot有问题
> > > > [image]
> > > 
> > > 
> > > scm call的时候吗？我看nwrt那个uboot这部分似乎与git.codelinaro.org上的一样。能不能让大雕把正确的uboot放出来呢？
> > 
> > 
> > 大雕应该不会去做这个东西的uboot
> 
> 他不是都已经能启动了吗？那应该就是做好了吧

不知道，你可以去问问他，而且我看rax3000q好像被移除了