coolsnowwolf/lede #13355: QWRT R24.12.12 使用移动宽带在断线后无法重连

FrancisNGG, 2025-02-21:
### 详细叙述

固件版本：QWRT R24.12.12 / LuCI Master (git-24.338.41706-d6ee217) 
宽带类型：中国移动PPPOE
问题叙述：看日志是移动宽带会7天断一次连接，openwrt重连后能获取到ip，但ipv6无法获取，此时状态是无法上网的。

请问下大佬们，这是什么原因引起的？尝试过使用自编译的版本，问题依旧。

### 重复 issue

- [x] 没有类似的 issue

### 具体型号

x86

### 详细日志

Fri Feb 21 01:33:15 2025 daemon.info pppd[8699]: LCP terminated by peer
Fri Feb 21 01:33:15 2025 daemon.info pppd[8699]: Connect time 10080.1 minutes.
Fri Feb 21 01:33:15 2025 daemon.info pppd[8699]: Sent 2864088356 bytes, received 289688384 bytes.
Fri Feb 21 01:33:15 2025 daemon.notice ttyd[11583]: [2025/02/21 01:33:15:8195] N: rops_handle_POLLIN_netlink: DELADDR
Fri Feb 21 01:33:15 2025 daemon.notice ttyd[11583]: [2025/02/21 01:33:15:8196] N: rops_handle_POLLIN_netlink: DELADDR
Fri Feb 21 01:33:15 2025 daemon.notice ttyd[11583]: [2025/02/21 01:33:15:8197] N: rops_handle_POLLIN_netlink: DELADDR
Fri Feb 21 01:33:15 2025 daemon.err odhcp6c[9039]: Failed to send RS (Permission denied)
Fri Feb 21 01:33:15 2025 daemon.err odhcp6c[9039]: Failed to send RELEASE message to ff02::1:2 (Permission denied)
Fri Feb 21 01:33:15 2025 daemon.notice netifd: Network device 'pppoe-wan' link is down
Fri Feb 21 01:33:15 2025 daemon.notice netifd: Network alias 'pppoe-wan' link is down
Fri Feb 21 01:33:15 2025 daemon.notice netifd: Interface 'wan_6' has link connectivity loss
Fri Feb 21 01:33:15 2025 daemon.notice pppd[8699]: Modem hangup
Fri Feb 21 01:33:15 2025 daemon.notice pppd[8699]: Connection terminated.
Fri Feb 21 01:33:15 2025 daemon.info pppd[8699]: Connect time 10080.1 minutes.
Fri Feb 21 01:33:15 2025 daemon.info pppd[8699]: Sent 2864088356 bytes, received 289688384 bytes.
Fri Feb 21 01:33:15 2025 daemon.notice netifd: Interface 'wan' has lost the connection
Fri Feb 21 01:33:15 2025 daemon.notice netifd: wan_6 (9039): Command failed: Permission denied
Fri Feb 21 01:33:15 2025 user.notice modem_hotplug: net slot: /devices/virtual/net/pppoe-wan action: remove
Fri Feb 21 01:33:15 2025 daemon.info pppd[8699]: Exit.
Fri Feb 21 01:33:15 2025 daemon.notice netifd: Interface 'wan' is now down
Fri Feb 21 01:33:15 2025 daemon.notice netifd: Interface 'wan' is setting up now
Fri Feb 21 01:33:15 2025 daemon.notice netifd: wan (25593): /sbin/modprobe: unrecognized option: a
Fri Feb 21 01:33:15 2025 daemon.info modprobe: Usage: 	modprobe [-q] filename
Fri Feb 21 01:33:15 2025 daemon.notice netifd: wan (25593): uci: Entry not found
Fri Feb 21 01:33:15 2025 daemon.notice netifd: wan (25593): sh: out of range
Fri Feb 21 01:33:15 2025 daemon.info pppd[25601]: Plugin pppoe.so loaded.
Fri Feb 21 01:33:15 2025 daemon.info pppd[25601]: PPPoE plugin from pppd 2.4.9
Fri Feb 21 01:33:16 2025 daemon.notice pppd[25601]: pppd 2.4.9 started by root, uid 0
Fri Feb 21 01:33:16 2025 daemon.info pppd[25601]: PPP session is 14052
Fri Feb 21 01:33:16 2025 daemon.warn pppd[25601]: Connected to 2c:97:b1:d0:b3:62 via interface eth6.41
Fri Feb 21 01:33:16 2025 daemon.info pppd[25601]: Using interface pppoe-wan
Fri Feb 21 01:33:16 2025 daemon.notice pppd[25601]: Connect: pppoe-wan <--> eth6.41
Fri Feb 21 01:33:16 2025 user.notice modem_hotplug: net slot: /devices/virtual/net/pppoe-wan action: add
Fri Feb 21 01:33:16 2025 daemon.info pppd[25601]: syncppp not active
Fri Feb 21 01:33:16 2025 daemon.info pppd[25601]: Remote message: Authentication success,Welcome!
Fri Feb 21 01:33:16 2025 daemon.notice pppd[25601]: PAP authentication succeeded
Fri Feb 21 01:33:16 2025 daemon.notice pppd[25601]: peer from calling number 2C:97:B1:D0:B3:62 authorized
Fri Feb 21 01:33:16 2025 daemon.notice netifd: Network device 'pppoe-wan' link is up
Fri Feb 21 01:33:16 2025 daemon.notice netifd: Network alias 'pppoe-wan' link is up
Fri Feb 21 01:33:16 2025 daemon.notice netifd: Interface 'wan_6' has link connectivity
Fri Feb 21 01:33:16 2025 daemon.notice pppd[25601]: local  LL address fe80::90bc:6c90:823a:93de
Fri Feb 21 01:33:16 2025 daemon.notice pppd[25601]: remote LL address fe80::2e97:b1ff:fed0:b362
Fri Feb 21 01:33:16 2025 daemon.notice netifd: Interface 'wan' is now up
Fri Feb 21 01:33:16 2025 daemon.notice pppd[25601]: local  IP address 172.70.228.82
Fri Feb 21 01:33:16 2025 daemon.notice ttyd[11583]: [2025/02/21 01:33:16:3662] N: rops_handle_POLLIN_netlink: DELADDR
Fri Feb 21 01:33:16 2025 daemon.notice pppd[25601]: remote IP address 172.70.0.1
Fri Feb 21 01:33:16 2025 daemon.notice pppd[25601]: primary   DNS address 120.196.165.24
Fri Feb 21 01:33:16 2025 daemon.notice pppd[25601]: secondary DNS address 211.136.192.6
Fri Feb 21 01:33:16 2025 user.notice firewall: Reloading firewall due to ifup of wan (pppoe-wan)
Fri Feb 21 01:33:16 2025 daemon.err odhcp6c[9039]: Failed to send SOLICIT message to ff02::1:2 (Permission denied)
Fri Feb 21 01:33:16 2025 daemon.notice netifd: Interface 'wan_6' is now down
Fri Feb 21 01:33:16 2025 daemon.notice netifd: Interface 'wan_6' is disabled
Fri Feb 21 01:33:16 2025 daemon.notice netifd: Interface 'wan_6' has link connectivity loss
Fri Feb 21 01:33:16 2025 user.notice ddns-scripts[26238]: myddns_ipv4: PID '26238' started at 2025-02-21 01:33
Fri Feb 21 01:33:16 2025 user.warn ddns-scripts[26238]: myddns_ipv4: Service section disabled! - TERMINATE
Fri Feb 21 01:33:16 2025 user.warn ddns-scripts[26238]: myddns_ipv4: PID '26238' exit WITH ERROR '1' at 2025-02-21 01:33
Fri Feb 21 01:33:16 2025 user.notice firewall: Reloading firewall due to ifupdate of wan (pppoe-wan)
Fri Feb 21 01:33:18 2025 user.info adblock-4.1.5-8[27015]: adblock is currently disabled, please set adb_enabled to '1' to use this service
Fri Feb 21 02:00:00 2025 cron.err crond[11475]: USER root pid 30992 cmd echo "" > /tmp/log/msd_lite.log
Fri Feb 21 02:00:00 2025 cron.err crond[11475]: USER root pid 30993 cmd /usr/share/shadowsocksr/ssrplusupdate.sh >/var/log/ssrplus.log
Fri Feb 21 03:33:17 2025 daemon.notice ttyd[11583]: [2025/02/21 03:33:17:4661] N: rops_handle_POLLIN_netlink: DELADDR
Fri Feb 21 04:33:38 2025 daemon.notice netifd: Network device 'eth5' link is down
Fri Feb 21 04:33:38 2025 kern.err kernel: [1189673.223475] bnx2x 0000:03:00.0 eth5: NIC Link is Down
Fri Feb 21 04:33:38 2025 kern.info kernel: [1189673.223945] br-lan: port 6(eth5) entered disabled state
Fri Feb 21 04:34:06 2025 daemon.notice netifd: Network device 'eth5' link is up
Fri Feb 21 04:34:06 2025 kern.info kernel: [1189700.804049] bnx2x 0000:03:00.0 eth5: NIC Link is Up, 2500 Mbps full duplex, Flow control: ON - receive & transmit
Fri Feb 21 04:34:06 2025 kern.info kernel: [1189700.804507] br-lan: port 6(eth5) entered blocking state
Fri Feb 21 04:34:06 2025 kern.info kernel: [1189700.804855] br-lan: port 6(eth5) entered forwarding state

yanxin152133, 2025-02-24:
最好最快最省心的解决办法就是禁用ipv6  ：）

viqbgrg, 2025-03-07:
hanwckf/immortalwrt-mt798x#269

FrancisNGG, 2025-04-01:
> [hanwckf/immortalwrt-mt798x#269](https://github.com/hanwckf/immortalwrt-mt798x/issues/269) [hanwckf/immortalwrt-mt798x#57](https://github.com/hanwckf/immortalwrt-mt798x/issues/57)

感谢大佬，已通过打补丁的方式解决，山不转水转

closed 2025-04-01 by FrancisNGG