coolsnowwolf/lede #13362: H69K风扇驱动不正常

CsoWhy, 2025-02-24:
### 详细叙述

求助大佬，H69K模板编译出来之后，风扇智能高速运转，dmesg | grep pwm 没有找到信息，看起来没加载成功。
配置 cooling_device0 的 cur_state 没用，iStoreOS 的看起来没有这个问题，看了半天 rk3568 相关的几个 dtsi，没看出什么问题。

### 重复 issue

- [x] 没有类似的 issue

### 具体型号

HINLINK H69K

### 详细日志

暂无

CsoWhy, 2025-02-27:
仔细看了 linux 内核对 rk356x.dtsi 也确定有 pwm0 的描述，hinlink 的 dts 也有 pwm-fan 的配置，但内核信息确实看不到 pwm-fan 的加载。

coolsnowwolf, 2025-03-02:
解决了

CsoWhy, 2025-03-02:
> 解决了

我看处理方式是内核配置里开启 CONFIG_SENSORS_PWM_FAN=y，我本地已经开启这个试过了，没用。内核仍然检测不到 pwm_fan，风扇还是会全速运转。

CsoWhy, 2025-03-02:
总感觉是驱动的问题

coolsnowwolf, 2025-03-03:
已修复

CsoWhy, 2025-03-03:
> 已修复

好滴 等着大佬的新代码😊

CsoWhy, 2025-03-03:
> 已修复

大神，代码已经推了吗？我看 Commit 最新的还是上次的 CONFIG_SENSORS_PWM_FAN=y 改动

aiamadeus, 2025-03-03:
改了，加了自动控制

CsoWhy, 2025-03-03:
> 你需要的可能是这样的一个插件
> 
> [image]

和这个插件没关系 因为手动在 CLI 里对 pwm 调节也是没用的

CsoWhy, 2025-03-03:
@coolsnowwolf 修复好了 非常感谢

CsoWhy, 2025-03-03:
请教下L大 我之前自己研究了好几天没解决 想学习一下原因

我看其实主要改动就是把 pwm-fan 的描述从父 dtsi 文件迁移到了具体的 dts 文件中
我做了对比测试，确实移动到 dts 文件中就好了😂 

想请教下具体原因，是因为 pwm-fan 在父 dtsi 文件中加载的时机顺序不对吗？