coolsnowwolf/lede #13365: Banana BPI-R4迅雷下载断流

fantasykan, 2025-02-25:
### 详细叙述

只要局域网内有主机在用迅雷下载，下载速度比较快的时候（譬如大于10MB/s），此时ping网关延迟会增加，并且会出现断流的情况，ping局域网中的其他机器，一切正常

### 重复 issue

- [x] 没有类似的 issue

### 具体型号

Banana BPI-R4

### 详细日志

[image]
[image]

coolsnowwolf, 2025-03-02:
具体日志？

fantasykan, 2025-03-03:
[image]
[image]

不知道这两个日志可以吗

fantasykan, 2025-03-18:
[image]

6.6内核编译失败，网上找了两个6.6内核的固件，问题依旧

closed 2026-09-05 by coolsnowwolf