coolsnowwolf/lede #13369: mt7925网卡wifi无法启动（无信号强度无噪声）

xiaouex, 2025-02-26:
### 详细叙述

[image]
配置如图，我也不奢求能用6ghz，开5g都不行，手机能搜到这个wifi，但是连接不了，web端这边会显示手机连接上了但是手机这边一直是正在连接，然后过一会就连接失败。

### 重复 issue

- [x] 没有类似的 issue

### 具体型号

mt7925

### 详细日志

Thu Feb 27 02:53:19 2025 daemon.notice hostapd: Remove interface 'wlan0'
Thu Feb 27 02:53:19 2025 daemon.notice hostapd: wlan0: interface state ENABLED->DISABLED
Thu Feb 27 02:53:19 2025 daemon.notice hostapd: wlan0: AP-DISABLED
Thu Feb 27 02:53:19 2025 daemon.notice hostapd: wlan0: CTRL-EVENT-TERMINATING
Thu Feb 27 02:53:19 2025 daemon.err hostapd: rmdir[ctrl_interface=/var/run/hostapd]: Permission denied
Thu Feb 27 02:53:19 2025 daemon.notice hostapd: nl80211: deinit ifname=wlan0 disabled_11b_rates=0
Thu Feb 27 02:53:20 2025 daemon.notice hostapd: Configuration file: /var/run/hostapd-phy0.conf (phy wlan0) --> new PHY
Thu Feb 27 02:53:20 2025 daemon.notice hostapd: wlan0: interface state UNINITIALIZED->COUNTRY_UPDATE
Thu Feb 27 02:53:20 2025 daemon.notice hostapd: wlan0: interface state COUNTRY_UPDATE->HT_SCAN
Thu Feb 27 02:53:21 2025 daemon.notice hostapd: Switch own primary and secondary channel to get secondary channel with no Beacons from other BSSes
Thu Feb 27 02:53:21 2025 daemon.notice hostapd: wlan0: interface state HT_SCAN->ENABLED
Thu Feb 27 02:53:21 2025 daemon.notice hostapd: wlan0: AP-ENABLED
Thu Feb 27 02:53:21 2025 daemon.notice netifd: Network device 'wlan0' link is up
Thu Feb 27 02:53:22 2025 daemon.notice ksmbd: Stopping Ksmbd userspace service.
Thu Feb 27 02:53:22 2025 daemon.notice ksmbd: Starting Ksmbd userspace service.
Thu Feb 27 02:53:23 2025 daemon.warn odhcpd[2226]: A default route is present but there is no public prefix on lan thus we don't announce a default route!
Thu Feb 27 02:53:33 2025 daemon.info hostapd: wlan0: STA 8e:37:5c:67:42:5a IEEE 802.11: authenticated
Thu Feb 27 02:53:33 2025 daemon.info hostapd: wlan0: STA 8e:37:5c:67:42:5a IEEE 802.11: associated (aid 1)
Thu Feb 27 02:53:33 2025 daemon.notice hostapd: wlan0: AP-STA-CONNECTED 8e:37:5c:67:42:5a auth_alg=sae
Thu Feb 27 02:53:33 2025 daemon.info hostapd: wlan0: STA 8e:37:5c:67:42:5a RADIUS: starting accounting session DFDEE85755EDBE72
Thu Feb 27 02:53:33 2025 daemon.info hostapd: wlan0: STA 8e:37:5c:67:42:5a WPA: pairwise key handshake completed (RSN)
Thu Feb 27 02:53:33 2025 daemon.notice hostapd: wlan0: EAPOL-4WAY-HS-COMPLETED 8e:37:5c:67:42:5a
Thu Feb 27 02:53:51 2025 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED 8e:37:5c:67:42:5a
Thu Feb 27 02:56:13 2025 daemon.warn odhcpd[2226]: A default route is present but there is no public prefix on lan thus we don't announce a default route!
Thu Feb 27 02:58:49 2025 daemon.notice hostapd: Remove interface 'wlan0'
Thu Feb 27 02:58:49 2025 daemon.notice hostapd: wlan0: interface state ENABLED->DISABLED
Thu Feb 27 02:58:49 2025 daemon.notice hostapd: wlan0: AP-DISABLED
Thu Feb 27 02:58:49 2025 daemon.notice hostapd: wlan0: CTRL-EVENT-TERMINATING
Thu Feb 27 02:58:49 2025 daemon.err hostapd: rmdir[ctrl_interface=/var/run/hostapd]: Permission denied
Thu Feb 27 02:58:49 2025 daemon.notice hostapd: nl80211: deinit ifname=wlan0 disabled_11b_rates=0
Thu Feb 27 02:58:49 2025 daemon.notice hostapd: Configuration file: /var/run/hostapd-phy0.conf (phy wlan0) --> new PHY
Thu Feb 27 02:58:50 2025 daemon.notice hostapd: wlan0: interface state UNINITIALIZED->COUNTRY_UPDATE
Thu Feb 27 02:58:50 2025 daemon.notice hostapd: wlan0: interface state COUNTRY_UPDATE->HT_SCAN
Thu Feb 27 02:58:51 2025 daemon.notice ksmbd: Stopping Ksmbd userspace service.
Thu Feb 27 02:58:51 2025 daemon.notice ksmbd: Starting Ksmbd userspace service.
Thu Feb 27 02:58:51 2025 daemon.notice hostapd: 20/40 MHz operation not permitted on channel pri=11 sec=7 based on overlapping BSSes
Thu Feb 27 02:58:51 2025 daemon.notice hostapd: Fallback to 20 MHz
Thu Feb 27 02:58:52 2025 daemon.notice hostapd: wlan0: interface state HT_SCAN->ENABLED
Thu Feb 27 02:58:52 2025 daemon.notice hostapd: wlan0: AP-ENABLED
Thu Feb 27 02:58:52 2025 daemon.warn odhcpd[2226]: A default route is present but there is no public prefix on lan thus we don't announce a default route!
Thu Feb 27 02:59:07 2025 daemon.info hostapd: wlan0: STA 8e:37:5c:67:42:5a IEEE 802.11: authenticated
Thu Feb 27 02:59:07 2025 daemon.info hostapd: wlan0: STA 8e:37:5c:67:42:5a IEEE 802.11: associated (aid 1)
Thu Feb 27 02:59:08 2025 daemon.notice hostapd: wlan0: AP-STA-CONNECTED 8e:37:5c:67:42:5a auth_alg=sae
Thu Feb 27 02:59:08 2025 daemon.info hostapd: wlan0: STA 8e:37:5c:67:42:5a RADIUS: starting accounting session 386BDCBA13BDAAA5
Thu Feb 27 02:59:08 2025 daemon.info hostapd: wlan0: STA 8e:37:5c:67:42:5a WPA: pairwise key handshake completed (RSN)
Thu Feb 27 02:59:08 2025 daemon.notice hostapd: wlan0: EAPOL-4WAY-HS-COMPLETED 8e:37:5c:67:42:5a
Thu Feb 27 02:59:26 2025 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED 8e:37:5c:67:42:5a

xiaouex, 2025-03-01:
> 不要想了，这个不好修，牵一发而动全身！

还真是，我尝试从openwrt搬东西，结果越搬东西越多

coolsnowwolf, 2025-03-02:
hostapd 要更新版本了

xiaouex, 2025-03-02:
有一说一，用immort和openwrt都编译过一遍，升级了hostapd后确实有了wifi7,但是确实还是用不了，wifi连上不能上网，但是确实能连上，所以可能是驱动的问题吧

aiamadeus, 2025-03-20:
[image]
6.6 内核上实测没有问题

coolsnowwolf, 2026-09-05:
当前 mt76 已更新到 2026-09-01 Git HEAD（commit 8f7aa30c44b），hostapd/mt76 的新版本已包含后续 Wi‑Fi 7/MT7925 兼容修复。原日志实际显示 SAE 认证、四次握手和 AP-STA-CONNECTED 均成功，随后客户端主动断开，没有驱动启动失败证据；当前无法复现为仓库源码问题，现关闭。

closed 2026-09-05 by coolsnowwolf