coolsnowwolf/lede #13371: fine3399设备无法启动

1967937965, 2025-02-27:
### 详细叙述

经测试只有这个版本的能正常启动
53661b6a154c17f3d52553decd160239be8a2808

### 重复 issue

- [x] 没有类似的 issue

### 具体型号

fine3399

### 详细日志

无

aiamadeus, 2025-02-28:
#13253 人家能启动，试下 https://github.com/coolsnowwolf/lede/commit/bac0af3dec67781112f47203fedb02dc035fe942

x2358473, 2025-03-03:
> [#13253](https://github.com/coolsnowwolf/lede/issues/13253) 人家能启动，试下 [bac0af3](https://github.com/coolsnowwolf/lede/commit/bac0af3dec67781112f47203fedb02dc035fe942)

请问那个补丁要怎么打。https://github.com/coolsnowwolf/lede/issues/13253 这个issues中说打上补丁就可以了。但要如何打补丁。我也是编译后启动直接不亮灯。

x2358473, 2025-03-07:
[image]使用tf卡启动报这个错。是什么原因。emmc已经清空了，里面只有loader。

aiamadeus, 2025-03-07:
trust.img.. 你用的 rkbin 啊。。要把 emmc 的 loader 给扬了

x2358473, 2025-03-07:
> trust.img.. 你用的 rkbin 啊。。要把 emmc 的 loader 给扬了

请问要用什么bin去扬loader

aiamadeus, 2025-03-12:
找了能启动的固件，启动后 dd 清掉 emmc 的前 32M

x2358473, 2025-03-21:
> 找个能启动的固件，启动后 dd 清掉 emmc 的前 32M

感谢。使用sudo dd if=/dev/zero of=/dev/mmcblkX bs=1M count=32 （mmcblkX目标为 eMMC 设备节点，需替换为实际设备名）把loader清了确实可以用了。还是大雕的固件方便。想怎么编译就怎么编译

x2358473, 2025-03-21:
> 找个能启动的固件，启动后 dd 清掉 emmc 的前 32M

大佬。把loader清了之后我用tf卡可以启动了。但是如果装了nvme的话。会导致重启无法启动。这个有什么办法吗。只能断电才能启动。变成无法重启了

aiamadeus, 2025-03-21:
revert 掉这个试试 https://github.com/coolsnowwolf/lede/commit/e43d4db7c057a52fe85843a37950d937670fcb04

fisher423, 2025-04-07:
> > 找个能启动的固件，启动后 dd 清掉 emmc 的前 32M
> 
> 大佬。把loader清了之后我用tf卡可以启动了。但是如果装了nvme的话。会导致重启无法启动。这个有什么办法吗。只能断电才能启动。变成无法重启了

是不是用emmc还是无法启动

coolsnowwolf, 2026-09-05:
根据后续测试，问题根因是设备 eMMC 中残留旧 loader；清理 eMMC 前 32M 后，TF 卡启动已恢复。该问题不需要 LEDE 源码变更，按 issue 中已验证的处理方式关闭。

closed 2026-09-05 by coolsnowwolf