coolsnowwolf/lede #13375: quectel_MHI无法在ipq807x平台下编译

c29tZSBvbmU, 2025-03-01:
### 详细叙述

好像是nss的问题 在ipq807x报错
`ERROR: modpost: "rmnet_nss_callbacks" [/home/op/lede/lede/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/pcie_mhi-1.3.6/pcie_mhi.ko] undefined!`
但是ipq60xx能正常编译

### 重复 issue

- [x] 没有类似的 issue

### 具体型号

aw1000

### 详细日志

make[3]: Entering directory '/home/op/lede/lede/package/wwan/driver/quectel_MHI'
rm -f /home/op/lede/lede/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/pcie_mhi-1.3.6/.built
touch /home/op/lede/lede/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/pcie_mhi-1.3.6/.built_check
make -C "/home/op/lede/lede/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.1.128" ARCH="arm64" CROSS_COMPILE="aarch64-openwrt-linux-musl-" CXXFLAGS="-Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/op/lede/lede/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/pcie_mhi-1.3.6=pcie_mhi-1.3.6 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" M="/home/op/lede/lede/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/pcie_mhi-1.3.6"  modules
make[4]: Entering directory '/home/op/lede/lede/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.1.128'
  MODPOST /home/op/lede/lede/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/pcie_mhi-1.3.6/Module.symvers
ERROR: modpost: "rmnet_nss_callbacks" [/home/op/lede/lede/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/pcie_mhi-1.3.6/pcie_mhi.ko] undefined!
make[5]: *** [scripts/Makefile.modpost:127: /home/op/lede/lede/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/pcie_mhi-1.3.6/Module.symvers] Error 1
make[4]: *** [Makefile:1961: modpost] Error 2
make[4]: Leaving directory '/home/op/lede/lede/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.1.128'
make[3]: *** [Makefile:49: /home/op/lede/lede/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/pcie_mhi-1.3.6/.built] Error 2
make[3]: Leaving directory '/home/op/lede/lede/package/wwan/driver/quectel_MHI'
time: package/wwan/driver/quectel_MHI/compile#0.23#0.51#0.75
    ERROR: package/wwan/driver/quectel_MHI failed to build.
make[2]: *** [package/Makefile:116: package/wwan/driver/quectel_MHI/compile] Error 1
make[2]: Leaving directory '/home/op/lede/lede'
make[1]: *** [package/Makefile:110: /home/op/lede/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/op/lede/lede'
make: *** [/home/op/lede/lede/include/toplevel.mk:231：world] 错误 2

xiaomeng9597, 2025-03-01:
广和通的驱动没人维护了的，修个der啊😂

c29tZSBvbmU, 2025-03-01:
> 广和通的驱动没人维护了的，修个der啊😂

这不是移远的吗😂

xiaomeng9597, 2025-03-01:
> > 广和通的驱动没人维护了的，修个der啊😂
> 
> 这不是移远的吗😂

唉，没注意看，还真是，不好意思啊😂

coolsnowwolf, 2025-03-02:
已修复

closed 2025-03-07 by aiamadeus