coolsnowwolf/lede #13377: 烦请大佬在百忙之中抽空升级下go的版本

bi4tms, 2025-03-04:
### 详细叙述

最新的xray要求1.24的编译环境，目前go的版本为1.23.6，导致编译失败
烦请大佬在百忙之中抽空升级go的版本
@coolsnowwolf 

### 重复 issue

- [x] 没有类似的 issue

### 具体型号

x86

### 详细日志

无

gssiwmd, 2025-03-04:
同样的问题！

GreatMichaelLee, 2025-03-05:
time: package/feeds/passwall/v2ray-plugin/compile#0.04#0.09#0.13
make[3]: Entering directory '/home/builder/lede_x86/feeds/passwall/xray-core'
rm -f /home/builder/lede_x86/build_dir/target-x86_64_musl/Xray-core-25.3.3/.built
touch /home/builder/lede_x86/build_dir/target-x86_64_musl/Xray-core-25.3.3/.built_check
CONFIG_GOLANG_MOD_CACHE_WORLD_READABLE="" GO_BUILD_CACHE_DIR="/home/builder/lede_x86/tmp/go-build" GO_MOD_CACHE_DIR="/home/builder/lede_x86/dl/go-mod-cache" GO_MOD_ARGS="-modcacherw" GO_PKG="github.com/xtls/xray-core" GO_INSTALL_EXTRA="" GO_INSTALL_ALL="" GO_SOURCE_ONLY="" GO_BUILD_PKG="github.com/xtls/xray-core/main" GO_EXCLUDES="" GO_GO_GENERATE="" GO_INSTALL_BIN_PATH="/usr/bin" BUILD_DIR="/home/builder/lede_x86/build_dir/target-x86_64_musl/Xray-core-25.3.3" GO_BUILD_DIR="/home/builder/lede_x86/build_dir/target-x86_64_musl/Xray-core-25.3.3/.go_work/build" GO_BUILD_BIN_DIR="/home/builder/lede_x86/build_dir/target-x86_64_musl/Xray-core-25.3.3/.go_work/build/bin" GO_BUILD_DEPENDS_PATH="/usr/share/gocode" GO_BUILD_DEPENDS_SRC="/home/builder/lede_x86/staging_dir/target-x86_64_musl/usr/share/gocode/src" GOOS="linux" GOARCH="amd64" GO386="" GOAMD64="v1" GOARM="" GOMIPS="" GOMIPS64="" GOPPC64="" CGO_ENABLED=1 CC="x86_64-openwrt-linux-musl-gcc" CXX="x86_64-openwrt-linux-musl-g++" CGO_CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/builder/lede_x86/build_dir/target-x86_64_musl/Xray-core-25.3.3=Xray-core-25.3.3 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CGO_CPPFLAGS="-I/home/builder/lede_x86/staging_dir/toolchain-x86_64_gcc-13.3.0_musl/usr/include -I/home/builder/lede_x86/staging_dir/toolchain-x86_64_gcc-13.3.0_musl/include -I/home/builder/lede_x86/staging_dir/toolchain-x86_64_gcc-13.3.0_musl/include/fortify" CGO_CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/builder/lede_x86/build_dir/target-x86_64_musl/Xray-core-25.3.3=Xray-core-25.3.3 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CGO_LDFLAGS="-L/home/builder/lede_x86/staging_dir/toolchain-x86_64_gcc-13.3.0_musl/usr/lib -L/home/builder/lede_x86/staging_dir/toolchain-x86_64_gcc-13.3.0_musl/lib -fuse-ld=bfd -znow -zrelro" GOPATH="/home/builder/lede_x86/build_dir/target-x86_64_musl/Xray-core-25.3.3/.go_work/build" GOCACHE="/home/builder/lede_x86/tmp/go-build" GOMODCACHE="/home/builder/lede_x86/dl/go-mod-cache" GOENV=off GOTOOLCHAIN=local /usr/bin/env bash /home/builder/lede_x86/feeds/packages/lang/golang//golang-build.sh build -v -buildvcs=false -trimpath -ldflags "all=-buildid '1741089616' -linkmode external -extldflags '-L/home/builder/lede_x86/staging_dir/toolchain-x86_64_gcc-13.3.0_musl/usr/lib -L/home/builder/lede_x86/staging_dir/toolchain-x86_64_gcc-13.3.0_musl/lib -fuse-ld=bfd -Wl,-z,now -Wl,-z,relro'"        -ldflags " -X github.com/xtls/xray-core/core.build=OpenWrt -X github.com/xtls/xray-core/core.version=25.3.3 -buildid '1741089616' -linkmode external -extldflags '-L/home/builder/lede_x86/staging_dir/toolchain-x86_64_gcc-13.3.0_musl/usr/lib -L/home/builder/lede_x86/staging_dir/toolchain-x86_64_gcc-13.3.0_musl/lib -fuse-ld=bfd -Wl,-z,now -Wl,-z,relro'"
Finding targets
go: ../../go.mod requires go >= 1.24 (running go 1.23.6; GOTOOLCHAIN=local)

Building targets
go: ../../go.mod requires go >= 1.24 (running go 1.23.6; GOTOOLCHAIN=local)

make[3]: *** [Makefile:64: /home/builder/lede_x86/build_dir/target-x86_64_musl/Xray-core-25.3.3/.built] Error 1
make[3]: Leaving directory '/home/builder/lede_x86/feeds/passwall/xray-core'
time: package/feeds/passwall/xray-core/compile#0.06#0.14#0.20
    ERROR: package/feeds/passwall/xray-core failed to build.
make[2]: *** [package/Makefile:116: package/feeds/passwall/xray-core/compile] Error 1
make[2]: Leaving directory '/home/builder/lede_x86'
make[1]: *** [package/Makefile:110: /home/builder/lede_x86/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/builder/lede_x86'
make: *** [/home/builder/lede_x86/include/toplevel.mk:231：world] 错误 2

bi4tms, 2025-03-06:
已有PR提出，烦请合并
https://github.com/coolsnowwolf/packages/pull/905
@coolsnowwolf